jerryscript-project/jerryscript #274: Assertion 'str_buf_iter_p <= str_buf_p + source_str_size' failed in convert_string_to_token_transform_escape_seq

renatahodovan, 2015-06-30:
###### Jerry version:
    Checked revision: 2aa0485
    Build: debug.linux
###### OS: 
    Ubuntu 14.10, x86_64
###### Test case:
```javascript
'�'
```
Hex: 27 EF BF BD 27

###### Backtrace:
```javascript
ICE: Assertion 'str_buf_iter_p <= str_buf_p + source_str_size' failed at jerryscript/jerry-core/parser/js/lexer.cpp(convert_string_to_token_transform_escape_seq):558.
Error: ERR_FAILED_INTERNAL_ASSERTION

Program received signal SIGABRT, Aborted.
0x0000000000485c23 in syscall_2_asm () at jerryscript/jerry-libc/target/linux/jerry-asm.S:32
32    SYSCALL_2
(gdb) bt
# 0  0x0000000000485c23 in syscall_2_asm () at jerryscript/jerry-libc/target/linux/jerry-asm.S:32
# 1  0x00000000004003c2 in syscall_2 (syscall_no=62, arg1=21158, arg2=6)
    at jerryscript/jerry-libc/target/linux/jerry-libc-target.c:96
# 2  0x0000000000400563 in abort () at jerryscript/jerry-libc/target/linux/jerry-libc-target.c:175
# 3  0x000000000043a7c7 in jerry_fatal (code=ERR_FAILED_INTERNAL_ASSERTION) at jerryscript/jerry-core/jrt/jrt-fatals.cpp:65
# 4  0x000000000043a816 in jerry_assert_fail (assertion=0x48ffe0 "str_buf_iter_p <= str_buf_p + source_str_size", 
    file=0x48ff88 "jerryscript/jerry-core/parser/js/lexer.cpp", 
    function=0x48daa0 <_ZZL44convert_string_to_token_transform_escape_seq10token_typePKhmE8__func__.lto_priv.171> "convert_string_to_token_transform_escape_seq", line=558) at jerryscript/jerry-core/jrt/jrt-fatals.cpp:92
# 5  0x000000000040c2b4 in convert_string_to_token_transform_escape_seq (tok_type=TOK_STRING, 
    source_str_p=0x6b5781 <_ZL13source_buffer.lto_priv.1070+1> "\357\277\275'", source_str_size=3)
    at jerryscript/jerry-core/parser/js/lexer.cpp:558
# 6  0x000000000040d904 in parse_string () at jerryscript/jerry-core/parser/js/lexer.cpp:951
# 7  0x000000000040dfaa in lexer_next_token_private () at jerryscript/jerry-core/parser/js/lexer.cpp:1125
# 8  0x000000000040f142 in lexer_next_token () at jerryscript/jerry-core/parser/js/lexer.cpp:1286
# 9  0x0000000000444612 in skip_token() [clone .lto_priv.528] () at jerryscript/jerry-core/parser/js/parser.cpp:102
# 10 0x0000000000444851 in skip_newlines() [clone .lto_priv.524] () at jerryscript/jerry-core/parser/js/parser.cpp:135
# 11 0x00000000004411e7 in parser_parse_program (source_p=0x6b5780 <_ZL13source_buffer.lto_priv.1070> "'\357\277\275'", source_size=5, 
    in_function=false, in_eval=false, is_strict=false, out_opcodes_p=0x7fffffffd5f0)
    at jerryscript/jerry-core/parser/js/parser.cpp:3075
# 12 0x0000000000441370 in parser_parse_script (source=0x6b5780 <_ZL13source_buffer.lto_priv.1070> "'\357\277\275'", source_size=5, 
    opcodes_p=0x7fffffffd5f0) at jerryscript/jerry-core/parser/js/parser.cpp:3145
# 13 0x00000000004856e0 in jerry_parse (source_p=0x6b5780 <_ZL13source_buffer.lto_priv.1070> "'\357\277\275'", source_size=5)
    at jerryscript/jerry-core/jerry.cpp:1341
# 14 0x00000000004842ed in main (argc=3, argv=0x7fffffffd898) at jerryscript/main-linux.cpp:282
```

kkristof, 2015-07-30:
Currently this is working for me, maybe we should close this issue.

egavrin, 2015-11-19:
Works on master, closing

closed 2015-11-19 by egavrin