jerryscript-project/jerryscript #278: Assertion 'code_p != NULL' failed in ecma_op_eval_chars_buffer

renatahodovan, 2015-06-30:
###### Jerry version:
    Checked revision: 2aa0485
    Build: debug.linux
###### OS: 
    Ubuntu 14.10, x86_64
###### Test case:
```javascript
eval(String());
```

###### Backtrace:
```javascript
ICE: Assertion 'code_p != NULL' failed at jerryscript/jerry-core/ecma/operations/ecma-eval.cpp(ecma_op_eval_chars_buffer):85.
Error: ERR_FAILED_INTERNAL_ASSERTION

Program received signal SIGABRT, Aborted.
0x0000000000485c23 in syscall_2_asm () at jerryscript/jerry-libc/target/linux/jerry-asm.S:32
32    SYSCALL_2
(gdb) bt
#0  0x0000000000485c23 in syscall_2_asm () at jerryscript/jerry-libc/target/linux/jerry-asm.S:32
#1  0x00000000004003c2 in syscall_2 (syscall_no=62, arg1=24981, arg2=6)
    at jerryscript/jerry-libc/target/linux/jerry-libc-target.c:96
#2  0x0000000000400563 in abort () at jerryscript/jerry-libc/target/linux/jerry-libc-target.c:175
#3  0x000000000043a7c7 in jerry_fatal (code=ERR_FAILED_INTERNAL_ASSERTION) at jerryscript/jerry-core/jrt/jrt-fatals.cpp:65
#4  0x000000000043a816 in jerry_assert_fail (assertion=0x49a067 "code_p != NULL", 
    file=0x499fc0 "jerryscript/jerry-core/ecma/operations/ecma-eval.cpp", 
    function=0x48b0e0 <_ZZ25ecma_op_eval_chars_bufferPKhmbbE8__func__.lto_priv.574> "ecma_op_eval_chars_buffer", line=85)
    at jerryscript/jerry-core/jrt/jrt-fatals.cpp:92
#5  0x0000000000447e44 in ecma_op_eval_chars_buffer (code_p=0x0, code_buffer_size=0, is_direct=true, is_called_from_strict_mode_code=false)
    at jerryscript/jerry-core/ecma/operations/ecma-eval.cpp:85
#6  0x0000000000447d94 in ecma_op_eval (code_p=0x7b5c90 <_ZL13mem_heap_area.lto_priv.1016+272>, is_direct=true, is_called_from_strict_mode_code=false)
    at jerryscript/jerry-core/ecma/operations/ecma-eval.cpp:63
#7  0x0000000000474802 in ecma_builtin_global_object_eval (this_arg=4, x=138)
    at jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-global.cpp:86
#8  0x00000000004745fa in ecma_builtin_global_dispatch_routine (builtin_routine_id=1, this_arg_value=4, 
    arguments_list=0x7f5b10 <_ZL13mem_heap_area.lto_priv.1016+262032>, arguments_number=1)
    at jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-global.inc.h:222
#9  0x000000000047a6fb in ecma_builtin_dispatch_routine (builtin_object_id=ECMA_BUILTIN_ID_GLOBAL, builtin_routine_id=1, this_arg_value=4, 
    arguments_list=0x7f5b10 <_ZL13mem_heap_area.lto_priv.1016+262032>, arguments_number=1)
    at jerryscript/jerry-core/ecma/builtin-objects/ecma-builtins.inc.h:302
#10 0x0000000000479b49 in ecma_builtin_dispatch_call (obj_p=0x7b5c30 <_ZL13mem_heap_area.lto_priv.1016+176>, this_arg_value=4, 
    arguments_list_p=0x7f5b10 <_ZL13mem_heap_area.lto_priv.1016+262032>, arguments_list_len=1)
    at jerryscript/jerry-core/ecma/builtin-objects/ecma-builtins.cpp:414
#11 0x0000000000449387 in ecma_op_function_call (func_obj_p=0x7b5c30 <_ZL13mem_heap_area.lto_priv.1016+176>, this_arg_value=4, 
    arguments_list_p=0x7f5b10 <_ZL13mem_heap_area.lto_priv.1016+262032>, arguments_list_len=1)
    at jerryscript/jerry-core/ecma/operations/ecma-function-object.cpp:696
#12 0x0000000000437fa2 in opfunc_call_n (opdata=..., int_data=0x7fffffffd4e0) at jerryscript/jerry-core/vm/opcodes.cpp:817
#13 0x000000000047e78f in vm_loop (int_data_p=0x7fffffffd4e0, run_scope_p=0x0) at jerryscript/jerry-core/vm/vm.cpp:466
#14 0x000000000047ea1c in vm_run_from_pos (opcodes_p=0x7b5d18 <_ZL13mem_heap_area.lto_priv.1016+408>, start_pos=1, this_binding_value=15, 
    lex_env_p=0x7b5ba8 <_ZL13mem_heap_area.lto_priv.1016+40>, is_strict=false, is_eval_code=false)
    at jerryscript/jerry-core/vm/vm.cpp:556
#15 0x000000000047e5f9 in vm_run_global () at jerryscript/jerry-core/vm/vm.cpp:395
#16 0x000000000048572d in jerry_run () at jerryscript/jerry-core/jerry.cpp:1373
#17 0x000000000048430c in main (argc=3, argv=0x7fffffffd8a8) at jerryscript/main-linux.cpp:291
```

egavrin, 2015-11-19:
Fails on current master f17f785b6ec0c71c2ef37e4e80a98aff1eff0963

egavrin, 2015-11-24:
Additional testcase:
```js
eval("")
```

commit cc60b0b3 (2016-01-13): Fix eval fail with empty string

Fixes #278

JerryScript-DCO-1.0-Signed-off-by: pius.lee pius.lee@samsung.com
closed 2016-01-13 by ruben-ayrapetyan