jerryscript-project/jerryscript #279: Assertion 'bytes_copied >= 0' failed in ecma_builtin_global_object_parse_int

renatahodovan, 2015-06-30:
###### Jerry version:
    Checked revision: 2aa0485
    Build: debug.linux
###### OS: 
    Ubuntu 14.10, x86_64
###### Test case:
```javascript
parseInt(RegExp("\u2029"));
```

###### Backtrace:
```javascript
ICE: Assertion 'bytes_copied >= 0' failed at jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-global.cpp(ecma_builtin_global_object_parse_int):119.
Error: ERR_FAILED_INTERNAL_ASSERTION

Program received signal SIGABRT, Aborted.
0x0000000000485c23 in syscall_2_asm () at jerryscript/jerry-libc/target/linux/jerry-asm.S:32
32    SYSCALL_2
(gdb) bt
# 0  0x0000000000485c23 in syscall_2_asm () at jerryscript/jerry-libc/target/linux/jerry-asm.S:32
# 1  0x00000000004003c2 in syscall_2 (syscall_no=62, arg1=30132, arg2=6)
    at jerryscript/jerry-libc/target/linux/jerry-libc-target.c:96
# 2  0x0000000000400563 in abort () at jerryscript/jerry-libc/target/linux/jerry-libc-target.c:175
# 3  0x000000000043a7c7 in jerry_fatal (code=ERR_FAILED_INTERNAL_ASSERTION) at jerryscript/jerry-core/jrt/jrt-fatals.cpp:65
# 4  0x000000000043a816 in jerry_assert_fail (assertion=0x4a2138 "bytes_copied >= 0", 
    file=0x4a2068 "jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-global.cpp", 
    function=0x486d80 <_ZZL36ecma_builtin_global_object_parse_intjjjE8__func__.lto_priv.952> "ecma_builtin_global_object_parse_int", line=119)
    at jerryscript/jerry-core/jrt/jrt-fatals.cpp:92
# 5  0x000000000047495a in ecma_builtin_global_object_parse_int (this_arg=4, string=151, radix=4)
    at jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-global.cpp:119
# 6  0x000000000047475f in ecma_builtin_global_dispatch_routine (builtin_routine_id=47, this_arg_value=4, 
    arguments_list=0x7f5b10 <_ZL13mem_heap_area.lto_priv.1016+262032>, arguments_number=1)
    at jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-global.inc.h:230
# 7  0x000000000047a6fb in ecma_builtin_dispatch_routine (builtin_object_id=ECMA_BUILTIN_ID_GLOBAL, builtin_routine_id=47, this_arg_value=4, 
    arguments_list=0x7f5b10 <_ZL13mem_heap_area.lto_priv.1016+262032>, arguments_number=1)
    at jerryscript/jerry-core/ecma/builtin-objects/ecma-builtins.inc.h:302
# 8  0x0000000000479b49 in ecma_builtin_dispatch_call (obj_p=0x7b5c30 <_ZL13mem_heap_area.lto_priv.1016+176>, this_arg_value=4, 
    arguments_list_p=0x7f5b10 <_ZL13mem_heap_area.lto_priv.1016+262032>, arguments_list_len=1)
    at jerryscript/jerry-core/ecma/builtin-objects/ecma-builtins.cpp:414
# 9  0x0000000000449387 in ecma_op_function_call (func_obj_p=0x7b5c30 <_ZL13mem_heap_area.lto_priv.1016+176>, this_arg_value=4, 
    arguments_list_p=0x7f5b10 <_ZL13mem_heap_area.lto_priv.1016+262032>, arguments_list_len=1)
    at jerryscript/jerry-core/ecma/operations/ecma-function-object.cpp:696
# 10 0x0000000000437fa2 in opfunc_call_n (opdata=..., int_data=0x7fffffffd4e0) at jerryscript/jerry-core/vm/opcodes.cpp:817
# 11 0x000000000047e78f in vm_loop (int_data_p=0x7fffffffd4e0, run_scope_p=0x0) at jerryscript/jerry-core/vm/vm.cpp:466
# 12 0x000000000047ea1c in vm_run_from_pos (opcodes_p=0x7b5d18 <_ZL13mem_heap_area.lto_priv.1016+408>, start_pos=1, this_binding_value=15, 
    lex_env_p=0x7b5ba8 <_ZL13mem_heap_area.lto_priv.1016+40>, is_strict=false, is_eval_code=false)
    at jerryscript/jerry-core/vm/vm.cpp:556
# 13 0x000000000047e5f9 in vm_run_global () at jerryscript/jerry-core/vm/vm.cpp:395
# 14 0x000000000048572d in jerry_run () at jerryscript/jerry-core/jerry.cpp:1373
# 15 0x000000000048430c in main (argc=3, argv=0x7fffffffd8a8) at jerryscript/main-linux.cpp:291
```

galpeter, 2015-07-14:
Proposed fix: #389

egavrin, 2015-07-16:
Merged at db9b74a1147a3698b3b628bc86fadd4a20372ed5

closed 2015-07-16 by egavrin