jerryscript-project/jerryscript #280: Assertion 'lit->get_type () == LIT_STR_T' failed in ecma_init_ecma_string_from_lit_cp

renatahodovan, 2015-06-30:
###### Jerry version:
    Checked revision: ec02ace
    Build: debug.linux
###### OS: 
    Ubuntu 14.10, x86_64
###### Test case:
```javascript
v_0 = {0: delete 3. instanceof foo() ^ {} }
```

###### Backtrace:
```javascript
ICE: Assertion 'lit->get_type () == LIT_STR_T' failed at jerryscript/jerry-core/ecma/base/ecma-helpers-string.cpp(ecma_init_ecma_string_from_lit_cp):347.
Error: ERR_FAILED_INTERNAL_ASSERTION

Program received signal SIGABRT, Aborted.
0x0000000000485ccd in syscall_2_asm () at jerryscript/jerry-libc/target/linux/jerry-asm.S:32
32    SYSCALL_2
(gdb) bt
# 0  0x0000000000485ccd in syscall_2_asm () at jerryscript/jerry-libc/target/linux/jerry-asm.S:32
# 1  0x00000000004003c2 in syscall_2 (syscall_no=62, arg1=1476, arg2=6)
    at jerryscript/jerry-libc/target/linux/jerry-libc-target.c:96
# 2  0x0000000000400563 in abort () at jerryscript/jerry-libc/target/linux/jerry-libc-target.c:175
# 3  0x000000000043a7db in jerry_fatal (code=ERR_FAILED_INTERNAL_ASSERTION) at jerryscript/jerry-core/jrt/jrt-fatals.cpp:65
# 4  0x000000000043a82a in jerry_assert_fail (assertion=0x49cbea "lit->get_type () == LIT_STR_T", 
    file=0x49c9e0 "jerryscript/jerry-core/ecma/base/ecma-helpers-string.cpp", 
    function=0x4899e0 <_ZZL33ecma_init_ecma_string_from_lit_cpP13ecma_string_tN15rcs_recordset_t8record_t10cpointer_tEbE8__func__.lto_priv.734> "ecma_init_ecma_string_from_lit_cp", line=347) at jerryscript/jerry-core/jrt/jrt-fatals.cpp:92
# 5  0x0000000000453c4a in ecma_init_ecma_string_from_lit_cp (string_p=0x7b5c30 <_ZL13mem_heap_area.lto_priv.1016+176>, lit_cp=..., is_stack_var=false)
    at jerryscript/jerry-core/ecma/base/ecma-helpers-string.cpp:347
# 6  0x00000000004543de in ecma_new_ecma_string_from_lit_cp (lit_cp=...)
    at jerryscript/jerry-core/ecma/base/ecma-helpers-string.cpp:571
# 7  0x0000000000430342 in opfunc_delete_var (opdata=..., int_data=0x7fffffffd4d0) at jerryscript/jerry-core/vm/opcodes.cpp:1617
# 8  0x000000000047e839 in vm_loop (int_data_p=0x7fffffffd4d0, run_scope_p=0x0) at jerryscript/jerry-core/vm/vm.cpp:466
# 9  0x00000000004387e7 in opfunc_obj_decl (opdata=..., int_data=0x7fffffffd4d0) at jerryscript/jerry-core/vm/opcodes.cpp:1030
# 10 0x000000000047e839 in vm_loop (int_data_p=0x7fffffffd4d0, run_scope_p=0x0) at jerryscript/jerry-core/vm/vm.cpp:466
# 11 0x000000000047eac6 in vm_run_from_pos (opcodes_p=0x7b5d58 <_ZL13mem_heap_area.lto_priv.1016+472>, start_pos=1, this_binding_value=15, 
    lex_env_p=0x7b5ba8 <_ZL13mem_heap_area.lto_priv.1016+40>, is_strict=false, is_eval_code=false)
    at jerryscript/jerry-core/vm/vm.cpp:556
# 12 0x000000000047e6a3 in vm_run_global () at jerryscript/jerry-core/vm/vm.cpp:395
# 13 0x00000000004857d7 in jerry_run () at jerryscript/jerry-core/jerry.cpp:1373
# 14 0x00000000004843b6 in main (argc=3, argv=0x7fffffffd898) at jerryscript/main-linux.cpp:291
```

kkristof, 2015-07-30:
I think this is solved because now I get reference error. Maybe we should close this issue.

egavrin, 2015-11-19:
```
$ ./build/bin/debug.linux/jerry /tmp/aa.js
$ echo $?
1
```

wateret, 2015-12-04:
@egavrin As @kkristof said, the reason why the TC ends with exit code 1 is that Jerry throws a ReferenceError. I'll create a PR with a regression test.

LaszloLango, 2016-02-08:
Fails on current master (91a0514).

renatahodovan, 2016-02-08:
This is a fuzzer-generated test. Its only expected result is not to fail with an assert.

LaszloLango, 2016-02-08:
Works on current master (91a0514).

closed 2016-02-08 by LaszloLango