jerryscript-project/jerryscript #281: Assertion 'locus < buffer_size' failed in lexer_seek

renatahodovan, 2015-06-30:
###### Jerry version:
    Checked revision: ec02ace
    Build: debug.linux
###### OS: 
    Ubuntu 14.10, x86_64
###### Test case:
```javascript
while(false) foo
```

###### Backtrace:
```javascript
ICE: Assertion 'locus < buffer_size' failed at jerryscript/jerry-core/parser/js/lexer.cpp(lexer_seek):1313.
Error: ERR_FAILED_INTERNAL_ASSERTION

Program received signal SIGABRT, Aborted.
0x0000000000485ccd in syscall_2_asm () at jerryscript/jerry-libc/target/linux/jerry-asm.S:32
32    SYSCALL_2
(gdb) bt
# 0  0x0000000000485ccd in syscall_2_asm () at jerryscript/jerry-libc/target/linux/jerry-asm.S:32
# 1  0x00000000004003c2 in syscall_2 (syscall_no=62, arg1=11434, arg2=6)
    at jerryscript/jerry-libc/target/linux/jerry-libc-target.c:96
# 2  0x0000000000400563 in abort () at jerryscript/jerry-libc/target/linux/jerry-libc-target.c:175
# 3  0x000000000043a7db in jerry_fatal (code=ERR_FAILED_INTERNAL_ASSERTION) at jerryscript/jerry-core/jrt/jrt-fatals.cpp:65
# 4  0x000000000043a82a in jerry_assert_fail (assertion=0x4907ce "locus < buffer_size", 
    file=0x48ffc8 "jerryscript/jerry-core/parser/js/lexer.cpp", 
    function=0x48db91 <_ZZ10lexer_seekmE8__func__.lto_priv.161> "lexer_seek", line=1313)
    at jerryscript/jerry-core/jrt/jrt-fatals.cpp:92
# 5  0x000000000040f232 in lexer_seek (locus=16) at jerryscript/jerry-core/parser/js/lexer.cpp:1313
# 6  0x000000000043fa83 in parse_while_statement (outermost_stmt_label_p=0x7fffffffd380)
    at jerryscript/jerry-core/parser/js/parser.cpp:2251
# 7  0x0000000000440315 in parse_iterational_statement (outermost_named_stmt_label_p=0x0)
    at jerryscript/jerry-core/parser/js/parser.cpp:2540
# 8  0x00000000004404df in parse_statement (outermost_stmt_label_p=0x0) at jerryscript/jerry-core/parser/js/parser.cpp:2668
# 9  0x0000000000440c18 in parse_source_element () at jerryscript/jerry-core/parser/js/parser.cpp:2820
# 10 0x000000000044109a in parse_source_element_list(bool) [clone .lto_priv.559] (is_global=true)
    at jerryscript/jerry-core/parser/js/parser.cpp:3045
# 11 0x000000000044123b in parser_parse_program (source_p=0x6b5780 <_ZL13source_buffer.lto_priv.1070> "while(false) foo", source_size=16, 
    in_function=false, in_eval=false, is_strict=false, out_opcodes_p=0x7fffffffd5f0)
    at jerryscript/jerry-core/parser/js/parser.cpp:3105
# 12 0x00000000004413ba in parser_parse_script (source=0x6b5780 <_ZL13source_buffer.lto_priv.1070> "while(false) foo", source_size=16, 
    opcodes_p=0x7fffffffd5f0) at jerryscript/jerry-core/parser/js/parser.cpp:3174
# 13 0x000000000048578a in jerry_parse (source_p=0x6b5780 <_ZL13source_buffer.lto_priv.1070> "while(false) foo", source_size=16)
    at jerryscript/jerry-core/jerry.cpp:1341
# 14 0x0000000000484397 in main (argc=3, argv=0x7fffffffd898) at jerryscript/main-linux.cpp:282

```

kkristof, 2015-07-30:
It is working for me, maybe we should close this issue.

egavrin, 2015-11-19:
Works on master (f17f785b6ec0c71c2ef37e4e80a98aff1eff0963)

closed 2015-11-19 by egavrin