jerryscript-project/jerryscript #282: Assertion 'token_is (TOK_EOF)' failed in parser_parse_program

renatahodovan, 2015-06-30:
###### Jerry version:
    Checked revision: ec02ace
    Build: debug.linux
###### OS: 
    Ubuntu 14.10, x86_64
###### Test case:
```javascript
switch(foo){} {}
```

###### Backtrace:
```javascript
ICE: Assertion 'token_is (TOK_EOF)' failed at jerryscript/jerry-core/parser/js/parser.cpp(parser_parse_program):3108.
Error: ERR_FAILED_INTERNAL_ASSERTION

Program received signal SIGABRT, Aborted.
0x0000000000485ccd in syscall_2_asm () at jerryscript/jerry-libc/target/linux/jerry-asm.S:32
32    SYSCALL_2
(gdb) bt
# 0  0x0000000000485ccd in syscall_2_asm () at jerryscript/jerry-libc/target/linux/jerry-asm.S:32
# 1  0x00000000004003c2 in syscall_2 (syscall_no=62, arg1=3261, arg2=6)
    at jerryscript/jerry-libc/target/linux/jerry-libc-target.c:96
# 2  0x0000000000400563 in abort () at jerryscript/jerry-libc/target/linux/jerry-libc-target.c:175
# 3  0x000000000043a7db in jerry_fatal (code=ERR_FAILED_INTERNAL_ASSERTION) at jerryscript/jerry-core/jrt/jrt-fatals.cpp:65
# 4  0x000000000043a82a in jerry_assert_fail (assertion=0x498bc8 "token_is (TOK_EOF)", 
    file=0x498958 "jerryscript/jerry-core/parser/js/parser.cpp", 
    function=0x48b740 <_ZZL20parser_parse_programPKhmbbbPPK8opcode_tE8__func__.lto_priv.509> "parser_parse_program", line=3108)
    at jerryscript/jerry-core/jrt/jrt-fatals.cpp:92
# 5  0x0000000000441273 in parser_parse_program (source_p=0x6b5780 <_ZL13source_buffer.lto_priv.1070> "switch(foo){} {}", source_size=16, 
    in_function=false, in_eval=false, is_strict=false, out_opcodes_p=0x7fffffffd5f0)
    at jerryscript/jerry-core/parser/js/parser.cpp:3108
# 6  0x00000000004413ba in parser_parse_script (source=0x6b5780 <_ZL13source_buffer.lto_priv.1070> "switch(foo){} {}", source_size=16, 
    opcodes_p=0x7fffffffd5f0) at jerryscript/jerry-core/parser/js/parser.cpp:3174
# 7  0x000000000048578a in jerry_parse (source_p=0x6b5780 <_ZL13source_buffer.lto_priv.1070> "switch(foo){} {}", source_size=16)
    at jerryscript/jerry-core/jerry.cpp:1341
# 8  0x0000000000484397 in main (argc=3, argv=0x7fffffffd898) at jerryscript/main-linux.cpp:282
```

egavrin, 2015-11-19:
Works on master (f17f785b6ec0c71c2ef37e4e80a98aff1eff0963)

Revealed other issue
```
switch(foo){} {}
               ^
ERROR: Ln 1, Col 16: Unexpected token
```

egavrin, 2015-11-19:
New issue: #725

closed 2015-11-19 by egavrin