jerryscript-project/jerryscript #285: Assertion 'builtin_id == ECMA_BUILTIN_ID_TYPE_ERROR_THROWER' failed in ecma_object_get_class_name

renatahodovan, 2015-07-01:
###### Jerry version:
    Checked revision: 505bead
    Build: debug.linux
###### OS: 
    Ubuntu 14.10, x86_64
###### Test case:
```javascript
NaN == RegExp;
```

###### Backtrace:
```javascript
ICE: Assertion 'builtin_id == ECMA_BUILTIN_ID_TYPE_ERROR_THROWER' failed at jerryscript/jerry-core/ecma/operations/ecma-objects.cpp(ecma_object_get_class_name):612.
Error: ERR_FAILED_INTERNAL_ASSERTION

Program received signal SIGABRT, Aborted.
0x0000000000485d22 in syscall_2_asm () at jerryscript/jerry-libc/target/linux/jerry-asm.S:32
32    SYSCALL_2
(gdb) bt
# 0  0x0000000000485d22 in syscall_2_asm () at jerryscript/jerry-libc/target/linux/jerry-asm.S:32
# 1  0x00000000004003c2 in syscall_2 (syscall_no=62, arg1=4299, arg2=6)
    at jerryscript/jerry-libc/target/linux/jerry-libc-target.c:96
# 2  0x0000000000400563 in abort () at jerryscript/jerry-libc/target/linux/jerry-libc-target.c:175
# 3  0x000000000043a7db in jerry_fatal (code=ERR_FAILED_INTERNAL_ASSERTION) at jerryscript/jerry-core/jrt/jrt-fatals.cpp:65
# 4  0x000000000043a82a in jerry_assert_fail (assertion=0x49aa90 "builtin_id == ECMA_BUILTIN_ID_TYPE_ERROR_THROWER", 
    file=0x49a830 "jerryscript/jerry-core/ecma/operations/ecma-objects.cpp", 
    function=0x48acd0 <_ZZ26ecma_object_get_class_nameP13ecma_object_tE8__func__.lto_priv.605> "ecma_object_get_class_name", line=612)
    at jerryscript/jerry-core/jrt/jrt-fatals.cpp:92
# 5  0x000000000044a86e in ecma_object_get_class_name (obj_p=0x7b5c60 <_ZL13mem_heap_area.lto_priv.1016+224>)
    at jerryscript/jerry-core/ecma/operations/ecma-objects.cpp:612
# 6  0x000000000044248b in ecma_op_general_object_default_value (obj_p=0x7b5c60 <_ZL13mem_heap_area.lto_priv.1016+224>, hint=ECMA_PREFERRED_TYPE_NO)
    at jerryscript/jerry-core/ecma/operations/ecma-objects-general.cpp:529
# 7  0x000000000044a54a in ecma_op_object_default_value (obj_p=0x7b5c60 <_ZL13mem_heap_area.lto_priv.1016+224>, hint=ECMA_PREFERRED_TYPE_NO)
    at jerryscript/jerry-core/ecma/operations/ecma-objects.cpp:388
# 8  0x00000000004462ed in ecma_op_to_primitive (value=115, preferred_type=ECMA_PREFERRED_TYPE_NO)
    at jerryscript/jerry-core/ecma/operations/ecma-conversion.cpp:169
# 9  0x000000000044368e in ecma_op_abstract_equality_compare (x=101, y=115)
    at jerryscript/jerry-core/ecma/operations/ecma-comparison.cpp:183
# 10 0x000000000047af84 in opfunc_equal_value (opdata=..., int_data=0x7fffffffd4b0)
    at jerryscript/jerry-core/vm/opcodes-ecma-equality.cpp:39
# 11 0x000000000047e88e in vm_loop (int_data_p=0x7fffffffd4b0, run_scope_p=0x0) at jerryscript/jerry-core/vm/vm.cpp:466
# 12 0x000000000047eb1b in vm_run_from_pos (opcodes_p=0x7b5d18 <_ZL13mem_heap_area.lto_priv.1016+408>, start_pos=1, this_binding_value=15, 
    lex_env_p=0x7b5ba8 <_ZL13mem_heap_area.lto_priv.1016+40>, is_strict=false, is_eval_code=false)
    at jerryscript/jerry-core/vm/vm.cpp:556
# 13 0x000000000047e6f8 in vm_run_global () at jerryscript/jerry-core/vm/vm.cpp:395
# 14 0x000000000048582c in jerry_run () at jerryscript/jerry-core/jerry.cpp:1373
# 15 0x000000000048440b in main (argc=3, argv=0x7fffffffd878) at jerryscript/main-linux.cpp:291
```

galpeter, 2015-07-06:
The problem is that in the `ecma_object_get_class_name` only the RegExp prototype is registered, and the object is not. This should fix the error:
```diff
diff --git a/jerry-core/ecma/operations/ecma-objects.cpp b/jerry-core/ecma/operations/ecma-objects.cpp
index dcbe78b..863e30a 100644
--- a/jerry-core/ecma/operations/ecma-objects.cpp
+++ b/jerry-core/ecma/operations/ecma-objects.cpp
@@ -607,6 +607,13 @@ ecma_object_get_class_name (ecma_object_t *obj_p) /**< object */
             break;
           }
 #endif /* !CONFIG_ECMA_COMPACT_PROFILE_DISABLE_ERROR_BUILTINS */
+#ifndef CONFIG_ECMA_COMPACT_PROFILE_DISABLE_REGEXP_BUILTIN
+          case ECMA_BUILTIN_ID_REGEXP:
+          {
+            class_name = LIT_MAGIC_STRING_REGEXP_UL;
+            break;
+          }
+#endif /* CONFIG_ECMA_COMPACT_PROFILE_DISABLE_REGEXP_BUILTIN */
           default:
           {
             JERRY_ASSERT (builtin_id == ECMA_BUILTIN_ID_TYPE_ERROR_THROWER);
```

Also please note that the case `NaN == Date` throws a similar error, and the cause is the same.

LaszloLango, 2015-07-06:
@galpeter, thanks for the fix

LaszloLango, 2015-07-06:
@galpeter, sorry, I forgot to add it. I've updated the PR, please check.

closed 2015-07-09 by LaszloLango