jerryscript-project/jerryscript #286: Assertion 'jmp_op_meta.op.op_idx == OPCODE (is_true_jmp_down)' failed in rewrite_case_clause

renatahodovan, 2015-07-01:
###### Jerry version:
    Checked revision: 505bead
    Build: debug.linux
###### OS: 
    Ubuntu 14.10, x86_64
###### Test case:
```javascript
switch(foo){ 
    default:
        while(false)
            foo() {
                case a : {}
            }
}
```

###### Backtrace:
```javascript
ICE: Assertion 'jmp_op_meta.op.op_idx == OPCODE (is_true_jmp_down)' failed at jerryscript/jerry-core/parser/js/opcodes-dumper.cpp(rewrite_case_clause):2321.
Error: ERR_FAILED_INTERNAL_ASSERTION

Program received signal SIGABRT, Aborted.
0x0000000000485d22 in syscall_2_asm () at jerryscript/jerry-libc/target/linux/jerry-asm.S:32
32    SYSCALL_2
(gdb) bt
#0  0x0000000000485d22 in syscall_2_asm () at jerryscript/jerry-libc/target/linux/jerry-asm.S:32
#1  0x00000000004003c2 in syscall_2 (syscall_no=62, arg1=30900, arg2=6)
    at jerryscript/jerry-libc/target/linux/jerry-libc-target.c:96
#2  0x0000000000400563 in abort () at jerryscript/jerry-libc/target/linux/jerry-libc-target.c:175
#3  0x000000000043a7db in jerry_fatal (code=ERR_FAILED_INTERNAL_ASSERTION) at jerryscript/jerry-core/jrt/jrt-fatals.cpp:65
#4  0x000000000043a82a in jerry_assert_fail (assertion=0x4910f0 "jmp_op_meta.op.op_idx == OPCODE (is_true_jmp_down)", 
    file=0x491000 "jerryscript/jerry-core/parser/js/opcodes-dumper.cpp", 
    function=0x48d910 <_ZZ19rewrite_case_clausevE8__func__.lto_priv.224> "rewrite_case_clause", line=2321)
    at jerryscript/jerry-core/jrt/jrt-fatals.cpp:92
#5  0x00000000004128a2 in rewrite_case_clause() () at jerryscript/jerry-core/parser/js/opcodes-dumper.cpp:2321
#6  0x000000000043fe3c in parse_switch_statement () at jerryscript/jerry-core/parser/js/parser.cpp:2368
#7  0x000000000044095f in parse_statement (outermost_stmt_label_p=0x0) at jerryscript/jerry-core/parser/js/parser.cpp:2744
#8  0x0000000000440c18 in parse_source_element () at jerryscript/jerry-core/parser/js/parser.cpp:2820
#9  0x000000000044109a in parse_source_element_list(bool) [clone .lto_priv.559] (is_global=true)
    at jerryscript/jerry-core/parser/js/parser.cpp:3045
#10 0x000000000044123b in parser_parse_program (
    source_p=0x6b5780 <_ZL13source_buffer.lto_priv.1070> "switch(foo){ \n    default:\n        while(false)\n", ' ' <repeats 12 times>, "foo() {\n", ' ' <repeats 16 times>, "case a : {}\n", ' ' <repeats 12 times>, "}\n}", source_size=111, in_function=false, in_eval=false, is_strict=false, 
    out_opcodes_p=0x7fffffffd5d0) at jerryscript/jerry-core/parser/js/parser.cpp:3105
#11 0x00000000004413ba in parser_parse_script (
    source=0x6b5780 <_ZL13source_buffer.lto_priv.1070> "switch(foo){ \n    default:\n        while(false)\n", ' ' <repeats 12 times>, "foo() {\n", ' ' <repeats 16 times>, "case a : {}\n", ' ' <repeats 12 times>, "}\n}", source_size=111, opcodes_p=0x7fffffffd5d0)
    at jerryscript/jerry-core/parser/js/parser.cpp:3174
#12 0x00000000004857df in jerry_parse (
    source_p=0x6b5780 <_ZL13source_buffer.lto_priv.1070> "switch(foo){ \n    default:\n        while(false)\n", ' ' <repeats 12 times>, "foo() {\n", ' ' <repeats 16 times>, "case a : {}\n", ' ' <repeats 12 times>, "}\n}", source_size=111) at jerryscript/jerry-core/jerry.cpp:1341
#13 0x00000000004843ec in main (argc=3, argv=0x7fffffffd878) at jerryscript/main-linux.cpp:282
```

###### Note:
Removing the case branch from the test we get another assertion failure:
```javascript
switch(foo){ 
    default:
        while(false)
            foo() {}
}
```
###### Backtrace:
```javascript
ICE: Assertion 'token_is (TOK_EOF)' failed at jerryscript/jerry-core/parser/js/parser.cpp(parser_parse_program):3108.
Error: ERR_FAILED_INTERNAL_ASSERTION

Program received signal SIGABRT, Aborted.
0x0000000000485d22 in syscall_2_asm () at jerryscript/jerry-libc/target/linux/jerry-asm.S:32
32    SYSCALL_2
(gdb) bt
#0  0x0000000000485d22 in syscall_2_asm () at jerryscript/jerry-libc/target/linux/jerry-asm.S:32
#1  0x00000000004003c2 in syscall_2 (syscall_no=62, arg1=32025, arg2=6)
    at jerryscript/jerry-libc/target/linux/jerry-libc-target.c:96
#2  0x0000000000400563 in abort () at jerryscript/jerry-libc/target/linux/jerry-libc-target.c:175
#3  0x000000000043a7db in jerry_fatal (code=ERR_FAILED_INTERNAL_ASSERTION) at jerryscript/jerry-core/jrt/jrt-fatals.cpp:65
#4  0x000000000043a82a in jerry_assert_fail (assertion=0x498c48 "token_is (TOK_EOF)", 
    file=0x4989d8 "jerryscript/jerry-core/parser/js/parser.cpp", 
    function=0x48b7c0 <_ZZL20parser_parse_programPKhmbbbPPK8opcode_tE8__func__.lto_priv.509> "parser_parse_program", line=3108)
    at jerryscript/jerry-core/jrt/jrt-fatals.cpp:92
#5  0x0000000000441273 in parser_parse_program (
    source_p=0x6b5780 <_ZL13source_buffer.lto_priv.1070> "switch(foo){ \n    default:\n        while(false)\n", ' ' <repeats 12 times>, "foo() {}\n}", source_size=70, in_function=false, in_eval=false, is_strict=false, out_opcodes_p=0x7fffffffd5d0)
    at jerryscript/jerry-core/parser/js/parser.cpp:3108
#6  0x00000000004413ba in parser_parse_script (
    source=0x6b5780 <_ZL13source_buffer.lto_priv.1070> "switch(foo){ \n    default:\n        while(false)\n", ' ' <repeats 12 times>, "foo() {}\n}", 
    source_size=70, opcodes_p=0x7fffffffd5d0) at jerryscript/jerry-core/parser/js/parser.cpp:3174
#7  0x00000000004857df in jerry_parse (
    source_p=0x6b5780 <_ZL13source_buffer.lto_priv.1070> "switch(foo){ \n    default:\n        while(false)\n", ' ' <repeats 12 times>, "foo() {}\n}", source_size=70) at jerryscript/jerry-core/jerry.cpp:1341
#8  0x00000000004843ec in main (argc=3, argv=0x7fffffffd878) at jerryscript/main-linux.cpp:282
```

kkristof, 2015-07-30:
I think this is fixed because I get SyntaxError.

egavrin, 2015-11-19:
Works on master (f17f785b6ec0c71c2ef37e4e80a98aff1eff0963)

```
            foo() {
                  ^
ERROR: Ln 4, Col 19: Expected either ';' or newline token
```

closed 2015-11-19 by egavrin