jerryscript-project/jerryscript #287: Improvement of ecma/operations/* logic.

egavrin, 2015-07-01:
Let's continue the discussion from #273 here.

>> Btw the engine suffers a lot from these reversed order construction mechanics. Could we globally change that?
>
> @zherczeg, could you, please, provide more details on the issue?

zherczeg, 2015-07-02:
Normally, if you construct properties in other engines, they reported in creation order.

Firefox example:
var a = {}; a.x = 1; a.y = 2; a.z = 3; for (i in a) console.log(i);
"x"
"y"
"z"

I had problems with this when I was working on JSON (it has a strict visiting order). I manually reverse the list now. However, wouldn't it be better to simply follow other engines?

Basically we would need two pointers for each object: a head and a last pointer. Appending new property should happen using the last pointer.

ruben-ayrapetyan, 2015-07-02:
@zherczeg, thank you for the point.

'Last' pointer is not stored to reduce memory usage. However, if it leads to issues, maybe this should be revised.

Could you, please, provide an example with reference to corresponding specification paragraph for the case where order of properties is important?

egavrin, 2015-07-02:
@ruben-ayrapetyan it's not covered by spec.

@zherczeg I'd argue about `reported in creation order.`. As far as now, ES5.1 doesn't covers this behaviour - its implementation defined. But following other engines - strong point.

Input script:
```js
 var a = {}; a.x = 1; a[2]=10; a.y = 3; a.z = 2; for (i in a) console.log( i );
```
Output Chrome:
```
VM95:2 2
VM95:2 x
VM95:2 y
VM95:2 z
```
Output Firefox:
```
"2"
"x"
"y"
"z"
```
IE9:
```
var a = {}; a.x = 1; a[2]=10; a.y = 3; a.z = 2; for (i in a) console.log( i ); 
LOG: 2 
LOG: x 
LOG: y 
LOG: z 
```
I even tried Opera:
```
VM677:2 2
VM677:2 x
VM677:2 y
VM677:2 z
```

zherczeg, 2015-07-02:
http://stackoverflow.com/questions/5525795/does-javascript-guarantee-object-property-order

Here they say only ES6 defines the order. The same order above is probably the result of the ES6 support of browsers.

ECMAScript 6 seems to formalize in which order the properties should be iterated over:

    integer indexes(*) in ascending order
    strings in creation order
    Symbols in creation order

Anyway, it would be good to have an insert after or some similar method.

egavrin, 2015-07-03:
@zherczeg in this case it makes sense.

galpeter, 2015-07-03:
Also there is a test in the test262 suite which expect the order to be the same as the elements were added (https://github.com/tc39/test262/blob/es5-tests/test/suite/ch15/15.2/15.2.3/15.2.3.14/15.2.3.14-4-1.js).

Additionally there take the following js code:
```js
var obj = {a: 1, b: 2, c: 3};
var obj2 = {}

for(var p in obj) {
    obj2[p] = obj[p];
}

print(Object.keys(obj));
print(Object.keys(obj2));
```

Will print out:
```js
c,b,a
a,b,c
```

I know the order is not defined in the standard, but still it's awkward to see that if I iterate over an object and create a new object from it then the order changes for the new object.

closed 2015-11-19 by egavrin