jerryscript-project/jerryscript #295: Assertion 'token_start == NULL' failed in lexer_next_token_private

renatahodovan, 2015-07-02:
###### Jerry version:
    Checked revision: 017aa6b
    Build: debug.linux
###### OS: 
    Ubuntu 14.10, x86_64
###### Test case:
```javascript
function f_0() {
    try {
      Function("for(w in\\");
    } catch (e) {}
}

f_0();
f_0();
```

###### Backtrace:
```javascript
for(w in\
      ^
ERROR: Ln 1, Col 7: Malformed escape sequence
ICE: Assertion 'token_start == NULL' failed at jerryscript/jerry-core/parser/js/lexer.cpp(lexer_next_token_private):1100.
Error: ERR_FAILED_INTERNAL_ASSERTION

Program received signal SIGABRT, Aborted.
0x0000000000487139 in syscall_2_asm () at jerryscript/jerry-libc/target/linux/jerry-asm.S:32
32    SYSCALL_2
(gdb) bt
# 0  0x0000000000487139 in syscall_2_asm () at jerryscript/jerry-libc/target/linux/jerry-asm.S:32
# 1  0x00000000004003c2 in syscall_2 (syscall_no=62, arg1=17362, arg2=6) at jerryscript/jerry-libc/target/linux/jerry-libc-target.c:96
# 2  0x0000000000400563 in abort () at jerryscript/jerry-libc/target/linux/jerry-libc-target.c:175
# 3  0x000000000043c365 in jerry_fatal (code=ERR_FAILED_INTERNAL_ASSERTION) at jerryscript/jerry-core/jrt/jrt-fatals.cpp:65
# 4  0x000000000043c3b4 in jerry_assert_fail (assertion=0x4918da "token_start == NULL", 
    file=0x491448 "jerryscript/jerry-core/parser/js/lexer.cpp", 
    function=0x48efe0 <_ZZL24lexer_next_token_privatevE8__func__.lto_priv.164> "lexer_next_token_private", line=1100)
    at jerryscript/jerry-core/jrt/jrt-fatals.cpp:92
# 5  0x000000000040df8d in lexer_next_token_private () at jerryscript/jerry-core/parser/js/lexer.cpp:1100
# 6  0x000000000040f1f2 in lexer_next_token () at jerryscript/jerry-core/parser/js/lexer.cpp:1286
# 7  0x000000000044650a in skip_token() [clone .lto_priv.520] () at jerryscript/jerry-core/parser/js/parser.cpp:102
# 8  0x0000000000446749 in skip_newlines() [clone .lto_priv.516] () at jerryscript/jerry-core/parser/js/parser.cpp:135
# 9  0x000000000043b1c7 in parser_parse_program (source_p=0x7f69d0 <_ZL13mem_heap_area.lto_priv.1062+261712> "for(w in\\", source_size=9, in_function=true, 
    in_eval=false, is_strict=false, out_opcodes_p=0x7fffffffcf18) at jerryscript/jerry-core/parser/js/parser.cpp:3114
# 10 0x000000000043b476 in parser_parse_new_function (params=0x7f6a50 <_ZL13mem_heap_area.lto_priv.1062+261840>, 
    params_size=0x7f6a10 <_ZL13mem_heap_area.lto_priv.1062+261776>, params_count=1, out_opcodes_p=0x7fffffffcf18)
    at jerryscript/jerry-core/parser/js/parser.cpp:3234
# 11 0x0000000000469643 in ecma_builtin_function_dispatch_construct (arguments_list_p=0x7f6ad0 <_ZL13mem_heap_area.lto_priv.1062+261968>, arguments_list_len=1)
    at jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-function.cpp:144
# 12 0x00000000004692a1 in ecma_builtin_function_dispatch_call (arguments_list_p=0x7f6ad0 <_ZL13mem_heap_area.lto_priv.1062+261968>, arguments_list_len=1)
    at jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-function.cpp:55
# 13 0x00000000004784aa in ecma_builtin_dispatch_call (obj_p=0x7b6dd8 <_ZL13mem_heap_area.lto_priv.1062+600>, this_arg_value=4, 
    arguments_list_p=0x7f6ad0 <_ZL13mem_heap_area.lto_priv.1062+261968>, arguments_list_len=1)
    at jerryscript/jerry-core/ecma/builtin-objects/ecma-builtins.inc.h:116
# 14 0x0000000000442c10 in ecma_op_function_call(ecma_object_t*, unsigned int, unsigned int const*, unsigned int) (
    func_obj_p=0x7b6dd8 <_ZL13mem_heap_area.lto_priv.1062+600>, this_arg_value=4, arguments_list_p=0x7f6ad0 <_ZL13mem_heap_area.lto_priv.1062+261968>, 
    arguments_list_len=1) at jerryscript/jerry-core/ecma/operations/ecma-function-object.cpp:631
# 15 0x00000000004389b2 in opfunc_call_n (opdata=..., int_data=0x7fffffffd2b0) at jerryscript/jerry-core/vm/opcodes.cpp:817
# 16 0x000000000047e546 in vm_loop (int_data_p=0x7fffffffd2b0, run_scope_p=0x7fffffffd200) at jerryscript/jerry-core/vm/vm.cpp:466
# 17 0x000000000048080b in opfunc_try_block (opdata=..., int_data=0x7fffffffd2b0)
    at jerryscript/jerry-core/vm/opcodes-ecma-try-catch-finally.cpp:41
# 18 0x000000000047e546 in vm_loop (int_data_p=0x7fffffffd2b0, run_scope_p=0x0) at jerryscript/jerry-core/vm/vm.cpp:466
# 19 0x000000000047e7d3 in vm_run_from_pos (opcodes_p=0x7b6d18 <_ZL13mem_heap_area.lto_priv.1062+408>, start_pos=5, this_binding_value=15, 
    lex_env_p=0x7b6ca0 <_ZL13mem_heap_area.lto_priv.1062+288>, is_strict=false, is_eval_code=false)
    at jerryscript/jerry-core/vm/vm.cpp:556
# 20 0x0000000000442e6a in ecma_op_function_call(ecma_object_t*, unsigned int, unsigned int const*, unsigned int) (
    func_obj_p=0x7b6c28 <_ZL13mem_heap_area.lto_priv.1062+168>, this_arg_value=4, arguments_list_p=0x0, arguments_list_len=0)
    at jerryscript/jerry-core/ecma/operations/ecma-function-object.cpp:692
# 21 0x00000000004389b2 in opfunc_call_n (opdata=..., int_data=0x7fffffffd500) at jerryscript/jerry-core/vm/opcodes.cpp:817
# 22 0x000000000047e546 in vm_loop (int_data_p=0x7fffffffd500, run_scope_p=0x0) at jerryscript/jerry-core/vm/vm.cpp:466
# 23 0x000000000047e7d3 in vm_run_from_pos (opcodes_p=0x7b6d18 <_ZL13mem_heap_area.lto_priv.1062+408>, start_pos=1, this_binding_value=15, 
    lex_env_p=0x7b6ba8 <_ZL13mem_heap_area.lto_priv.1062+40>, is_strict=false, is_eval_code=false)
    at jerryscript/jerry-core/vm/vm.cpp:556
# 24 0x000000000048508b in vm_run_global() () at jerryscript/jerry-core/vm/vm.cpp:395
# 25 0x0000000000486c43 in jerry_run () at jerryscript/jerry-core/jerry.cpp:1373
# 26 0x0000000000485822 in main (argc=3, argv=0x7fffffffd8c8) at jerryscript/main-linux.cpp:291
```

egavrin, 2015-11-19:
Works on master (f17f785b6ec0c71c2ef37e4e80a98aff1eff0963)

Revealed other issue:
```
(function(){for(w in\})
                      ^
ERROR: Ln 1, Col 23: Illegal identifier name
(function(){for(w in\})
                      ^
ERROR: Ln 1, Col 23: Illegal identifier name
```

egavrin, 2015-11-19:
Follow up issue: #724

closed 2015-11-19 by egavrin