jerryscript-project/jerryscript #313: Prerequisite script fails.

LaszloLango, 2015-07-06:
Prerequisite script fails with the following message:
```
Cloning into './tmp.viAM17He42/nuttx'...
fatal: reference is not a tree: 36a655eddec29754cc93631b6083fe6409817861
Failed to setup 'nuttx' prerequisite. Failed to prepare nuttx headers
```

seanshpark, 2015-07-08:
@egavrin , can you please add some explanation about removing nuttx support? 
for me, it's ok. I think removing can reduce efforts to maintain, and support for various platforms can be added to wiki guide pages.

egavrin, 2015-07-08:
@seanshpark the main reason to add nuttx was to check integration, not to add functionality. Same result can be achieved with the current infrastructure. For me, build-only testing is useless since it doesn't reveals any issues at run-time, like #296. #304. #231. etc. We should setup proper integration testing with IoTjs.

PS Regarding guides, you're absolutely right - current guides does not provide enough information for doing such things.

seanshpark, 2015-07-08:
Thank you for detailed explanation!

egavrin, 2015-07-08:
Fixed at ddc3f0d6e85e8e1257f3d16b577b579e48c7233b

closed 2015-07-08 by egavrin