jerryscript-project/jerryscript #323: Implement toLowerCase toUpperCase

zherczeg, 2015-07-07:
String manipulation functions and regular expressions require lowercase/uppercase conversion. See

http://www.ecma-international.org/ecma-262/5.1/#sec-15.5.4.16

To implement them, we need some utilities in the core.

1) We need a variant, which converts a character to its uppercase described in (starting from Canonicalize part)
http://www.ecma-international.org/ecma-262/5.1/#sec-15.10.2.8

2) We also need variants, which converts characters to 1 or more upper/lowercase characters. E.g. the uppercase form of german sharp s is SS. This is needed for String.prototype.toUpperCase String.prototype.toLowerCase.

3) Fortunately LocaleCompare is platform dependent.

Could you raise the priority of implementing these functions in the core?

sand1k, 2015-07-08:
@zherczeg, these functions are not in our nearest aims, as we are concentrating on fixing test262 and  improving the parser. We could implement them in three weeks.
Maybe you can implement them roughly for now, supporting ASCII only for example, and we'll fix them later?

zherczeg, 2015-07-08:
Ok, I will sketch the appropriate functions and you can implement them later.

sand1k, 2015-07-08:
@zherczeg, thank you.

LaszloLango, 2016-07-14:
@zherczeg, is this still valid?

zherczeg, 2016-07-14:
Thanks for reminding me to this. This is very much valid.

robertsipka, 2017-02-10:
#1575

PR #1575 (2017-02-15): Improve `toLowerCase` and `toUpperCase` functions.
Language-sensitive mappings are not processed now.
Binary size: 136160	 -> 137684
"Fixes #323"
commit 1b5f839d (2017-02-15): Improve `toLowerCase` and `toUpperCase` functions. (#1575)

Language-sensitive mappings are not processed now.

Fixes #323

JerryScript-DCO-1.0-Signed-off-by: Robert Sipka rsipka.uszeged@partner.samsung.com
closed 2017-02-15 by LaszloLango