jerryscript-project/jerryscript #332: Lowering required CMake version

ILyoan, 2015-07-08:
JerryScript requires CMake 2.8.12.
```
cmake_minimum_required (VERSION 2.8.12)
```

However, It seems like some version of ubuntu only support CMake 2.8.7 with 'apt-get'  that developers running ubuntu 12.04 should manually download and build CMake 2.8.12. 

Is there any problem lowering minimum required version of CMake to 2.8? If there are no problems why don't we lower the required version.

galpeter, 2015-07-08:
On ubuntu 12.04 which has CMake 2.8.7 the gcc is also old. It's only 4.6 which can not compile the engine (AFAIK). So on that systems the user/developer should also install a new gcc. I think this is a bit more general question on what is the minimum required version of tools for which we support the build.

I'm not against modifying the required version for the cmake, I'm just concerned that on such systems this step is not enough to build the engine.

egavrin, 2015-07-08:
I think latest Ubuntu LTS should be fine, so we may just make it default for developers.

ILyoan, 2015-07-09:
I found that we could not build jerry without changing CMake script since we use `target_include_directories`, `target_compile_difinitions` and maybe more which are added from CMake 2.8.11. That my conclusion is that ubuntu 12.04 is not sufficient to build Jerry because of CMake version problem as well as gcc and probably other infrastructure.

Closing this issue for won't fix.

closed 2015-07-09 by ILyoan