jerryscript-project/jerryscript #341: Segmentation fault in longjmp

renatahodovan, 2015-07-08:
###### Jerry version:
    Checked revision: 146ac15
    Build: debug.linux
###### OS: 
    Ubuntu 15.04, x86_64
###### Test case:
```javascript
�
```

Hex: BF 0A 0A

###### Backtrace:
```javascript

^
ERROR: Ln 1, Col 1: Invalid source encoding

Program received signal SIGSEGV, Segmentation fault.
0x0000000000497a84 in longjmp () at jerryscript/jerry-libc/target/linux/jerry-asm.S:69
69	  _LONGJMP
(gdb) bt
# 0  0x0000000000497a84 in longjmp () at jerryscript/jerry-libc/target/linux/jerry-asm.S:69
Backtrace stopped: Cannot access memory at address 0x0

```

ruben-ayrapetyan, 2015-07-14:
The failure is caused by invocation of `syntax_raise_error` before `setjmp` was performed in `parser_parse_program`.

ruben-ayrapetyan, 2015-07-14:
Pull request with fix: #390

ruben-ayrapetyan, 2015-07-14:
Fixed.

closed 2015-07-14 by ruben-ayrapetyan