jerryscript-project/jerryscript #342: Heap size limit for embed system

seanshpark, 2015-07-08:
Nuttx heap memory size limit is now gone for #313, commit 8a815fe35742fb45070ccf461998f137aa9b3e9e
it would be nice to set CONFIG_MEM_HEAP_AREA_SIZE  from command line.
I couldn't find an issue about it so I'm adding this one.

seanshpark, 2015-07-09:
adding  `-DCONFIG_MEM_HEAP_AREA_SIZE=(some number)` to compile flags limits the heap size.
@egavrin , do you think it's ok to use like this or add another variable and let that variable set  `CONFIG_MEM_HEAP_AREA_SIZE` value?

ruben-ayrapetyan, 2015-07-09:
@seanshpark, could you, please, check the changes?

The patch could be applied above changes from https://github.com/Samsung/jerryscript/issues/344#issuecomment-120143978
```
diff --git a/CMakeLists.txt b/CMakeLists.txt
index 83f4060..1748949 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -83,6 +83,7 @@ project (Jerry CXX C ASM)
 
   set(EXTERNAL_LIBC_INTERFACE "UNDEFINED" CACHE STRING "Path to external libc include directory")
   set(EXTERNAL_LIBS_INTERFACE "UNDEFINED" CACHE STRING "Path to external libraries' include directory")
+  set(EXTERNAL_MEM_HEAP_SIZE_KB "256" CACHE STRING "Size of memory heap, in kilobytes")
  else()
   message(FATAL_ERROR "Platform '${PLATFORM}' is not supported")
  endif()
diff --git a/jerry-core/CMakeLists.txt b/jerry-core/CMakeLists.txt
index 0e79726..62f0639 100644
--- a/jerry-core/CMakeLists.txt
+++ b/jerry-core/CMakeLists.txt
@@ -88,6 +88,10 @@ project (JerryCore CXX C ASM)
     math(EXPR MEM_HEAP_AREA_SIZE_16K "16 * 1024")
     set(DEFINES_JERRY_MCU_STM32F4 CONFIG_MEM_HEAP_AREA_SIZE=${MEM_HEAP_AREA_SIZE_16K})
 
+  # External
+   math(EXPR MEM_HEAP_AREA_SIZE_EXTERNAL "${EXTERNAL_MEM_HEAP_SIZE_KB} * 1024")
+   set(DEFINES_JERRY_EXTERNAL CONFIG_MEM_HEAP_AREA_SIZE=${MEM_HEAP_AREA_SIZE_EXTERNAL})
+
 # Include directories
  set(INCLUDE_CORE
      ${CMAKE_SOURCE_DIR}/jerry-core
diff --git a/jerry-core/config.h b/jerry-core/config.h
index 2495749..b282050 100644
--- a/jerry-core/config.h
+++ b/jerry-core/config.h
@@ -53,6 +53,8 @@
  */
 #ifndef CONFIG_MEM_HEAP_AREA_SIZE
 # define CONFIG_MEM_HEAP_AREA_SIZE (256 * 1024)
+#elif CONFIG_MEM_HEAP_AREA_SIZE > (256 * 1024)
+# error "Currently, maximum 256 kilobytes heap size is supported"
 #endif /* !CONFIG_MEM_HEAP_AREA_SIZE */
 
 /**
```

seanshpark, 2015-07-09:
@ruben-ayrapetyan , I couldn't have done it better. I works fine.
I'll update PR #345 with this as separate commit commits with your sign also.

ruben-ayrapetyan, 2015-07-10:
@seanshpark, great! Thank you for checking.

seanshpark, 2015-07-14:
close as PR landed

closed 2015-07-14 by seanshpark