jerryscript-project/jerryscript #344: iotjs build break for nuttx

seanshpark, 2015-07-09:
iotjs build is using nuttx parts in cmake scripts and as #330 landed iotjs nuttx build fails by the changes.
I'm trying to fix the scripts with using `PLATFORM` as `EXTERNAL` and not getting anywhere.
@egavrin , can you please quick fix this?
there are items that needs some fix
* using `CMAKE_SYSTEM_VERSION` as platform name
* complicate mode and hard tracking built target

seanshpark, 2015-07-09:
this is what I've tried.
```
diff --git a/CMakeLists.txt b/CMakeLists.txt
index 9c139f3..5bd8c36 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -78,13 +78,13 @@ project (Jerry CXX C ASM)
   option(STRIP_RELEASE_BINARY "Strip symbols from release binaries" OFF)
   set(MCU_SCRIPT_FILE "tests/blinky.js" CACHE STRING "Script to run on MCU")
  elseif("${PLATFORM}" STREQUAL "EXTERNAL")
-  set(PLATFORM_EXT "${CMAKE_SYSTEM_VERSION}")
-  set(EXTERNAL_BUILD TRUE)
+  if (NOT DEFINED PLATFORM_EXT)
+    message(FATAL_ERROR "Please define 'PLATFORM_EXT' for name of this platform")
+  endif()
 
-  set(EXTERNAL_LIBC_INTERFACE "UNDEFINED" CACHE STRING "Path to external libc include directory")
-  set(EXTERNAL_LIBS_INTERFACE "UNDEFINED" CACHE STRING "Path to external libraries' include directory")
+  set(EXTERNAL_BUILD TRUE)
  else()
-  message(FATAL_ERROR "Platform is not supported")
+  message(FATAL_ERROR "Platform '${PLATFORM}' is not supported")
  endif()
 
 # Intermediate files
@@ -182,6 +182,14 @@ project (Jerry CXX C ASM)
       "COMPACT_PROFILE"
       "COMPACT_PROFILE_MINIMAL")
 
+  # EXTERNAL
+   if("${PLATFORM}" STREQUAL "EXTERNAL")
+    set(MODIFIERS_LISTS_${PLATFORM_EXT}
+      "FULL_PROFILE"
+      "COMPACT_PROFILE"
+      "COMPACT_PROFILE_MINIMAL")
+   endif()
+
 # Compiler / Linker flags
  set(COMPILE_FLAGS_JERRY "-fno-builtin")
  set(LINKER_FLAGS_COMMON "-Wl,-z,noexecstack")
```

error is something like this
```
You have called ADD_LIBRARY for library debug.nuttx.nuttx.main without any source files. This typically indicates a problem with your CMakeLists.txt file
...
CMake Error: CMake can not determine linker language for target: debug.nuttx.nuttx.main
CMake Error: Cannot determine link language for target "release.nuttx-cp.nuttx.main".
```

seanshpark, 2015-07-09:
found fix for error and added RR #345

ruben-ayrapetyan, 2015-07-09:
@seanshpark, I've tried to reproduce the issue with nuttx build, using jerry revision 8d7cdebee09d9fcccb85b47efa4e84341f065607 and iotjs revision ed5e309c6c833dea924a8699de86d94a39cf748e, but build was successful and the issue didn't appear.

Could you, please, check the changes?
The changes are based on pull request #345; additionally, they remove main module target for external build, as it is supposed that external targets provide their own main module, and set `PLATFORM_EXT` to `EXTERNAL` for external targets.
```
diff --git a/CMakeLists.txt b/CMakeLists.txt
index 9c139f3..83f4060 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -78,13 +78,13 @@ project (Jerry CXX C ASM)
   option(STRIP_RELEASE_BINARY "Strip symbols from release binaries" OFF)
   set(MCU_SCRIPT_FILE "tests/blinky.js" CACHE STRING "Script to run on MCU")
  elseif("${PLATFORM}" STREQUAL "EXTERNAL")
-  set(PLATFORM_EXT "${CMAKE_SYSTEM_VERSION}")
+  set(PLATFORM_EXT "EXTERNAL")
   set(EXTERNAL_BUILD TRUE)
 
   set(EXTERNAL_LIBC_INTERFACE "UNDEFINED" CACHE STRING "Path to external libc include directory")
   set(EXTERNAL_LIBS_INTERFACE "UNDEFINED" CACHE STRING "Path to external libraries' include directory")
  else()
-  message(FATAL_ERROR "Platform is not supported")
+  message(FATAL_ERROR "Platform '${PLATFORM}' is not supported")
  endif()
 
 # Intermediate files
@@ -182,6 +182,14 @@ project (Jerry CXX C ASM)
       "COMPACT_PROFILE"
       "COMPACT_PROFILE_MINIMAL")
 
+ # External
+   set(MODIFIERS_LISTS_EXTERNAL
+     "FULL_PROFILE"
+     "COMPACT_PROFILE"
+     "COMPACT_PROFILE_MINIMAL"
+     "FULL_PROFILE MEMORY_STATISTICS"
+     "COMPACT_PROFILE MEMORY_STATISTICS")
+
 # Compiler / Linker flags
  set(COMPILE_FLAGS_JERRY "-fno-builtin")
  set(LINKER_FLAGS_COMMON "-Wl,-z,noexecstack")
@@ -366,28 +374,20 @@ project (Jerry CXX C ASM)
      endif()
     endif()
    else()
-    set(MAIN_MODULE_TARGET_NAME ${TARGET_NAME}.${PLATFORM_L}.main)
-    add_library(${MAIN_MODULE_TARGET_NAME} ${SOURCE_JERRY_STANDALONE_MAIN})
-    set_property(TARGET ${MAIN_MODULE_TARGET_NAME}
-                 PROPERTY COMPILE_FLAGS "${COMPILE_FLAGS_JERRY} ${CXX_FLAGS_JERRY} ${FLAGS_COMMON_${BUILD_MODE}}")
-    target_compile_definitions(${MAIN_MODULE_TARGET_NAME} PRIVATE ${DEFINES_JERRY})
-    target_include_directories(${MAIN_MODULE_TARGET_NAME} PRIVATE ${INCLUDE_CORE_INTERFACE})
-    target_include_directories(${MAIN_MODULE_TARGET_NAME} SYSTEM PRIVATE ${INCLUDE_LIBC_INTERFACE})
-    target_include_directories(${MAIN_MODULE_TARGET_NAME} SYSTEM PRIVATE ${INCLUDE_EXTERNAL_LIBS_INTERFACE})
-
     add_custom_target(${TARGET_NAME} ALL)
-    add_dependencies(${TARGET_NAME} ${MAIN_MODULE_TARGET_NAME} ${FDLIBM_TARGET_NAME}${CORE_TARGET_NAME})
 
+    add_dependencies(${TARGET_NAME} ${FDLIBM_TARGET_NAME} ${CORE_TARGET_NAME})
     add_custom_command(TARGET ${TARGET_NAME}
                        POST_BUILD
                        COMMAND mkdir -p ${CMAKE_BINARY_DIR}/${TARGET_NAME}
-                       COMMAND echo
-                       $<TARGET_FILE:${MAIN_MODULE_TARGET_NAME}>
-                       $<TARGET_FILE:${FDLIBM_TARGET_NAME}>
-                       $<TARGET_FILE:${CORE_TARGET_NAME}> > ${CMAKE_BINARY_DIR}/${TARGET_NAME}/list)
+                       COMMAND echo $<TARGET_FILE:${FDLIBM_TARGET_NAME}> > ${CMAKE_BINARY_DIR}/${TARGET_NAME}/list
+                       COMMAND echo $<TARGET_FILE:${CORE_TARGET_NAME}> >> ${CMAKE_BINARY_DIR}/${TARGET_NAME}/list)
 
     if(${USE_JERRY_LIBC})
      add_dependencies(${TARGET_NAME} ${LIBC_TARGET_NAME})
+     add_custom_command(TARGET ${TARGET_NAME}
+                        POST_BUILD
+                        COMMAND echo $<TARGET_FILE:${LIBC_TARGET_NAME}> >> ${CMAKE_BINARY_DIR}/${TARGET_NAME}/list)
     endif()
 
     add_cppcheck_target(${TARGET_NAME})
diff --git a/build/configs/toolchain_external.cmake b/build/configs/toolchain_external.cmake
index 5ad75cc..97b4df2 100644
--- a/build/configs/toolchain_external.cmake
+++ b/build/configs/toolchain_external.cmake
@@ -15,7 +15,7 @@
 include(CMakeForceCompiler)
 
 set(CMAKE_SYSTEM_NAME EXTERNAL)
-set(CMAKE_SYSTEM_VERSION ${EXTERNAL_TARGET_NAME})
+set(CMAKE_SYSTEM_PROCESSOR "${EXTERNAL_CMAKE_SYSTEM_PROCESSOR}")
 
 CMAKE_FORCE_C_COMPILER(${EXTERNAL_CMAKE_C_COMPILER} GNU)
 CMAKE_FORCE_CXX_COMPILER(${EXTERNAL_CMAKE_CXX_COMPILER} GNU)
```

seanshpark, 2015-07-09:
@ruben-ayrapetyan , works just great! thank you. I'll update PR #345 with your sign also.

ruben-ayrapetyan, 2015-07-10:
@seanshpark, great! Thank you too for checking.

seanshpark, 2015-07-14:
close as #345 landed

closed 2015-07-14 by seanshpark