jerryscript-project/jerryscript #354: Infinite loop on big objects

galpeter, 2015-07-09:
If a js source contains an object with more than 255 key then the engine goes into infinite loop in the `syntax_check_for_duplication_of_prop_names` function (`jerry-core/parser/js/syntax-errors.cpp`).

Testcase:
```js
var data = {
'1': '', '2': '', '3': '', '4': '', '5': '', '6': '', '7': '', '8': '',
'9': '', '10': '', '11': '', '12': '', '13': '', '14': '', '15': '', '16': '',
'17': '', '18': '', '19': '', '20': '', '21': '', '22': '', '23': '', '24': '',
'25': '', '26': '', '27': '', '28': '', '29': '', '30': '', '31': '', '32': '',
'33': '', '34': '', '35': '', '36': '', '37': '', '38': '', '39': '', '40': '',
'41': '', '42': '', '43': '', '44': '', '45': '', '46': '', '47': '', '48': '',
'49': '', '50': '', '51': '', '52': '', '53': '', '54': '', '55': '', '56': '',
'57': '', '58': '', '59': '', '60': '', '61': '', '62': '', '63': '', '64': '',
'65': '', '66': '', '67': '', '68': '', '69': '', '70': '', '71': '', '72': '',
'73': '', '74': '', '75': '', '76': '', '77': '', '78': '', '79': '', '80': '',
'81': '', '82': '', '83': '', '84': '', '85': '', '86': '', '87': '', '88': '',
'89': '', '90': '', '91': '', '92': '', '93': '', '94': '', '95': '', '96': '',
'97': '', '98': '', '99': '', '100': '', '101': '', '102': '', '103': '', '104': '',
'105': '', '106': '', '107': '', '108': '', '109': '', '110': '', '111': '', '112': '',
'113': '', '114': '', '115': '', '116': '', '117': '', '118': '', '119': '', '120': '',
'121': '', '122': '', '123': '', '124': '', '125': '', '126': '', '127': '', '128': '',
'129': '', '130': '', '131': '', '132': '', '133': '', '134': '', '135': '', '136': '',
'137': '', '138': '', '139': '', '140': '', '141': '', '142': '', '143': '', '144': '',
'145': '', '146': '', '147': '', '148': '', '149': '', '150': '', '151': '', '152': '',
'153': '', '154': '', '155': '', '156': '', '157': '', '158': '', '159': '', '160': '',
'161': '', '162': '', '163': '', '164': '', '165': '', '166': '', '167': '', '168': '',
'169': '', '170': '', '171': '', '172': '', '173': '', '174': '', '175': '', '176': '',
'177': '', '178': '', '179': '', '180': '', '181': '', '182': '', '183': '', '184': '',
'185': '', '186': '', '187': '', '188': '', '189': '', '190': '', '191': '', '192': '',
'193': '', '194': '', '195': '', '196': '', '197': '', '198': '', '199': '', '200': '',
'201': '', '202': '', '203': '', '204': '', '205': '', '206': '', '207': '', '208': '',
'209': '', '210': '', '211': '', '212': '', '213': '', '214': '', '215': '', '216': '',
'217': '', '218': '', '219': '', '220': '', '221': '', '222': '', '223': '', '224': '',
'225': '', '226': '', '227': '', '228': '', '229': '', '230': '', '231': '', '232': '',
'233': '', '234': '', '235': '', '236': '', '237': '', '238': '', '239': '', '240': '',
'241': '', '242': '', '243': '', '244': '', '245': '', '246': '', '247': '', '248': '',
'249': '', '250': '', '251': '', '252': '', '253': '', '254': '', '255': '',
'256': '',
};
```

ruben-ayrapetyan, 2015-07-09:
// this can be partial solution for large objects https://github.com/Samsung/jerryscript/tree/Ruben-fix-syntax-check-for-large-object-literals

galpeter, 2015-07-09:
What I found out is that the infinite loop is caused by size of the stack used in the `syntax-errors.cpp`. There the stack is used with `uint8_t` type. I've tried to change it to `unit16_t` to see what happens, and then the infinite loop disappeared, and an assert appeared:
```
ICE: Assertion 'ecma_is_completion_value_throw (completion) || ecma_is_completion_value_return (completion)' failed at /mnt/work/projects/jerryscript_main_repo/jerry-core/vm/vm.cpp(vm_run_from_pos):559.
Error: ERR_FAILED_INTERNAL_ASSERTION
```

ruben-ayrapetyan, 2015-07-09:
Yes, because of this the solution in the branch is only partial.

ruben-ayrapetyan, 2015-07-14:
Pull request with fix of `syntax_check_for_duplication_of_prop_names`.

LaszloLango, 2016-02-08:
Works on current master (91a0514).

closed 2016-02-08 by LaszloLango