jerryscript-project/jerryscript #355: unittests/unit-test-date-helpers  fail

seanshpark, 2015-07-10:
When I do pre-commit test I get
```
make[1]: Entering directory `/.../harmony/jerryscript'
Running ./build/bin/unittests/unit-test-api... OK
Running ./build/bin/unittests/unit-test-date-helpers... FAILED
Unit tests run failed. See ./build/bin/unittests/unit_tests_run.log for details.
make[1]: *** [unittests_run] Error 1
```
`./build/bin/unittests/unit_tests_run.log` says
```
ICE: Assertion 'ecma_date_day (MS_PER_DAY) == 1' failed at /.../harmony/jerryscript/tests/unit/test-date-helpers.cpp(main):32.
Error: ERR_FAILED_INTERNAL_ASSERTION
```

is this a problem for only me or a bug? or duplicate issue?

egavrin, 2015-07-10:
Could you please share your environment?

On dbf3708255217f360a72c02fadb185a76e5fbf67 :
```bash
$ make unittests_run 
Running ./build/bin/unittests/unit-test-api... OK
Running ./build/bin/unittests/unit-test-date-helpers... OK
Running ./build/bin/unittests/unit-test-heap... OK
Running ./build/bin/unittests/unit-test-literal-storage... OK
Running ./build/bin/unittests/unit-test-longjmp... OK
Running ./build/bin/unittests/unit-test-number-to-integer... OK
Running ./build/bin/unittests/unit-test-number-to-string... OK
Running ./build/bin/unittests/unit-test-parser... OK
Running ./build/bin/unittests/unit-test-pool... OK
Running ./build/bin/unittests/unit-test-poolman... OK
Running ./build/bin/unittests/unit-test-recordset... OK
Running ./build/bin/unittests/unit-test-strings... OK
Running ./build/bin/unittests/unit-test-string-to-number... OK
```

```bash
$ make unittests_run TOOLCHAIN="./build/configs/toolchain_linux_armv7l-el.cmake"
Running ./build/bin/unittests/unit-test-api... OK
Running ./build/bin/unittests/unit-test-date-helpers... OK
Running ./build/bin/unittests/unit-test-heap... OK
Running ./build/bin/unittests/unit-test-literal-storage... OK
Running ./build/bin/unittests/unit-test-longjmp... OK
Running ./build/bin/unittests/unit-test-number-to-integer... OK
Running ./build/bin/unittests/unit-test-number-to-string... OK
Running ./build/bin/unittests/unit-test-parser... OK
Running ./build/bin/unittests/unit-test-pool... OK
Running ./build/bin/unittests/unit-test-poolman... OK
Running ./build/bin/unittests/unit-test-recordset... OK
Running ./build/bin/unittests/unit-test-strings... OK
Running ./build/bin/unittests/unit-test-string-to-number... OK
```

seanshpark, 2015-07-12:
I think it's related with 32bit.  I use Ubuntu 14.04 32bit for main development.
64bit Ubuntu 14.04 test passes successfully.
Source 13cf31413f6a3a1683c62ac4ed3f19d98d9b2cd2

galpeter, 2015-07-24:
Fix landed: https://github.com/Samsung/jerryscript/commit/021fc626526a141a4deae4f2e60669546209cf2c

closed 2015-07-24 by galpeter