jerryscript-project/jerryscript #356: Allow unset values when creating arrays

galpeter, 2015-07-10:
In js it is possible to create arrays with unset values.
Example:
```js
var data = [0,,,'3'];
```
In this case the `data` array's `1` and `2` properties should not exist.
```js
data.hasOwnProperty(1) == false
data.hasOwnProperty(2) == false
```

Currently both the 1 and 2 is set to undefined which is incorrect.

dbatyai, 2015-07-22:
The following test262 tests fail because of this issue:
```
15.4.4.10/S15.4.4.10_A3_T3
15.4.4.14/15.4.4.14-9-a-10
15.4.4.14/15.4.4.14-9-b-1
15.4.4.14/15.4.4.14-9-b-i-15
15.4.4.14/15.4.4.14-9-b-i-7
15.4.4.15/15.4.4.15-8-a-10
15.4.4.15/15.4.4.15-8-b-1
15.4.4.15/15.4.4.15-8-b-i-15
15.4.4.15/15.4.4.15-8-b-i-7
15.4.4.16/15.4.4.16-7-b-11
15.4.4.16/15.4.4.16-7-b-7
15.4.4.16/15.4.4.16-7-c-i-16
15.4.4.16/15.4.4.16-7-c-i-8
15.4.4.17/15.4.4.17-7-b-11
15.4.4.17/15.4.4.17-7-b-7
15.4.4.17/15.4.4.17-7-c-i-16
15.4.4.17/15.4.4.17-7-c-i-8
15.4.4.18/15.4.4.18-7-b-11
15.4.4.18/15.4.4.18-7-b-7
15.4.4.18/15.4.4.18-7-c-i-16
15.4.4.18/15.4.4.18-7-c-i-8
15.4.4.19/15.4.4.19-8-b-11
15.4.4.19/15.4.4.19-8-b-7
15.4.4.19/15.4.4.19-8-c-i-16
15.4.4.19/15.4.4.19-8-c-i-8
15.4.4.20/15.4.4.20-9-b-11
15.4.4.20/15.4.4.20-9-b-7
15.4.4.20/15.4.4.20-9-c-i-16
15.4.4.20/15.4.4.20-9-c-i-8
15.4.4.21/15.4.4.21-8-b-iii-1-16
15.4.4.21/15.4.4.21-8-b-iii-1-8
15.4.4.21/15.4.4.21-9-b-11
15.4.4.21/15.4.4.21-9-b-20
15.4.4.21/15.4.4.21-9-b-24
15.4.4.21/15.4.4.21-9-b-7
15.4.4.21/15.4.4.21-9-c-i-16
15.4.4.21/15.4.4.21-9-c-i-8
15.4.4.22/15.4.4.22-8-b-iii-1-16
15.4.4.22/15.4.4.22-8-b-iii-1-8
15.4.4.22/15.4.4.22-9-b-11
15.4.4.22/15.4.4.22-9-b-20
15.4.4.22/15.4.4.22-9-b-24
15.4.4.22/15.4.4.22-9-b-7
15.4.4.22/15.4.4.22-9-c-i-16
15.4.4.22/15.4.4.22-9-c-i-8
```

zherczeg, 2015-08-13:
I think we should do something with this, since ~50 fails is much.

ruben-ayrapetyan, 2015-08-26:
// TODO:
//  1. introduce empty 'varg' opcode arguments
//  2. support them in `array_decl` opcode handler,
//      i.e. change the opcode handler to perform calls
//      to ECMA, according to ECMA-262 v5, 11.1.4

egavrin, 2015-11-19:
Works on master (f17f785b6ec0c71c2ef37e4e80a98aff1eff0963)

closed 2015-11-19 by egavrin