jerryscript-project/jerryscript #358: Assertion 'label_p->continue_tgt_oc != MAX_OPCODES' failed in jsp_label_rewrite_jumps_and_pop

renatahodovan, 2015-07-10:
###### Jerry version:
    Checked revision: 9b24d8f
    Build: debug.linux
###### OS: 
    Ubuntu 15.04, x86_64
###### Test case:
```javascript
label_0: 
    if (false)
        do  {
            continue label_0;
        } while (false);
```

###### Backtrace:
```javascript
ICE: Assertion 'label_p->continue_tgt_oc != MAX_OPCODES' failed at jerryscript/jerry-core/parser/js/jsp-label.cpp(jsp_label_rewrite_jumps_and_pop):125.
Error: ERR_FAILED_INTERNAL_ASSERTION

Program received signal SIGABRT, Aborted.
0x0000000000499ae6 in syscall_2_asm () at jerryscript/jerry-libc/target/linux/jerry-asm.S:32
32    SYSCALL_2
(gdb) bt
# 0  0x0000000000499ae6 in syscall_2_asm () at jerryscript/jerry-libc/target/linux/jerry-asm.S:32
# 1  0x00000000004003c2 in syscall_2 (syscall_no=62, arg1=9091, arg2=6) at jerryscript/jerry-libc/target/linux/jerry-libc-target.c:96
# 2  0x0000000000400563 in abort () at jerryscript/jerry-libc/target/linux/jerry-libc-target.c:175
# 3  0x000000000044242f in jerry_fatal (code=ERR_FAILED_INTERNAL_ASSERTION) at jerryscript/jerry-core/jrt/jrt-fatals.cpp:65
# 4  0x000000000044247e in jerry_assert_fail (assertion=0x4a6350 "label_p->continue_tgt_oc != MAX_OPCODES", 
    file=0x4a6210 "jerryscript/jerry-core/parser/js/jsp-label.cpp", 
    function=0x4a1560 <_ZZ31jsp_label_rewrite_jumps_and_popP11jsp_label_ttE8__func__.lto_priv.303> "jsp_label_rewrite_jumps_and_pop", line=125)
    at jerryscript/jerry-core/jrt/jrt-fatals.cpp:92
# 5  0x000000000041c6cf in jsp_label_rewrite_jumps_and_pop (label_p=0x7fffffffd3d0, break_tgt_oc=7)
    at jerryscript/jerry-core/parser/js/jsp-label.cpp:125
# 6  0x000000000044a7df in parse_statement (outermost_stmt_label_p=0x0) at jerryscript/jerry-core/parser/js/parser.cpp:2795
# 7  0x000000000044a88e in parse_source_element () at jerryscript/jerry-core/parser/js/parser.cpp:2835
# 8  0x000000000044ad2f in parse_source_element_list(bool) [clone .lto_priv.560] (is_global=true)
    at jerryscript/jerry-core/parser/js/parser.cpp:3074
# 9  0x000000000044aec3 in parser_parse_program (
    source_p=0x6cb780 <_ZL13source_buffer.lto_priv.1117> "label_0: \n    if (false)\n        do  {\n", ' ' <repeats 12 times>, "continue label_0;\n        } while (false);", source_size=93, in_function=false, in_eval=false, is_strict=false, out_opcodes_p=0x7fffffffd620)
    at jerryscript/jerry-core/parser/js/parser.cpp:3134
# 10 0x000000000044b042 in parser_parse_script (
    source=0x6cb780 <_ZL13source_buffer.lto_priv.1117> "label_0: \n    if (false)\n        do  {\n", ' ' <repeats 12 times>, "continue label_0;\n        } while (false);", source_size=93, opcodes_p=0x7fffffffd620) at jerryscript/jerry-core/parser/js/parser.cpp:3203
# 11 0x00000000004991f6 in jerry_parse (
    source_p=0x6cb780 <_ZL13source_buffer.lto_priv.1117> "label_0: \n    if (false)\n        do  {\n", ' ' <repeats 12 times>, "continue label_0;\n        } while (false);", source_size=93) at jerryscript/jerry-core/jerry.cpp:1341
# 12 0x0000000000497e03 in main (argc=3, argv=0x7fffffffd8c8) at jerryscript/main-linux.cpp:285
```

egavrin, 2015-11-19:
Same on master (f17f785b6ec0c71c2ef37e4e80a98aff1eff0963)

egavrin, 2015-11-24:
New testcase:

```js
u:L: continue u:
```

LaszloLango, 2016-02-08:
@renatahodovan, what is the expected result?
On current master (91a0514) fails with the following:
PARSE_ERROR: Labelled statement targeted by a continue noty found. (line: 1, column: 15)

renatahodovan, 2016-02-08:
This is a fuzzer-generated test. Its only expected result is not to fail with an assert.

LaszloLango, 2016-02-08:
Works on current master (91a0514).

closed 2016-02-08 by LaszloLango