jerryscript-project/jerryscript #378: Assertion 'bytes_copied > 0' failed in ecma_builtin_json_quote

renatahodovan, 2015-07-13:
###### Jerry version:
    Checked revision: 13cf314
    Build: debug.linux
###### OS: 
    Ubuntu 15.04, x86_64
###### Test case:
```javascript
JSON.stringify("");
```

###### Backtrace:
```javascript
ICE: Assertion 'bytes_copied > 0' failed at jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-json.cpp(ecma_builtin_json_quote):1132.
Error: ERR_FAILED_INTERNAL_ASSERTION

Program received signal SIGABRT, Aborted.
0x0000000000499b11 in syscall_2_asm () at jerryscript/jerry-libc/target/linux/jerry-asm.S:32
32    SYSCALL_2
(gdb) bt
# 0  0x0000000000499b11 in syscall_2_asm () at jerryscript/jerry-libc/target/linux/jerry-asm.S:32
# 1  0x00000000004003c2 in syscall_2 (syscall_no=62, arg1=21502, arg2=6) at jerryscript/jerry-libc/target/linux/jerry-libc-target.c:96
# 2  0x0000000000400563 in abort () at jerryscript/jerry-libc/target/linux/jerry-libc-target.c:175
# 3  0x0000000000442455 in jerry_fatal (code=ERR_FAILED_INTERNAL_ASSERTION) at jerryscript/jerry-core/jrt/jrt-fatals.cpp:65
# 4  0x00000000004424a4 in jerry_assert_fail (assertion=0x4b4354 "bytes_copied > 0", 
    file=0x4b3de8 "jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-json.cpp", 
    function=0x49d0b0 <_ZZL23ecma_builtin_json_quoteP13ecma_string_tE8__func__.lto_priv.800> "ecma_builtin_json_quote", line=1132)
    at jerryscript/jerry-core/jrt/jrt-fatals.cpp:92
# 5  0x0000000000475f11 in ecma_builtin_json_quote (string_p=0x7cbcd0 <_ZL13mem_heap_area.lto_priv.1056+272>)
    at jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-json.cpp:1132
# 6  0x0000000000476ace in ecma_builtin_json_str(ecma_string_t*, ecma_object_t*, stringify_context_t*) [clone .lto_priv.797] (
    key_p=0x7cbcf0 <_ZL13mem_heap_area.lto_priv.1056+304>, holder_p=0x7cbce8 <_ZL13mem_heap_area.lto_priv.1056+296>, context_p=0x7fffffffd130)
    at jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-json.cpp:1391
# 7  0x0000000000475d65 in ecma_builtin_json_stringify (this_arg=95, arg1=138, arg2=4, arg3=4)
    at jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-json.cpp:1087
# 8  0x0000000000473912 in ecma_builtin_json_dispatch_routine (builtin_routine_id=46, this_arg_value=95, 
    arguments_list=0x80bb50 <_ZL13mem_heap_area.lto_priv.1056+262032>, arguments_number=1)
    at jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-json.inc.h:47
# 9  0x000000000048a6e6 in ecma_builtin_dispatch_routine (builtin_object_id=ECMA_BUILTIN_ID_JSON, builtin_routine_id=46, this_arg_value=95, 
    arguments_list=0x80bb50 <_ZL13mem_heap_area.lto_priv.1056+262032>, arguments_number=1)
    at jerryscript/jerry-core/ecma/builtin-objects/ecma-builtins.inc.h:135
# 10 0x0000000000489ce7 in ecma_builtin_dispatch_call (obj_p=0x7cbca0 <_ZL13mem_heap_area.lto_priv.1056+224>, this_arg_value=95, 
    arguments_list_p=0x80bb50 <_ZL13mem_heap_area.lto_priv.1056+262032>, arguments_list_len=1)
    at jerryscript/jerry-core/ecma/builtin-objects/ecma-builtins.cpp:414
# 11 0x000000000044bbcf in ecma_op_function_call (func_obj_p=0x7cbca0 <_ZL13mem_heap_area.lto_priv.1056+224>, this_arg_value=95, 
    arguments_list_p=0x80bb50 <_ZL13mem_heap_area.lto_priv.1056+262032>, arguments_list_len=1)
    at jerryscript/jerry-core/ecma/operations/ecma-function-object.cpp:714
# 12 0x000000000043ca41 in opfunc_call_n (opdata=..., int_data=0x7fffffffd510) at jerryscript/jerry-core/vm/opcodes.cpp:817
# 13 0x00000000004909be in vm_loop (int_data_p=0x7fffffffd510, run_scope_p=0x0) at jerryscript/jerry-core/vm/vm.cpp:466
# 14 0x0000000000490c4b in vm_run_from_pos (opcodes_p=0x7cbd58 <_ZL13mem_heap_area.lto_priv.1056+408>, start_pos=1, this_binding_value=23, 
    lex_env_p=0x7cbbf8 <_ZL13mem_heap_area.lto_priv.1056+56>, is_strict=false, is_eval_code=false)
    at jerryscript/jerry-core/vm/vm.cpp:556
# 15 0x0000000000490828 in vm_run_global () at jerryscript/jerry-core/vm/vm.cpp:395
# 16 0x000000000049926e in jerry_run () at jerryscript/jerry-core/jerry.cpp:1373
# 17 0x0000000000497e4d in main (argc=3, argv=0x7fffffffd8d8) at jerryscript/main-linux.cpp:294
```

rtakacs, 2015-07-14:
Merge request: #387

egavrin, 2015-07-15:
Fix is on master 292d99ba4e3160590387b3d3d9cedf3a64eb4914

@renatahodovan may we close this issue?

renatahodovan, 2015-07-15:
@egavrin yeah, it looks good. Thanks!

closed 2015-07-15 by ruben-ayrapetyan