jerryscript-project/jerryscript #380: Assertion 'copied > 0' failed in ecma_is_string_magic_longpath

renatahodovan, 2015-07-13:
###### Jerry version:
    Checked revision: 13cf314
    Build: debug.linux
###### OS: 
    Ubuntu 15.04, x86_64
###### Test case:
```javascript
var v_0 = [];
var v_1 = [['\u200E', '\u0601', '\u0602', '\u0603', '\u06DD', '\u070F'],[0x1107, 0x1175, 0x11B0]];
v_0[v_1];
```

###### Backtrace:
```javascript
ICE: Assertion 'copied > 0' failed at jerryscript/jerry-core/ecma/base/ecma-helpers-string.cpp(ecma_is_string_magic_longpath):1675.
Error: ERR_FAILED_INTERNAL_ASSERTION

Program received signal SIGABRT, Aborted.
0x0000000000499b11 in syscall_2_asm () at jerryscript/jerry-libc/target/linux/jerry-asm.S:32
32    SYSCALL_2
(gdb) bt
# 0  0x0000000000499b11 in syscall_2_asm () at jerryscript/jerry-libc/target/linux/jerry-asm.S:32
# 1  0x00000000004003c2 in syscall_2 (syscall_no=62, arg1=23442, arg2=6) at jerryscript/jerry-libc/target/linux/jerry-libc-target.c:96
# 2  0x0000000000400563 in abort () at jerryscript/jerry-libc/target/linux/jerry-libc-target.c:175
# 3  0x0000000000442455 in jerry_fatal (code=ERR_FAILED_INTERNAL_ASSERTION) at jerryscript/jerry-core/jrt/jrt-fatals.cpp:65
# 4  0x00000000004424a4 in jerry_assert_fail (assertion=0x4b09de "copied > 0", 
    file=0x4b0238 "jerryscript/jerry-core/ecma/base/ecma-helpers-string.cpp", 
    function=0x49deb0 <_ZZL29ecma_is_string_magic_longpathPK13ecma_string_tP21lit_magic_string_id_tE8__func__.lto_priv.708> "ecma_is_string_magic_longpath", 
    line=1675) at jerryscript/jerry-core/jrt/jrt-fatals.cpp:92
# 5  0x000000000045b09e in ecma_is_string_magic_longpath (string_p=0x7cc158 <_ZL13mem_heap_area.lto_priv.1056+1432>, out_id_p=0x7fffffffd144)
    at jerryscript/jerry-core/ecma/base/ecma-helpers-string.cpp:1675
# 6  0x000000000045b1ca in ecma_is_string_magic (string_p=0x7cc158 <_ZL13mem_heap_area.lto_priv.1056+1432>, out_id_p=0x7fffffffd144)
    at jerryscript/jerry-core/ecma/base/ecma-helpers-string.cpp:1723
# 7  0x0000000000438b0b in ecma_builtin_array_prototype_try_to_instantiate_property (obj_p=0x7cbc80 <_ZL13mem_heap_area.lto_priv.1056+192>, 
    prop_name_p=0x7cc158 <_ZL13mem_heap_area.lto_priv.1056+1432>)
    at jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-internal-routines-template.inc.h:112
# 8  0x000000000048970a in ecma_builtin_try_to_instantiate_property (object_p=0x7cbc80 <_ZL13mem_heap_area.lto_priv.1056+192>, 
    string_p=0x7cc158 <_ZL13mem_heap_area.lto_priv.1056+1432>) at jerryscript/jerry-core/ecma/builtin-objects/ecma-builtins.inc.h:37
# 9  0x0000000000456a60 in ecma_op_object_get_own_property_longpath (obj_p=0x7cbc80 <_ZL13mem_heap_area.lto_priv.1056+192>, 
    property_name_p=0x7cc158 <_ZL13mem_heap_area.lto_priv.1056+1432>) at jerryscript/jerry-core/ecma/operations/ecma-objects.cpp:145
# 10 0x0000000000456b2c in ecma_op_object_get_own_property (obj_p=0x7cbc80 <_ZL13mem_heap_area.lto_priv.1056+192>, 
    property_name_p=0x7cc158 <_ZL13mem_heap_area.lto_priv.1056+1432>) at jerryscript/jerry-core/ecma/operations/ecma-objects.cpp:177
# 11 0x000000000044ca8a in ecma_op_general_object_get_property (obj_p=0x7cbc80 <_ZL13mem_heap_area.lto_priv.1056+192>, 
    property_name_p=0x7cc158 <_ZL13mem_heap_area.lto_priv.1056+1432>)
    at jerryscript/jerry-core/ecma/operations/ecma-objects-general.cpp:225
# 12 0x0000000000456bd7 in ecma_op_object_get_property (obj_p=0x7cbc80 <_ZL13mem_heap_area.lto_priv.1056+192>, 
    property_name_p=0x7cc158 <_ZL13mem_heap_area.lto_priv.1056+1432>) at jerryscript/jerry-core/ecma/operations/ecma-objects.cpp:218
# 13 0x000000000044cac5 in ecma_op_general_object_get_property (obj_p=0x7cbc90 <_ZL13mem_heap_area.lto_priv.1056+208>, 
    property_name_p=0x7cc158 <_ZL13mem_heap_area.lto_priv.1056+1432>)
    at jerryscript/jerry-core/ecma/operations/ecma-objects-general.cpp:239
# 14 0x0000000000456bd7 in ecma_op_object_get_property (obj_p=0x7cbc90 <_ZL13mem_heap_area.lto_priv.1056+208>, 
    property_name_p=0x7cc158 <_ZL13mem_heap_area.lto_priv.1056+1432>) at jerryscript/jerry-core/ecma/operations/ecma-objects.cpp:218
# 15 0x000000000044c8d9 in ecma_op_general_object_get (obj_p=0x7cbc90 <_ZL13mem_heap_area.lto_priv.1056+208>, 
    property_name_p=0x7cc158 <_ZL13mem_heap_area.lto_priv.1056+1432>)
    at jerryscript/jerry-core/ecma/operations/ecma-objects-general.cpp:149
# 16 0x0000000000456962 in ecma_op_object_get (obj_p=0x7cbc90 <_ZL13mem_heap_area.lto_priv.1056+208>, 
    property_name_p=0x7cc158 <_ZL13mem_heap_area.lto_priv.1056+1432>) at jerryscript/jerry-core/ecma/operations/ecma-objects.cpp:80
# 17 0x0000000000457478 in ecma_op_get_value_object_base (ref=...) at jerryscript/jerry-core/ecma/operations/ecma-get-put-value.cpp:103
# 18 0x000000000043dd0c in opfunc_prop_getter (opdata=..., int_data=0x7fffffffd520) at jerryscript/jerry-core/vm/opcodes.cpp:1234
# 19 0x00000000004909be in vm_loop (int_data_p=0x7fffffffd520, run_scope_p=0x0) at jerryscript/jerry-core/vm/vm.cpp:466
# 20 0x0000000000490c4b in vm_run_from_pos (opcodes_p=0x7cbdd8 <_ZL13mem_heap_area.lto_priv.1056+536>, start_pos=1, this_binding_value=23, 
    lex_env_p=0x7cbbf8 <_ZL13mem_heap_area.lto_priv.1056+56>, is_strict=false, is_eval_code=false)
    at jerryscript/jerry-core/vm/vm.cpp:556
# 21 0x0000000000490828 in vm_run_global () at jerryscript/jerry-core/vm/vm.cpp:395
# 22 0x000000000049926e in jerry_run () at jerryscript/jerry-core/jerry.cpp:1373
# 23 0x0000000000497e4d in main (argc=3, argv=0x7fffffffd8e8) at jerryscript/main-linux.cpp:294
```

egavrin, 2015-11-19:
Same on master (f17f785b6ec0c71c2ef37e4e80a98aff1eff0963)

LaszloLango, 2016-02-08:
Fails on current master (91a0514).

ktorpi, 2016-07-08:
Works on current master (97be8bf).

closed 2016-07-08 by LaszloLango