jerryscript-project/jerryscript #381: Assertion 'STACK_ELEMENT (props, j).type == VARG' failed in syntax_check_for_syntax_errors_in_formal_param_list

renatahodovan, 2015-07-13:
###### Jerry version:
    Checked revision: 13cf314
    Build: debug.linux
###### OS: 
    Ubuntu 15.04, x86_64
###### Test case:
```javascript
function f_0(){
    'use strict';
    var v_0 = {
        p_0: function(a_0, a_1) {}
    };
    f_1(function(a_2) {});
}
```

###### Backtrace:
```javascript
ICE: Assertion 'STACK_ELEMENT (props, j).type == VARG' failed at jerryscript/jerry-core/parser/js/syntax-errors.cpp(syntax_check_for_syntax_errors_in_formal_param_list):220.
Error: ERR_FAILED_INTERNAL_ASSERTION

Program received signal SIGABRT, Aborted.
0x0000000000499b11 in syscall_2_asm () at jerryscript/jerry-libc/target/linux/jerry-asm.S:32
32    SYSCALL_2
(gdb) bt
# 0  0x0000000000499b11 in syscall_2_asm () at jerryscript/jerry-libc/target/linux/jerry-asm.S:32
# 1  0x00000000004003c2 in syscall_2 (syscall_no=62, arg1=24749, arg2=6) at jerryscript/jerry-libc/target/linux/jerry-libc-target.c:96
# 2  0x0000000000400563 in abort () at jerryscript/jerry-libc/target/linux/jerry-libc-target.c:175
# 3  0x0000000000442455 in jerry_fatal (code=ERR_FAILED_INTERNAL_ASSERTION) at jerryscript/jerry-core/jrt/jrt-fatals.cpp:65
# 4  0x00000000004424a4 in jerry_assert_fail (assertion=0x4a6820 "STACK_ELEMENT (props, j).type == VARG", 
    file=0x4a6508 "jerryscript/jerry-core/parser/js/syntax-errors.cpp", 
    function=0x4a16e0 <_ZZ51syntax_check_for_syntax_errors_in_formal_param_listb23lit_utf8_iterator_pos_tE8__func__.lto_priv.293> "syntax_check_for_syntax_errors_in_formal_param_list", line=220) at jerryscript/jerry-core/jrt/jrt-fatals.cpp:92
# 5  0x000000000041d8d7 in syntax_check_for_syntax_errors_in_formal_param_list (is_strict=true, loc=...)
    at jerryscript/jerry-core/parser/js/syntax-errors.cpp:220
# 6  0x0000000000447240 in parse_function_declaration () at jerryscript/jerry-core/parser/js/parser.cpp:662
# 7  0x000000000044a8a8 in parse_source_element () at jerryscript/jerry-core/parser/js/parser.cpp:2831
# 8  0x000000000044ad55 in parse_source_element_list(bool) [clone .lto_priv.560] (is_global=true)
    at jerryscript/jerry-core/parser/js/parser.cpp:3074
# 9  0x000000000044aee9 in parser_parse_program (
    source_p=0x6cb780 <_ZL13source_buffer.lto_priv.1117> "function f_0(){\n    'use strict';\n    var v_0 = {\n        p_0: function(a_0, a_1) {}\n    };\n    f_1(function(a_2) {});\n}", source_size=120, in_function=false, in_eval=false, is_strict=false, out_opcodes_p=0x7fffffffd640)
    at jerryscript/jerry-core/parser/js/parser.cpp:3134
# 10 0x000000000044b068 in parser_parse_script (
    source=0x6cb780 <_ZL13source_buffer.lto_priv.1117> "function f_0(){\n    'use strict';\n    var v_0 = {\n        p_0: function(a_0, a_1) {}\n    };\n    f_1(function(a_2) {});\n}", source_size=120, opcodes_p=0x7fffffffd640) at jerryscript/jerry-core/parser/js/parser.cpp:3203
# 11 0x0000000000499221 in jerry_parse (
    source_p=0x6cb780 <_ZL13source_buffer.lto_priv.1117> "function f_0(){\n    'use strict';\n    var v_0 = {\n        p_0: function(a_0, a_1) {}\n    };\n    f_1(function(a_2) {});\n}", source_size=120) at jerryscript/jerry-core/jerry.cpp:1341
# 12 0x0000000000497e2e in main (argc=3, argv=0x7fffffffd8e8) at jerryscript/main-linux.cpp:285
```

ruben-ayrapetyan, 2015-07-20:
Pull request with fix: #412

egavrin, 2015-11-19:
Works on master (f17f785b6ec0c71c2ef37e4e80a98aff1eff0963)

closed 2015-11-19 by egavrin