jerryscript-project/jerryscript #384: Remove redundant !lit_utf8_iterator_is_eos (&source_str_iter) checks from lexer

sand1k, 2015-07-13:
Refactor the lexer's code so that iterator is checked for EOS only in one place.

egavrin, 2015-11-24:
Version: 17a5dd27bbbadc036d83be0448f9edfa90800147

System:
```
Linux eg 4.2.0-18-generic #22-Ubuntu SMP Fri Nov 6 18:25:50 UTC 2015 x86_64 x86_64 x86_64 GNU/Linux
```

Output:
```

ICE: Assertion '!lit_utf8_iterator_is_eos (&iter)' failed at /home/eg/dev/jr/jerry-core/parser/js/lexer.cpp(lexer_is_no_escape_sequences_in_token_string):1865.
Error: ERR_FAILED_INTERNAL_ASSERTION

```
Testcase:

```js
// Copyrnse.

'use stricto
```

```js
// C_pyrnsq.

'use strictI
```

```js
// Copyrnse. Functionn f_0(){
    'use strictT
```

```js
// Copyrnse.

'use strict)
```

```js
// Copyrnse.

'use strict;
```

LaszloLango, 2016-02-08:
Works on current master (91a0514).

closed 2016-02-08 by LaszloLango