jerryscript-project/jerryscript #406: Assertion 'bytes_copied > 0 || !string_len' failed in ecma_builtin_json_quote

renatahodovan, 2015-07-15:
###### Jerry version:
    Checked revision: abc2b55
    Build: debug.linux
###### OS: 
    Ubuntu 15.04, x86_64
###### Test case:
```javascript
JSON.stringify('\u2028')
```

###### Backtrace:
```javascript
ICE: Assertion 'bytes_copied > 0 || !string_len' failed at jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-json.cpp(ecma_builtin_json_quote):1137.
Error: ERR_FAILED_INTERNAL_ASSERTION

Program received signal SIGABRT, Aborted.
0x00000000004a0a9e in syscall_2_asm () at jerryscript/jerry-libc/target/linux/jerry-asm.S:32
32    SYSCALL_2
(gdb) bt
# 0  0x00000000004a0a9e in syscall_2_asm () at jerryscript/jerry-libc/target/linux/jerry-asm.S:32
# 1  0x00000000004003c2 in syscall_2 (syscall_no=62, arg1=27033, arg2=6) at jerryscript/jerry-libc/target/linux/jerry-libc-target.c:96
# 2  0x0000000000400563 in abort () at jerryscript/jerry-libc/target/linux/jerry-libc-target.c:175
# 3  0x0000000000440580 in jerry_fatal(jerry_fatal_code_t) (code=ERR_FAILED_INTERNAL_ASSERTION)
    at jerryscript/jerry-core/jrt/jrt-fatals.cpp:65
# 4  0x00000000004405cf in jerry_assert_fail(char const*, char const*, char const*, unsigned int) (assertion=0x4bb680 "bytes_copied > 0 || !string_len", 
    file=0x4bb108 "jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-json.cpp", 
    function=0x4a4010 <_ZZL23ecma_builtin_json_quoteP13ecma_string_tE8__func__.lto_priv.798> "ecma_builtin_json_quote", line=1137)
    at jerryscript/jerry-core/jrt/jrt-fatals.cpp:92
# 5  0x0000000000478a8e in ecma_builtin_json_quote(ecma_string_t*) [clone .lto_priv.796] (string_p=0x7d2cd0 <_ZL13mem_heap_area.lto_priv.1061+272>)
    at jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-json.cpp:1137
# 6  0x000000000047964b in ecma_builtin_json_str(ecma_string_t*, ecma_object_t*, stringify_context_t*) [clone .lto_priv.795] (
    key_p=0x7d2cf0 <_ZL13mem_heap_area.lto_priv.1061+304>, holder_p=0x7d2ce8 <_ZL13mem_heap_area.lto_priv.1061+296>, context_p=0x7fffffffd100)
    at jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-json.cpp:1396
# 7  0x00000000004788d1 in ecma_builtin_json_stringify (this_arg=95, arg1=138, arg2=4, arg3=4)
    at jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-json.cpp:1092
# 8  0x0000000000476461 in ecma_builtin_json_dispatch_routine (builtin_routine_id=46, this_arg_value=95, 
    arguments_list=0x812b50 <_ZL13mem_heap_area.lto_priv.1061+262032>, arguments_number=1)
    at jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-json.inc.h:47
# 9  0x0000000000495ffc in ecma_builtin_dispatch_routine (builtin_object_id=ECMA_BUILTIN_ID_JSON, builtin_routine_id=46, this_arg_value=95, 
    arguments_list=0x812b50 <_ZL13mem_heap_area.lto_priv.1061+262032>, arguments_number=1)
    at jerryscript/jerry-core/ecma/builtin-objects/ecma-builtins.inc.h:135
# 10 0x00000000004955fd in ecma_builtin_dispatch_call (obj_p=0x7d2ca0 <_ZL13mem_heap_area.lto_priv.1061+224>, this_arg_value=95, 
    arguments_list_p=0x812b50 <_ZL13mem_heap_area.lto_priv.1061+262032>, arguments_list_len=1)
    at jerryscript/jerry-core/ecma/builtin-objects/ecma-builtins.cpp:414
# 11 0x000000000044c89f in ecma_op_function_call (func_obj_p=0x7d2ca0 <_ZL13mem_heap_area.lto_priv.1061+224>, this_arg_value=95, 
    arguments_list_p=0x812b50 <_ZL13mem_heap_area.lto_priv.1061+262032>, arguments_list_len=1)
    at jerryscript/jerry-core/ecma/operations/ecma-function-object.cpp:787
# 12 0x000000000043d276 in opfunc_call_n (opdata=..., int_data=0x7fffffffd500) at jerryscript/jerry-core/vm/opcodes.cpp:866
# 13 0x000000000049d978 in vm_loop (int_data_p=0x7fffffffd500, run_scope_p=0x0) at jerryscript/jerry-core/vm/vm.cpp:466
# 14 0x000000000049dc05 in vm_run_from_pos (opcodes_p=0x7d2d58 <_ZL13mem_heap_area.lto_priv.1061+408>, start_pos=1, this_binding_value=23, 
    lex_env_p=0x7d2bf8 <_ZL13mem_heap_area.lto_priv.1061+56>, is_strict=false, is_eval_code=false)
    at jerryscript/jerry-core/vm/vm.cpp:556
# 15 0x000000000049d7e2 in vm_run_global () at jerryscript/jerry-core/vm/vm.cpp:395
# 16 0x000000000049fe6b in jerry_run () at jerryscript/jerry-core/jerry.cpp:1375
# 17 0x000000000049ea16 in main (argc=3, argv=0x7fffffffd8c8) at jerryscript/main-linux.cpp:294
```

rtakacs, 2015-07-29:
PR: #491

kkristof, 2015-08-06:
Fixed in #491.

closed 2015-08-06 by kkristof