jerryscript-project/jerryscript #411: Out of memory error with JSON.stringify

renatahodovan, 2015-07-15:
###### Jerry version:
    Checked revision: abc2b55
    Build: debug.linux
###### OS: 
    Ubuntu 15.04, x86_64
###### Description:
Calling JSON.stringify with a not too big (length > 38) array ends in an out of memory error.
###### Test case:
```javascript
var a = ["a","a","a","a","a","a","a","a","a","a","a","a","a","a","a","a","a","a","a","a","a","a","a","a","a","a","a","a","a","a","a","a","a","a","a","a","a","a","a"];
print(JSON.stringify(a));
// a.length: 39
```

rtakacs, 2015-07-20:
This happens during ecma_concat_ecma_strings. I think this bug is related to the core.

ruben-ayrapetyan, 2015-07-20:
@rtakacs, the bug is related to `realloc_list` function (ecma-builtin-helpers-json.cpp).

ruben-ayrapetyan, 2015-07-20:
Seems that calculation of block sizes and `current_p` / `block_end_p` pointers assumes that difference of pointers would be equal to arithmetic difference of addresses. Actually, value is the difference divided by sizeof (void*). Usage of incorrect pointers subsequently leads to overwrite of heap structures.

rtakacs, 2015-08-06:
#543 will resolve the bug :)

closed 2015-08-07 by renatahodovan