jerryscript-project/jerryscript #414: Assertion 'ecma_number_is_negative (ret_num)' failed in ecma_builtin_math_object_max

renatahodovan, 2015-07-15:
###### Jerry version:
    Checked revision: abc2b55
    Build: debug.linux
###### OS: 
    Ubuntu 15.04, x86_64
###### Test case:
```javascript
Math.max(Number.POSITIVE_INFINITY, 234);
```

###### Backtrace:
```javascript
ICE: Assertion 'ecma_number_is_negative (ret_num)' failed at jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-math.cpp(ecma_builtin_math_object_max):367.
Error: ERR_FAILED_INTERNAL_ASSERTION

Program received signal SIGABRT, Aborted.
0x00000000004ab0d7 in syscall_2_asm () at jerryscript/jerry-libc/target/linux/jerry-asm.S:32
32    SYSCALL_2
(gdb) bt
# 0  0x00000000004ab0d7 in syscall_2_asm () at jerryscript/jerry-libc/target/linux/jerry-asm.S:32
# 1  0x00000000004003c2 in syscall_2 (syscall_no=62, arg1=1176, arg2=6) at jerryscript/jerry-libc/target/linux/jerry-libc-target.c:96
# 2  0x0000000000400563 in abort () at jerryscript/jerry-libc/target/linux/jerry-libc-target.c:175
# 3  0x0000000000444144 in jerry_fatal(jerry_fatal_code_t) (code=ERR_FAILED_INTERNAL_ASSERTION) at jerryscript/jerry-core/jrt/jrt-fatals.cpp:65
# 4  0x0000000000444193 in jerry_assert_fail(char const*, char const*, char const*, unsigned int) (assertion=0x4b9850 "ecma_number_is_negative (ret_num)", 
    file=0x4b9730 "jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-math.cpp", 
    function=0x4b2dd0 <_ZZL28ecma_builtin_math_object_maxjPKjjE8__func__.lto_priv.388> "ecma_builtin_math_object_max", line=367) at jerryscript/jerry-core/jrt/jrt-fatals.cpp:92
# 5  0x000000000041ec4f in ecma_builtin_math_object_max(unsigned int, unsigned int const*, unsigned int) [clone .lto_priv.427] (this_arg=5159, args=0x81fb50 <_ZL13mem_heap_area.lto_priv.1140+262032>, args_number=2)
    at jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-math.cpp:367
# 6  0x0000000000428f04 in ecma_builtin_math_dispatch_routine (builtin_routine_id=93, this_arg_value=5159, arguments_list=0x81fb50 <_ZL13mem_heap_area.lto_priv.1140+262032>, arguments_number=2)
    at jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-math.inc.h:114
# 7  0x0000000000497e21 in ecma_builtin_dispatch_routine(ecma_builtin_id_t, unsigned short, unsigned int, unsigned int const*, unsigned int) [clone .lto_priv.1098] (builtin_object_id=ECMA_BUILTIN_ID_MATH, 
    builtin_routine_id=93, this_arg_value=5159, arguments_list=0x81fb50 <_ZL13mem_heap_area.lto_priv.1140+262032>, arguments_number=2)
    at jerryscript/jerry-core/ecma/builtin-objects/ecma-builtins.inc.h:125
# 8  0x00000000004a094f in ecma_builtin_dispatch_call (obj_p=0x7e2740 <_ZL13mem_heap_area.lto_priv.1140+11136>, this_arg_value=5159, arguments_list_p=0x81fb50 <_ZL13mem_heap_area.lto_priv.1140+262032>, 
    arguments_list_len=2) at jerryscript/jerry-core/ecma/builtin-objects/ecma-builtins.cpp:419
# 9  0x0000000000450fcc in ecma_op_function_call(ecma_object_t*, unsigned int, unsigned int const*, unsigned int) (func_obj_p=0x7e2740 <_ZL13mem_heap_area.lto_priv.1140+11136>, this_arg_value=5159, 
    arguments_list_p=0x81fb50 <_ZL13mem_heap_area.lto_priv.1140+262032>, arguments_list_len=2) at jerryscript/jerry-core/ecma/operations/ecma-function-object.cpp:787
# 10 0x0000000000441908 in opfunc_call_n (opdata=..., int_data=0x7fffffffdb90) at jerryscript/jerry-core/vm/opcodes.cpp:866
# 11 0x00000000004aacb5 in vm_loop (int_data_p=0x7fffffffdb90, run_scope_p=0x0) at jerryscript/jerry-core/vm/vm.cpp:466
# 12 0x00000000004aaf42 in vm_run_from_pos (opcodes_p=0x7e2cd8 <_ZL13mem_heap_area.lto_priv.1140+12568>, start_pos=1, this_binding_value=835, lex_env_p=0x7e2bc8 <_ZL13mem_heap_area.lto_priv.1140+12296>, 
    is_strict=false, is_eval_code=false) at jerryscript/jerry-core/vm/vm.cpp:556
# 13 0x00000000004aab1f in vm_run_global () at jerryscript/jerry-core/vm/vm.cpp:395
# 14 0x00000000004a69c1 in jerry_run () at jerryscript/jerry-core/jerry.cpp:1375
# 15 0x00000000004a556c in main (argc=3, argv=0x7fffffffdf58) at jerryscript/main-linux.cpp:294
```

renatahodovan, 2015-07-16:
Similar assertion failure happens if calling Math.min with negative infinity:

```javascript
Math.min(Number.NEGATIVE_INFINITY, 234);
```

galpeter, 2015-07-24:
Fixed by #452.

closed 2015-07-24 by galpeter