jerryscript-project/jerryscript #420: Assertion 'ecma_date_month_from_time (time) == mn' failed in ecma_date_make_day

renatahodovan, 2015-07-16:
###### Jerry version:
    Checked revision: b414329
    Build: debug.linux
###### OS: 
    Ubuntu 15.04, x86_64
###### Test case:
```javascript
new Date(2013, -1);
```

###### Backtrace:
```javascript
ICE: Assertion 'ecma_date_month_from_time (time) == mn' failed at jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-helpers-date.cpp(ecma_date_make_day):737.
Error: ERR_FAILED_INTERNAL_ASSERTION

Program received signal SIGABRT, Aborted.
0x00000000004a0ac9 in syscall_2_asm () at jerryscript/jerry-libc/target/linux/jerry-asm.S:32
32    SYSCALL_2
(gdb) bt
# 0  0x00000000004a0ac9 in syscall_2_asm () at jerryscript/jerry-libc/target/linux/jerry-asm.S:32
# 1  0x00000000004003c2 in syscall_2 (syscall_no=62, arg1=18955, arg2=6) at jerryscript/jerry-libc/target/linux/jerry-libc-target.c:96
# 2  0x0000000000400563 in abort () at jerryscript/jerry-libc/target/linux/jerry-libc-target.c:175
# 3  0x000000000044059a in jerry_fatal(jerry_fatal_code_t) (code=ERR_FAILED_INTERNAL_ASSERTION)
    at jerryscript/jerry-core/jrt/jrt-fatals.cpp:65
# 4  0x00000000004405e9 in jerry_assert_fail(char const*, char const*, char const*, unsigned int) (
    assertion=0x4baf98 "ecma_date_month_from_time (time) == mn", 
    file=0x4bae38 "jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-helpers-date.cpp", 
    function=0x4a3e90 <_ZZ18ecma_date_make_daydddE8__func__.lto_priv.807> "ecma_date_make_day", line=737)
    at jerryscript/jerry-core/jrt/jrt-fatals.cpp:92
# 5  0x0000000000475b0b in ecma_date_make_day (year=2013, month=-1, date=1)
    at jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-helpers-date.cpp:737
# 6  0x0000000000421f69 in ecma_date_construct_helper (args=0x812b50 <_ZL13mem_heap_area.lto_priv.1061+262032>, args_len=2)
    at jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-date.cpp:165
# 7  0x0000000000422f73 in ecma_builtin_date_dispatch_construct(unsigned int const*, unsigned int) (
    arguments_list_p=0x812b50 <_ZL13mem_heap_area.lto_priv.1061+262032>, arguments_list_len=2)
    at jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-date.cpp:541
# 8  0x0000000000495cb2 in ecma_builtin_dispatch_construct (obj_p=0x7d2c70 <_ZL13mem_heap_area.lto_priv.1061+176>, 
    arguments_list_p=0x812b50 <_ZL13mem_heap_area.lto_priv.1061+262032>, arguments_list_len=2)
    at jerryscript/jerry-core/ecma/builtin-objects/ecma-builtins.inc.h:153
# 9  0x000000000044ce57 in ecma_op_function_construct (func_obj_p=0x7d2c70 <_ZL13mem_heap_area.lto_priv.1061+176>, 
    arguments_list_p=0x812b50 <_ZL13mem_heap_area.lto_priv.1061+262032>, arguments_list_len=2)
    at jerryscript/jerry-core/ecma/operations/ecma-function-object.cpp:959
# 10 0x000000000043d661 in opfunc_construct_n (opdata=..., int_data=0x7fffffffd510) at jerryscript/jerry-core/vm/opcodes.cpp:957
# 11 0x000000000049d9a3 in vm_loop (int_data_p=0x7fffffffd510, run_scope_p=0x0) at jerryscript/jerry-core/vm/vm.cpp:466
# 12 0x000000000049dc30 in vm_run_from_pos (opcodes_p=0x7d2d58 <_ZL13mem_heap_area.lto_priv.1061+408>, start_pos=1, this_binding_value=23, 
    lex_env_p=0x7d2bf8 <_ZL13mem_heap_area.lto_priv.1061+56>, is_strict=false, is_eval_code=false)
    at jerryscript/jerry-core/vm/vm.cpp:556
# 13 0x000000000049d80d in vm_run_global () at jerryscript/jerry-core/vm/vm.cpp:395
# 14 0x000000000049fe96 in jerry_run () at jerryscript/jerry-core/jerry.cpp:1375
# 15 0x000000000049ea41 in main (argc=3, argv=0x7fffffffd8d8) at jerryscript/main-linux.cpp:294
```

szledan, 2015-07-16:
Could you please assign this to me?

kkristof, 2015-07-16:
Sure :)

kkristof, 2015-07-22:
The #422 pull request also solved this assertion.

closed 2015-07-22 by kkristof