jerryscript-project/jerryscript #428: Segmentation fault in ecma_builtin_global_object_parse_int

renatahodovan, 2015-07-17:
###### Jerry version:
    Checked revision: 6038173
    Build: debug.linux
###### OS: 
    Ubuntu 15.04, x86_64
###### Test case:
```javascript
parseInt('');
```

###### Backtrace:
```javascript
Program received signal SIGSEGV, Segmentation fault.
0x000000000048dba2 in ecma_builtin_global_object_parse_int (this_arg=4, string=118, radix=4)
    at jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-global.cpp:121
121   utf8_string_buff[str_size] = LIT_BYTE_NULL;
(gdb) bt
# 0  0x000000000048dba2 in ecma_builtin_global_object_parse_int (this_arg=4, string=118, radix=4)
    at jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-global.cpp:121
# 1  0x000000000048d9a2 in ecma_builtin_global_dispatch_routine (builtin_routine_id=48, this_arg_value=4, 
    arguments_list=0x812b50 <_ZL13mem_heap_area.lto_priv.1058+262032>, arguments_number=1)
    at jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-global.inc.h:232
# 2  0x00000000004961ff in ecma_builtin_dispatch_routine (builtin_object_id=ECMA_BUILTIN_ID_GLOBAL, builtin_routine_id=48, this_arg_value=4, 
    arguments_list=0x812b50 <_ZL13mem_heap_area.lto_priv.1058+262032>, arguments_number=1)
    at jerryscript/jerry-core/ecma/builtin-objects/ecma-builtins.inc.h:312
# 3  0x0000000000495601 in ecma_builtin_dispatch_call (obj_p=0x7d2c70 <_ZL13mem_heap_area.lto_priv.1058+176>, this_arg_value=4, 
    arguments_list_p=0x812b50 <_ZL13mem_heap_area.lto_priv.1058+262032>, arguments_list_len=1)
    at jerryscript/jerry-core/ecma/builtin-objects/ecma-builtins.cpp:414
# 4  0x000000000044c8cf in ecma_op_function_call (func_obj_p=0x7d2c70 <_ZL13mem_heap_area.lto_priv.1058+176>, this_arg_value=4, 
    arguments_list_p=0x812b50 <_ZL13mem_heap_area.lto_priv.1058+262032>, arguments_list_len=1)
    at jerryscript/jerry-core/ecma/operations/ecma-function-object.cpp:787
# 5  0x000000000043d28a in opfunc_call_n (opdata=..., int_data=0x7fffffffd530) at jerryscript/jerry-core/vm/opcodes.cpp:866
# 6  0x000000000049d97c in vm_loop (int_data_p=0x7fffffffd530, run_scope_p=0x0) at jerryscript/jerry-core/vm/vm.cpp:466
# 7  0x000000000049dc09 in vm_run_from_pos (opcodes_p=0x7d2d58 <_ZL13mem_heap_area.lto_priv.1058+408>, start_pos=1, this_binding_value=23, 
    lex_env_p=0x7d2bf8 <_ZL13mem_heap_area.lto_priv.1058+56>, is_strict=false, is_eval_code=false)
    at jerryscript/jerry-core/vm/vm.cpp:556
# 8  0x000000000049d7e6 in vm_run_global () at jerryscript/jerry-core/vm/vm.cpp:395
# 9  0x000000000049fe6f in jerry_run () at jerryscript/jerry-core/jerry.cpp:1375
# 10 0x000000000049ea1a in main (argc=2, argv=0x7fffffffd8f8) at jerryscript/main-linux.cpp:294
```

galpeter, 2015-07-17:
The problem is that the buffer size calculation is off by one. There was a `+1` for the array alloc, but it was removed in the PR: https://github.com/Samsung/jerryscript/pull/389

galpeter, 2015-07-24:
#430 fixed it, closing issue

closed 2015-07-24 by galpeter