jerryscript-project/jerryscript #429: Assertion 'ecma_is_completion_value_throw (completion) || ecma_is_completion_value_return (completion)' failed in vm_run_from_pos

renatahodovan, 2015-07-17:
###### Jerry version:
    Checked revision: 6038173
    Build: debug.linux
###### OS: 
    Ubuntu 15.04, x86_64
###### Test case:
```javascript
var v_0 = [
    0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15,
    16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31,
    32, 33, 34, 35, 36, 37, 38, 39, 40, 41, 42, 43, 44, 45, 46, 47,
    48, 49, 50, 51, 52, 53, 54, 55, 56, 57, 58, 59, 60, 61, 62, 63,
    64, 65, 66, 67, 68, 69, 70, 71, 72, 73, 74, 75, 76, 77, 78, 79,
    80, 81, 82, 83, 84, 85, 86, 87, 88, 89, 90, 91, 92, 93, 94, 95,
    96, 97, 98, 99, 100, 101, 102, 103, 104, 105, 106, 107, 108, 109, 110, 111,
    112, 113, 114, 115, 116, 117, 118, 119, 120, 121, 122, 123, 124, 125, 126, 127,
    128, 129, 130, 131, 132, 133, 134, 135, 136, 137, 138, 139, 140, 141, 142, 143,
    144, 145, 146, 147, 148, 149, 150, 151, 152, 153, 154, 155, 156, 157, 158, 159,
    160, 161, 162, 163, 164, 165, 166, 167, 168, 169, 170, 171, 172, 173, 174, 175,
    176, 177, 178, 179, 180, 181, 182, 183, 184, 185, 186, 187, 188, 189, 190, 191,
    192, 193, 194, 195, 196, 197, 198, 199, 200, 201, 202, 203, 204, 205, 206, 207,
    208, 209, 210, 211, 212, 213, 214, 215, 216, 217, 218, 219, 220, 221, 222, 223,
    224, 225, 226, 227, 228, 229, 230, 231, 232, 233, 234, 235, 236, 237, 238, 239,
    240, 241, 242, 243, 244, 245, 246, 247, 248, 249, 250, 251, 252, 253, 254, 255
];
```

###### Backtrace:
```javascript
ICE: Assertion 'ecma_is_completion_value_throw (completion) || ecma_is_completion_value_return (completion)' failed at jerryscript/jerry-core/vm/vm.cpp(vm_run_from_pos):559.
Error: ERR_FAILED_INTERNAL_ASSERTION

Program received signal SIGABRT, Aborted.
0x00000000004a0aa2 in syscall_2_asm () at jerryscript/jerry-libc/target/linux/jerry-asm.S:32
32    SYSCALL_2
(gdb) bt
# 0  0x00000000004a0aa2 in syscall_2_asm () at jerryscript/jerry-libc/target/linux/jerry-asm.S:32
# 1  0x00000000004003c2 in syscall_2 (syscall_no=62, arg1=12843, arg2=6) at jerryscript/jerry-libc/target/linux/jerry-libc-target.c:96
# 2  0x0000000000400563 in abort () at jerryscript/jerry-libc/target/linux/jerry-libc-target.c:175
# 3  0x000000000044059a in jerry_fatal(jerry_fatal_code_t) (code=ERR_FAILED_INTERNAL_ASSERTION)
    at jerryscript/jerry-core/jrt/jrt-fatals.cpp:65
# 4  0x00000000004405e9 in jerry_assert_fail(char const*, char const*, char const*, unsigned int) (
    assertion=0x4c2ff8 "ecma_is_completion_value_throw (completion) || ecma_is_completion_value_return (completion)", 
    file=0x4c2de0 "jerryscript/jerry-core/vm/vm.cpp", 
    function=0x4a1600 <_ZZ15vm_run_from_posPK8opcode_ttjP13ecma_object_tbbE8__func__.lto_priv.1049> "vm_run_from_pos", line=559)
    at jerryscript/jerry-core/jrt/jrt-fatals.cpp:92
# 5  0x000000000049dc60 in vm_run_from_pos (opcodes_p=0x7d2d58 <_ZL13mem_heap_area.lto_priv.1058+408>, start_pos=1, this_binding_value=23, 
    lex_env_p=0x7d2bf8 <_ZL13mem_heap_area.lto_priv.1058+56>, is_strict=false, is_eval_code=false)
    at jerryscript/jerry-core/vm/vm.cpp:558
# 6  0x000000000049d7e6 in vm_run_global () at jerryscript/jerry-core/vm/vm.cpp:395
# 7  0x000000000049fe6f in jerry_run () at jerryscript/jerry-core/jerry.cpp:1375
# 8  0x000000000049ea1a in main (argc=3, argv=0x7fffffffd8d8) at jerryscript/main-linux.cpp:294
```

wateret, 2015-10-21:
This example works OK now. I think we can close this.

egavrin, 2015-10-21:
@wateret seams it was fixed/hidden. We need to add this test into source tree, to prevent regressions.

egavrin, 2015-11-19:
Works on master (f17f785b6ec0c71c2ef37e4e80a98aff1eff0963)

closed 2015-11-19 by egavrin