jerryscript-project/jerryscript #439: Segmentation fault at `jerry_api_string_to_char_buffer()` for empty string.

ILyoan, 2015-07-21:
`jerry_api_string_to_char_buffer()` was designed to return negation of required bytes for buffer when given buffer is not sufficient.

This function can take `NULL` for buffer to calculate required bytes to allocate buffer for a string. But if the string is actually empty string ('') then this will result segmentation fault.
```
  ssize_t ret_val = ecma_string_to_utf8_string (string_p, (lit_utf8_byte_t *) buffer_p, buffer_size);
  if (ret_val >= 0)
  {
    buffer_p[ret_val] = 0;
    ret_val++;
  }
  else
  {
    ret_val--;
  }
``` 
here, `ecma_string_to_utf8_string` returns `0`, next if statement hold, and then seg fault.

sand1k, 2015-07-22:
@ILyoan, thanks for reporting, (NULL, 0) case should be processed gracefully.

Maybe, we should also change this function, so it doesn't write '\0' at the end of buffer, because unicode strings may contain 0 in the middle? 
@ILyoan, @ruben-ayrapetyan what do you think?

ruben-ayrapetyan, 2015-07-22:
@sand1k, I think removing `\0` from end of buffer is good solution, as it would also make API more accurate and clear in relation to possible presence of '\0' inside of character set.

ILyoan, 2015-07-24:
@sand1k I don't mind removing `\0` at the end. Actually I didn't know that this API was adding `\0` before encounter this segmentation fault.

ILyoan, 2015-08-16:
@sand1k Can we close this?

egavrin, 2015-11-19:
@ILyoan closing.

closed 2015-11-19 by egavrin