jerryscript-project/jerryscript #440: Regression after #397 has been landed

kkristof, 2015-07-21:
The following test can reproduce the problem:
```
(function () {
  var d = {};
  var a = [arguments];
})();
```
This now throws `ReferenceError`.
It also makes 29 test262 tests to fail.

ruben-ayrapetyan, 2015-07-21:
@kkristof, thank you for raising the issue.

In my environment, ReferenceError is also raised at 005eb044a93b5ecaf7ac385c8fa9c866e3554b38 (just before #397 - abc2b55297eff13538fca2440d127ba79cdcc8b3 - was merged).

So, maybe, the issue is not connected with the #397 patch.
Could you, please, check this in your environment?

ruben-ayrapetyan, 2015-07-21:
@kkristof, with the following patch test finishes successfully in my environment.

Could you, please, also check the following patch for this and other 29 failing tests in your environment, and list tests that are fixed with the patch?

Thank you in advance,

```
diff --git a/jerry-core/parser/js/parser.cpp b/jerry-core/parser/js/parser.cpp
index 082d8d6..6f60d73 100644
--- a/jerry-core/parser/js/parser.cpp
+++ b/jerry-core/parser/js/parser.cpp
@@ -2949,9 +2949,18 @@ preparse_scope (bool is_global)
     }
   }
 
-  while (!token_is (end_tt))
+  size_t nesting_level = 0;
+  while (nesting_level > 0 || !token_is (end_tt))
   {
-    if (token_is (TOK_NAME))
+    if (token_is (TOK_OPEN_BRACE))
+    {
+      nesting_level++;
+    }
+    else if (token_is (TOK_CLOSE_BRACE))
+    {
+      nesting_level--;
+    }
+    else if (token_is (TOK_NAME))
     {
       if (lit_literal_equal_type_cstr (lit_get_literal_by_cp (token_data_as_lit_cp ()), "arguments"))
       {
```

ruben-ayrapetyan, 2015-07-21:
Failure caused by missing check for current nesting level during search for identifier tokens.

kkristof, 2015-07-21:
Sorry, the minimized example was wrong, The original test case was the following:
```
(function () {
    function getFunc() {
        return 10;
    }
    Object.defineProperty(arguments, "0", {
        get: getFunc,
        enumerable: true,
        configurable: false
    });
    try {
        Object.defineProperty(arguments, "0", {
            configurable: true
        });
    } catch (e) {
        return e instanceof TypeError;
    }
    return false;
}(0, 1, 2));
```

The following test262 test cases were affected:
* ch15/15.2/15.2.3/15.2.3.6/15.2.3.6-4-291-1
* ch15/15.2/15.2.3/15.2.3.6/15.2.3.6-4-291
* ch15/15.2/15.2.3/15.2.3.6/15.2.3.6-4-297-1
* ch15/15.2/15.2.3/15.2.3.6/15.2.3.6-4-297
* ch15/15.2/15.2.3/15.2.3.6/15.2.3.6-4-298-1
* ch15/15.2/15.2.3/15.2.3.6/15.2.3.6-4-298
* ch15/15.2/15.2.3/15.2.3.6/15.2.3.6-4-299-1
* ch15/15.2/15.2.3/15.2.3.6/15.2.3.6-4-299
* ch15/15.2/15.2.3/15.2.3.6/15.2.3.6-4-300-1
* ch15/15.2/15.2.3/15.2.3.6/15.2.3.6-4-300
* ch15/15.2/15.2.3/15.2.3.6/15.2.3.6-4-303
* ch15/15.2/15.2.3/15.2.3.6/15.2.3.6-4-309
* ch15/15.2/15.2.3/15.2.3.6/15.2.3.6-4-310
* ch15/15.2/15.2.3/15.2.3.6/15.2.3.6-4-311
* ch15/15.2/15.2.3/15.2.3.6/15.2.3.6-4-312
* ch15/15.2/15.2.3/15.2.3.6/15.2.3.6-4-314-1
* ch15/15.2/15.2.3/15.2.3.6/15.2.3.6-4-314
* ch15/15.2/15.2.3/15.2.3.6/15.2.3.6-4-321-1
* ch15/15.2/15.2.3/15.2.3.6/15.2.3.6-4-321
* ch15/15.2/15.2.3/15.2.3.6/15.2.3.6-4-322-1
* ch15/15.2/15.2.3/15.2.3.6/15.2.3.6-4-322
* ch15/15.2/15.2.3/15.2.3.6/15.2.3.6-4-323-1
* ch15/15.2/15.2.3/15.2.3.6/15.2.3.6-4-323
* ch15/15.2/15.2.3/15.2.3.6/15.2.3.6-4-324-1
* ch15/15.2/15.2.3/15.2.3.6/15.2.3.6-4-324
* ch15/15.3/15.3.4/15.3.4.3/S15.3.4.3_A7_T9
* ch15/15.3/15.3.4/15.3.4.4/S15.3.4.4_A6_T9
* ch15/15.3/15.3.4/15.3.4.5.2/15.3.4.5.2-4-1
* ch15/15.3/15.3.4/15.3.4.5.2/15.3.4.5.2-4-2
* ch15/15.4/15.4.4/15.4.4.22/15.4.4.22-9-c-ii-12

Your patch solves all of them and also the ReferenceError in the description. Thank you.

ruben-ayrapetyan, 2015-07-21:
@kkristof, thank you!

ruben-ayrapetyan, 2015-07-22:
Pull request with fix: #442

closed 2015-07-22 by ruben-ayrapetyan