jerryscript-project/jerryscript #445: Assertion 'buff_index <= buff_size' failed in ecma_builtin_number_prototype_object_to_string

renatahodovan, 2015-07-22:
###### Jerry version:
    Checked revision: 7c16228
    Build: debug.linux
###### OS: 
    Ubuntu 15.04, x86_64
###### Test case:
```javascript
var v_0 = Number
var v_1 = new v_0("1e73")
v_1.toString(35);
```

###### Backtrace:
```javascript
ICE: Assertion 'buff_index <= buff_size' failed at jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-number-prototype.cpp(ecma_builtin_number_prototype_object_to_string):303.
Error: ERR_FAILED_INTERNAL_ASSERTION

Program received signal SIGABRT, Aborted.
0x00000000004a1235 in syscall_2_asm () at jerryscript/jerry-libc/target/linux/jerry-asm.S:32
32    SYSCALL_2
(gdb) bt
# 0  0x00000000004a1235 in syscall_2_asm () at jerryscript/jerry-libc/target/linux/jerry-asm.S:32
# 1  0x00000000004003c2 in syscall_2 (syscall_no=62, arg1=27152, arg2=6) at jerryscript/jerry-libc/target/linux/jerry-libc-target.c:96
# 2  0x0000000000400563 in abort () at jerryscript/jerry-libc/target/linux/jerry-libc-target.c:175
# 3  0x00000000004405e6 in jerry_fatal(jerry_fatal_code_t) (code=ERR_FAILED_INTERNAL_ASSERTION)
    at jerryscript/jerry-core/jrt/jrt-fatals.cpp:65
# 4  0x0000000000440635 in jerry_assert_fail(char const*, char const*, char const*, unsigned int) (assertion=0x4bdb00 "buff_index <= buff_size", 
    file=0x4bda68 "jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-number-prototype.cpp", 
    function=0x4a3ae0 <_ZZL46ecma_builtin_number_prototype_object_to_stringjPKjjE8__func__.lto_priv.861> "ecma_builtin_number_prototype_object_to_string", 
    line=303) at jerryscript/jerry-core/jrt/jrt-fatals.cpp:92
# 5  0x0000000000483991 in ecma_builtin_number_prototype_object_to_string(unsigned int, unsigned int const*, unsigned int) [clone .lto_priv.857] (
    this_arg=147, arguments_list_p=0x813b50 <_ZL13mem_heap_area.lto_priv.1061+262032>, arguments_list_len=1)
    at jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-number-prototype.cpp:303
# 6  0x0000000000482cac in ecma_builtin_number_prototype_dispatch_routine (builtin_routine_id=103, this_arg_value=147, 
    arguments_list=0x813b50 <_ZL13mem_heap_area.lto_priv.1061+262032>, arguments_number=1)
    at jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-number-prototype.inc.h:47
# 7  0x000000000049670c in ecma_builtin_dispatch_routine (builtin_object_id=ECMA_BUILTIN_ID_NUMBER_PROTOTYPE, builtin_routine_id=103, this_arg_value=147, 
    arguments_list=0x813b50 <_ZL13mem_heap_area.lto_priv.1061+262032>, arguments_number=1)
    at jerryscript/jerry-core/ecma/builtin-objects/ecma-builtins.inc.h:91
# 8  0x0000000000495d8f in ecma_builtin_dispatch_call (obj_p=0x7d3e18 <_ZL13mem_heap_area.lto_priv.1061+600>, this_arg_value=147, 
    arguments_list_p=0x813b50 <_ZL13mem_heap_area.lto_priv.1061+262032>, arguments_list_len=1)
    at jerryscript/jerry-core/ecma/builtin-objects/ecma-builtins.cpp:414
# 9  0x000000000044c91b in ecma_op_function_call (func_obj_p=0x7d3e18 <_ZL13mem_heap_area.lto_priv.1061+600>, this_arg_value=147, 
    arguments_list_p=0x813b50 <_ZL13mem_heap_area.lto_priv.1061+262032>, arguments_list_len=1)
    at jerryscript/jerry-core/ecma/operations/ecma-function-object.cpp:787
# 10 0x000000000043d1d6 in opfunc_call_n (opdata=..., int_data=0x7fffffffd500) at jerryscript/jerry-core/vm/opcodes.cpp:866
# 11 0x000000000049e10a in vm_loop (int_data_p=0x7fffffffd500, run_scope_p=0x0) at jerryscript/jerry-core/vm/vm.cpp:466
# 12 0x000000000049e397 in vm_run_from_pos (opcodes_p=0x7d3d58 <_ZL13mem_heap_area.lto_priv.1061+408>, start_pos=1, this_binding_value=23, 
    lex_env_p=0x7d3bf8 <_ZL13mem_heap_area.lto_priv.1061+56>, is_strict=false, is_eval_code=false) at jerryscript/jerry-core/vm/vm.cpp:556
# 13 0x000000000049df74 in vm_run_global () at jerryscript/jerry-core/vm/vm.cpp:395
# 14 0x00000000004a05fd in jerry_run () at jerryscript/jerry-core/jerry.cpp:1375
# 15 0x000000000049f1a8 in main (argc=3, argv=0x7fffffffd8c8) at jerryscript/main-linux.cpp:294
```

renatahodovan, 2015-07-22:
Probably a related crash with the following test:

```javascript
new Number(865/23561782344444444444444444444444444444444*44444444444444444444444444444444444444444444444444444444444444444444444561856134).toString(3);
```

Backtrace:
```javascript
Program received signal SIGSEGV, Segmentation fault.
0x000000000048353a in ecma_builtin_number_prototype_object_to_string(unsigned int, unsigned int const*, unsigned int) [clone .lto_priv.857] (this_arg=151, 
    arguments_list_p=0x813b50 <_ZL13mem_heap_area.lto_priv.1061+262032>, arguments_list_len=1)
    at jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-number-prototype.cpp:195
195         buff[buff_index++] = (lit_utf8_byte_t) (whole % radix);
(gdb) bt
# 0  0x000000000048353a in ecma_builtin_number_prototype_object_to_string(unsigned int, unsigned int const*, unsigned int) [clone .lto_priv.857] (
    this_arg=151, arguments_list_p=0x813b50 <_ZL13mem_heap_area.lto_priv.1061+262032>, arguments_list_len=1)
    at jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-number-prototype.cpp:195
# 1  0x0000000000482cac in ecma_builtin_number_prototype_dispatch_routine (builtin_routine_id=103, this_arg_value=151, 
    arguments_list=0x813b50 <_ZL13mem_heap_area.lto_priv.1061+262032>, arguments_number=1)
    at jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-number-prototype.inc.h:47
# 2  0x000000000049670c in ecma_builtin_dispatch_routine (builtin_object_id=ECMA_BUILTIN_ID_NUMBER_PROTOTYPE, builtin_routine_id=103, this_arg_value=151, 
    arguments_list=0x813b50 <_ZL13mem_heap_area.lto_priv.1061+262032>, arguments_number=1)
    at jerryscript/jerry-core/ecma/builtin-objects/ecma-builtins.inc.h:91
# 3  0x0000000000495d8f in ecma_builtin_dispatch_call (obj_p=0x7d3c90 <_ZL13mem_heap_area.lto_priv.1061+208>, this_arg_value=151, 
    arguments_list_p=0x813b50 <_ZL13mem_heap_area.lto_priv.1061+262032>, arguments_list_len=1)
    at jerryscript/jerry-core/ecma/builtin-objects/ecma-builtins.cpp:414
# 4  0x000000000044c91b in ecma_op_function_call (func_obj_p=0x7d3c90 <_ZL13mem_heap_area.lto_priv.1061+208>, this_arg_value=151, 
    arguments_list_p=0x813b50 <_ZL13mem_heap_area.lto_priv.1061+262032>, arguments_list_len=1)
    at jerryscript/jerry-core/ecma/operations/ecma-function-object.cpp:787
# 5  0x000000000043d1d6 in opfunc_call_n (opdata=..., int_data=0x7fffffffd510) at jerryscript/jerry-core/vm/opcodes.cpp:866
# 6  0x000000000049e10a in vm_loop (int_data_p=0x7fffffffd510, run_scope_p=0x0) at jerryscript/jerry-core/vm/vm.cpp:466
# 7  0x000000000049e397 in vm_run_from_pos (opcodes_p=0x7d3d58 <_ZL13mem_heap_area.lto_priv.1061+408>, start_pos=1, this_binding_value=23, 
    lex_env_p=0x7d3bf8 <_ZL13mem_heap_area.lto_priv.1061+56>, is_strict=false, is_eval_code=false) at jerryscript/jerry-core/vm/vm.cpp:556
# 8  0x000000000049df74 in vm_run_global () at jerryscript/jerry-core/vm/vm.cpp:395
# 9  0x00000000004a05fd in jerry_run () at jerryscript/jerry-core/jerry.cpp:1375
# 10 0x000000000049f1a8 in main (argc=2, argv=0x7fffffffd8d8) at jerryscript/main-linux.cpp:294
```

galpeter, 2015-08-07:
Closing as fix was landed into master and the assert doesn't occur now.

closed 2015-08-07 by galpeter