jerryscript-project/jerryscript #447: Assertion 'lit_cp.packed_value != MEM_CP_NULL' failed in get_variable_value

renatahodovan, 2015-07-22:
###### Jerry version:
    Checked revision: 7c16228
    Build: debug.linux
###### OS: 
    Ubuntu 15.04, x86_64
###### Test case:
```javascript
for (v_0 in v_1) {};
f_0(v_2.v_3 === 5);
```

###### Backtrace:
```javascript
ICE: Assertion 'lit_cp.packed_value != MEM_CP_NULL' failed at jerryscript/jerry-core/vm/opcodes-helpers-variables.cpp(get_variable_value):95.
Error: ERR_FAILED_INTERNAL_ASSERTION

Program received signal SIGABRT, Aborted.
0x00000000004a1235 in syscall_2_asm () at jerryscript/jerry-libc/target/linux/jerry-asm.S:32
32    SYSCALL_2
(gdb) bt
# 0  0x00000000004a1235 in syscall_2_asm () at jerryscript/jerry-libc/target/linux/jerry-asm.S:32
# 1  0x00000000004003c2 in syscall_2 (syscall_no=62, arg1=6489, arg2=6) at jerryscript/jerry-libc/target/linux/jerry-libc-target.c:96
# 2  0x0000000000400563 in abort () at jerryscript/jerry-libc/target/linux/jerry-libc-target.c:175
# 3  0x00000000004405e6 in jerry_fatal(jerry_fatal_code_t) (code=ERR_FAILED_INTERNAL_ASSERTION)
    at jerryscript/jerry-core/jrt/jrt-fatals.cpp:65
# 4  0x0000000000440635 in jerry_assert_fail(char const*, char const*, char const*, unsigned int) (assertion=0x4c1420 "lit_cp.packed_value != MEM_CP_NULL", 
    file=0x4c1360 "jerryscript/jerry-core/vm/opcodes-helpers-variables.cpp", 
    function=0x4a1f70 <_ZZ18get_variable_valueP10int_data_thbE8__func__.lto_priv.1031> "get_variable_value", line=95)
    at jerryscript/jerry-core/jrt/jrt-fatals.cpp:92
# 5  0x00000000004984a7 in get_variable_value (int_data=0x7fffffffd500, var_idx=0 '\000', do_eval_or_arguments_check=false)
    at jerryscript/jerry-core/vm/opcodes-helpers-variables.cpp:95
# 6  0x000000000049363b in opfunc_for_in (opdata=..., int_data_p=0x7fffffffd500) at jerryscript/jerry-core/vm/opcodes-for-in.cpp:198
# 7  0x000000000049e10a in vm_loop (int_data_p=0x7fffffffd500, run_scope_p=0x0) at jerryscript/jerry-core/vm/vm.cpp:466
# 8  0x000000000049e397 in vm_run_from_pos (opcodes_p=0x7d3d98 <_ZL13mem_heap_area.lto_priv.1061+472>, start_pos=1, this_binding_value=23, 
    lex_env_p=0x7d3bf8 <_ZL13mem_heap_area.lto_priv.1061+56>, is_strict=false, is_eval_code=false) at jerryscript/jerry-core/vm/vm.cpp:556
# 9  0x000000000049df74 in vm_run_global () at jerryscript/jerry-core/vm/vm.cpp:395
# 10 0x00000000004a05fd in jerry_run () at jerryscript/jerry-core/jerry.cpp:1375
# 11 0x000000000049f1a8 in main (argc=3, argv=0x7fffffffd8c8) at jerryscript/main-linux.cpp:294
```

egavrin, 2015-11-19:
Same on master (f17f785b6ec0c71c2ef37e4e80a98aff1eff0963)

LaszloLango, 2016-02-08:
Works on current master (91a0514).

closed 2016-02-08 by LaszloLango