jerryscript-project/jerryscript #463: Assertion 'mem_heap.allocated_bytes >= bytes' failed in mem_heap_free_block

renatahodovan, 2015-07-24:
###### Jerry version:
    Checked revision: 021fc62
    Build: debug.linux
###### OS: 
    Ubuntu 15.04, x86_64
###### Test case:
```javascript
var v_0, v_1, v_2, v_3, v_4, v_5, v_6, v_7, v_8, v_9, v_10, v_11, v_12, v_13, v_14, v_15, v_16, v_17, v_18, v_19, v_20, v_21, v_22, v_23, v_24, v_25, v26, v_27, v_28, v_29;

function f_0() { JSON.stringify(this); }
function f_1() {}
function f_2() {}
function f_3() {}
function f_4() {}
function f_5() {}
function f_6() {}
function f_7() {}
f_0();
```

###### Backtrace:
```javascript
ICE: Assertion 'mem_heap.allocated_bytes >= bytes' failed at jerryscript/jerry-core/mem/mem-heap.cpp(mem_heap_free_block):788.
Error: ERR_FAILED_INTERNAL_ASSERTION

Program received signal SIGABRT, Aborted.
0x00000000004a26b2 in syscall_2_asm () at jerryscript/jerry-libc/target/linux/jerry-asm.S:32
32    SYSCALL_2
(gdb) bt
# 0  0x00000000004a26b2 in syscall_2_asm () at jerryscript/jerry-libc/target/linux/jerry-asm.S:32
# 1  0x00000000004003c2 in syscall_2 (syscall_no=62, arg1=14163, arg2=6) at jerryscript/jerry-libc/target/linux/jerry-libc-target.c:96
# 2  0x0000000000400563 in abort () at jerryscript/jerry-libc/target/linux/jerry-libc-target.c:175
# 3  0x000000000044102f in jerry_fatal(jerry_fatal_code_t) (code=ERR_FAILED_INTERNAL_ASSERTION)
    at jerryscript/jerry-core/jrt/jrt-fatals.cpp:65
# 4  0x000000000044107e in jerry_assert_fail(char const*, char const*, char const*, unsigned int) (assertion=0x4c4ca8 "mem_heap.allocated_bytes >= bytes", 
    file=0x4c46b0 "jerryscript/jerry-core/mem/mem-heap.cpp", 
    function=0x4a2e50 <_ZZ19mem_heap_free_blockPvE8__func__.lto_priv.1056> "mem_heap_free_block", line=788)
    at jerryscript/jerry-core/jrt/jrt-fatals.cpp:92
# 5  0x000000000049f084 in mem_heap_free_block (ptr=0x8149d0 <_ZL13mem_heap_area.lto_priv.1048+261648>)
    at jerryscript/jerry-core/mem/mem-heap.cpp:788
# 6  0x00000000004171b9 in free_list (ctx_p=0x7fffffffced8) at jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-helpers-json.cpp:169
# 7  0x0000000000478774 in ecma_builtin_json_stringify (this_arg=1495, arg1=23, arg2=4, arg3=4)
    at jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-json.cpp:1193
# 8  0x000000000047602b in ecma_builtin_json_dispatch_routine (builtin_routine_id=46, this_arg_value=1495, 
    arguments_list=0x814b10 <_ZL13mem_heap_area.lto_priv.1048+261968>, arguments_number=1)
    at jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-json.inc.h:47
# 9  0x0000000000491e29 in ecma_builtin_dispatch_routine (builtin_object_id=ECMA_BUILTIN_ID_JSON, builtin_routine_id=46, this_arg_value=1495, 
    arguments_list=0x814b10 <_ZL13mem_heap_area.lto_priv.1048+261968>, arguments_number=1)
    at jerryscript/jerry-core/ecma/builtin-objects/ecma-builtins.inc.h:135
# 10 0x000000000049142a in ecma_builtin_dispatch_call (obj_p=0x7d5790 <_ZL13mem_heap_area.lto_priv.1048+3024>, this_arg_value=1495, 
    arguments_list_p=0x814b10 <_ZL13mem_heap_area.lto_priv.1048+261968>, arguments_list_len=1)
    at jerryscript/jerry-core/ecma/builtin-objects/ecma-builtins.cpp:414
# 11 0x000000000044dcef in ecma_op_function_call(ecma_object_t*, unsigned int, unsigned int const*, unsigned int) (
    func_obj_p=0x7d5790 <_ZL13mem_heap_area.lto_priv.1048+3024>, this_arg_value=1495, arguments_list_p=0x814b10 <_ZL13mem_heap_area.lto_priv.1048+261968>, 
    arguments_list_len=1) at jerryscript/jerry-core/ecma/operations/ecma-function-object.cpp:787
# 12 0x000000000043450e in opfunc_call_n (opdata=..., frame_ctx_p=0x7fffffffd2c0) at jerryscript/jerry-core/vm/opcodes.cpp:866
# 13 0x00000000004a0209 in vm_loop (int_data_p=0x7fffffffd2c0, run_scope_p=0x0) at jerryscript/jerry-core/vm/vm.cpp:470
# 14 0x0000000000498ea8 in vm_run_from_pos (opcodes_p=0x7d4fd8 <_ZL13mem_heap_area.lto_priv.1048+1048>, start_pos=35, this_binding_value=23, 
    lex_env_p=0x7d5738 <_ZL13mem_heap_area.lto_priv.1048+2936>, is_strict=false, is_eval_code=false) at jerryscript/jerry-core/vm/vm.cpp:560
# 15 0x000000000044dc62 in ecma_op_function_call(ecma_object_t*, unsigned int, unsigned int const*, unsigned int) (
    func_obj_p=0x7d5338 <_ZL13mem_heap_area.lto_priv.1048+1912>, this_arg_value=4, arguments_list_p=0x0, arguments_list_len=0)
    at jerryscript/jerry-core/ecma/operations/ecma-function-object.cpp:765
# 16 0x000000000043450e in opfunc_call_n (opdata=..., frame_ctx_p=0x7fffffffd530) at jerryscript/jerry-core/vm/opcodes.cpp:866
# 17 0x00000000004a0209 in vm_loop (int_data_p=0x7fffffffd530, run_scope_p=0x0) at jerryscript/jerry-core/vm/vm.cpp:470
# 18 0x0000000000498ea8 in vm_run_from_pos (opcodes_p=0x7d4fd8 <_ZL13mem_heap_area.lto_priv.1048+1048>, start_pos=1, this_binding_value=23, 
    lex_env_p=0x7d4bf8 <_ZL13mem_heap_area.lto_priv.1048+56>, is_strict=false, is_eval_code=false) at jerryscript/jerry-core/vm/vm.cpp:560
# 19 0x00000000004a0073 in vm_run_global () at jerryscript/jerry-core/vm/vm.cpp:399
# 20 0x00000000004a1e0a in jerry_run () at jerryscript/jerry-core/jerry.cpp:1375
# 21 0x00000000004a09b5 in main (argc=3, argv=0x7fffffffd8f8) at jerryscript/main-linux.cpp:294
```

rtakacs, 2015-08-07:
This bug is already fixed. There is no assert with this test.

ruben-ayrapetyan, 2015-08-07:
@rtakacs, was the bug fixed by #543?

rtakacs, 2015-08-07:
Yes :)

ruben-ayrapetyan, 2015-08-07:
Ok, thanks.

closed 2015-08-07 by renatahodovan