jerryscript-project/jerryscript #464: Assertion 'lit_is_utf8_string_valid (string_p, string_size)' failed in ecma_new_ecma_string_from_utf8

renatahodovan, 2015-07-24:
###### Jerry version:
    Checked revision: 021fc62
    Build: debug.linux
###### OS: 
    Ubuntu 15.04, x86_64
###### Test case:
```javascript
'\n'
/./.exec("񉞍");
```

The content of exec in hex: F1 89 9E 8D

###### Backtrace:
```javascript
ICE: Assertion 'lit_is_utf8_string_valid (string_p, string_size)' failed at jerryscript/jerry-core/ecma/base/ecma-helpers-string.cpp(ecma_new_ecma_string_from_utf8):416.
Error: ERR_FAILED_INTERNAL_ASSERTION

Program received signal SIGABRT, Aborted.
0x00000000004a26b2 in syscall_2_asm () at jerryscript/jerry-libc/target/linux/jerry-asm.S:32
32    SYSCALL_2
(gdb) bt
# 0  0x00000000004a26b2 in syscall_2_asm () at jerryscript/jerry-libc/target/linux/jerry-asm.S:32
# 1  0x00000000004003c2 in syscall_2 (syscall_no=62, arg1=21746, arg2=6) at jerryscript/jerry-libc/target/linux/jerry-libc-target.c:96
# 2  0x0000000000400563 in abort () at jerryscript/jerry-libc/target/linux/jerry-libc-target.c:175
# 3  0x000000000044102f in jerry_fatal(jerry_fatal_code_t) (code=ERR_FAILED_INTERNAL_ASSERTION)
    at jerryscript/jerry-core/jrt/jrt-fatals.cpp:65
# 4  0x000000000044107e in jerry_assert_fail(char const*, char const*, char const*, unsigned int) (
    assertion=0x4b9ba0 "lit_is_utf8_string_valid (string_p, string_size)", 
    file=0x4b9958 "jerryscript/jerry-core/ecma/base/ecma-helpers-string.cpp", 
    function=0x4a66b0 <_ZZ30ecma_new_ecma_string_from_utf8PKhjE8__func__.lto_priv.721> "ecma_new_ecma_string_from_utf8", line=416)
    at jerryscript/jerry-core/jrt/jrt-fatals.cpp:92
# 5  0x000000000045f53e in ecma_new_ecma_string_from_utf8 (string_p=0x814b10 <_ZL13mem_heap_area.lto_priv.1048+261968> "\361\211\236\215", string_size=1)
    at jerryscript/jerry-core/ecma/base/ecma-helpers-string.cpp:416
# 6  0x000000000041a382 in ecma_regexp_exec_helper (obj_p=0x7d4c88 <_ZL13mem_heap_area.lto_priv.1048+200>, 
    bc_p=0x8149dc <_ZL13mem_heap_area.lto_priv.1048+261660> "\r\001", str_p=0x814b10 <_ZL13mem_heap_area.lto_priv.1048+261968> "\361\211\236\215", str_size=4)
    at jerryscript/jerry-core/ecma/operations/ecma-regexp-object.cpp:1326
# 7  0x000000000041eaf9 in ecma_builtin_regexp_prototype_exec (this_arg=103, arg=290)
    at jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-regexp-prototype.cpp:89
# 8  0x000000000041e85b in ecma_builtin_regexp_prototype_dispatch_routine (builtin_routine_id=195, this_arg_value=103, 
    arguments_list=0x814b50 <_ZL13mem_heap_area.lto_priv.1048+262032>, arguments_number=1)
    at jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-regexp-prototype.inc.h:43
# 9  0x0000000000491e77 in ecma_builtin_dispatch_routine (builtin_object_id=ECMA_BUILTIN_ID_REGEXP_PROTOTYPE, builtin_routine_id=195, this_arg_value=103, 
    arguments_list=0x814b50 <_ZL13mem_heap_area.lto_priv.1048+262032>, arguments_number=1)
    at jerryscript/jerry-core/ecma/builtin-objects/ecma-builtins.inc.h:163
# 10 0x000000000049142a in ecma_builtin_dispatch_call (obj_p=0x7d4cf8 <_ZL13mem_heap_area.lto_priv.1048+312>, this_arg_value=103, 
    arguments_list_p=0x814b50 <_ZL13mem_heap_area.lto_priv.1048+262032>, arguments_list_len=1)
    at jerryscript/jerry-core/ecma/builtin-objects/ecma-builtins.cpp:414
# 11 0x000000000044dcef in ecma_op_function_call(ecma_object_t*, unsigned int, unsigned int const*, unsigned int) (
    func_obj_p=0x7d4cf8 <_ZL13mem_heap_area.lto_priv.1048+312>, this_arg_value=103, arguments_list_p=0x814b50 <_ZL13mem_heap_area.lto_priv.1048+262032>, 
    arguments_list_len=1) at jerryscript/jerry-core/ecma/operations/ecma-function-object.cpp:787
# 12 0x000000000043450e in opfunc_call_n (opdata=..., frame_ctx_p=0x7fffffffd4f0) at jerryscript/jerry-core/vm/opcodes.cpp:866
# 13 0x00000000004a0209 in vm_loop (int_data_p=0x7fffffffd4f0, run_scope_p=0x0) at jerryscript/jerry-core/vm/vm.cpp:470
# 14 0x0000000000498ea8 in vm_run_from_pos (opcodes_p=0x7d4d58 <_ZL13mem_heap_area.lto_priv.1048+408>, start_pos=1, this_binding_value=23, 
    lex_env_p=0x7d4bf8 <_ZL13mem_heap_area.lto_priv.1048+56>, is_strict=false, is_eval_code=false) at jerryscript/jerry-core/vm/vm.cpp:560
# 15 0x00000000004a0073 in vm_run_global () at jerryscript/jerry-core/vm/vm.cpp:399
# 16 0x00000000004a1e0a in jerry_run () at jerryscript/jerry-core/jerry.cpp:1375
# 17 0x00000000004a09b5 in main (argc=3, argv=0x7fffffffd8b8) at jerryscript/main-linux.cpp:294
```

kkristof, 2015-07-29:
On the current master (https://github.com/Samsung/jerryscript/commit/41aef56cbaec0a91ad9089028676c6b7bca6a53c) it works for me.

galpeter, 2015-09-16:
@renatahodovan , could you please check if this issue is still valid?

egavrin, 2015-11-19:
Works on master (f17f785b6ec0c71c2ef37e4e80a98aff1eff0963), revealed other issue.
```
$ ./build/bin/debug.linux/jerry /tmp/aa.js 
/./.exec("__");
 ^
ERROR: Ln 2, Col 2: Unknown token .
```

closed 2015-11-19 by egavrin