jerryscript-project/jerryscript #465: Segmentation fault in re_parse_char_class

renatahodovan, 2015-07-24:
###### Jerry version:
    Checked revision: 021fc62
    Build: debug.linux
###### OS: 
    Ubuntu 15.04, x86_64
###### Test case:
```javascript
var v_0 = /[\0]C/;
```

###### Backtrace:
```javascript
Program received signal SIGSEGV, Segmentation fault.
0x000000000040a0eb in re_parse_char_class (parser_ctx_p=0x7fffffffd2b0, append_char_class=0x4087e3 <re_append_char_class(void*, uint32_t, uint32_t)>, 
    re_ctx_p=0x7fffffffd270, out_token_p=0x7fffffffd290) at jerryscript/jerry-core/parser/regexp/re-parser.cpp:429
429             || lit_char_is_decimal_digit (RE_LOOKUP (*pattern_p, 1)))
(gdb) bt
# 0  0x000000000040a0eb in re_parse_char_class (parser_ctx_p=0x7fffffffd2b0, append_char_class=0x4087e3 <re_append_char_class(void*, uint32_t, uint32_t)>, 
    re_ctx_p=0x7fffffffd270, out_token_p=0x7fffffffd290) at jerryscript/jerry-core/parser/regexp/re-parser.cpp:429
# 1  0x0000000000409139 in re_parse_alternative (re_ctx_p=0x7fffffffd270, expect_eof=true)
    at jerryscript/jerry-core/parser/regexp/re-compiler.cpp:567
# 2  0x00000000004093f2 in re_compile_bytecode (bytecode_p=0x7d4dd8 <_ZL13mem_heap_area.lto_priv.1048+536>, 
    pattern_str_p=0x7d4c78 <_ZL13mem_heap_area.lto_priv.1048+184>, flags=0 '\000')
    at jerryscript/jerry-core/parser/regexp/re-compiler.cpp:667
# 3  0x0000000000417cff in ecma_op_create_regexp_object (pattern_p=0x7d4c78 <_ZL13mem_heap_area.lto_priv.1048+184>, flags_str_p=0x0)
    at jerryscript/jerry-core/ecma/operations/ecma-regexp-object.cpp:222
# 4  0x000000000043e5c2 in opfunc_assignment (opdata=..., frame_ctx_p=0x7fffffffd520) at jerryscript/jerry-core/vm/opcodes.cpp:207
# 5  0x00000000004a0209 in vm_loop (int_data_p=0x7fffffffd520, run_scope_p=0x0) at jerryscript/jerry-core/vm/vm.cpp:470
# 6  0x0000000000498ea8 in vm_run_from_pos (opcodes_p=0x7d4d58 <_ZL13mem_heap_area.lto_priv.1048+408>, start_pos=1, this_binding_value=23, 
    lex_env_p=0x7d4bf8 <_ZL13mem_heap_area.lto_priv.1048+56>, is_strict=false, is_eval_code=false) at jerryscript/jerry-core/vm/vm.cpp:560
# 7  0x00000000004a0073 in vm_run_global () at jerryscript/jerry-core/vm/vm.cpp:399
# 8  0x00000000004a1e0a in jerry_run () at jerryscript/jerry-core/jerry.cpp:1375
# 9  0x00000000004a09b5 in main (argc=2, argv=0x7fffffffd8e8) at jerryscript/main-linux.cpp:294
```

LaszloLango, 2015-08-03:
@dbatyai, could you check that #505 is fixing this or not?

galpeter, 2015-08-07:
@LaszloLango, with the #505 there is no assert now.

closed 2015-08-11 by dbatyai