jerryscript-project/jerryscript #470: C++

egavrin, 2015-07-24:
Hi folks, 
We'd like to use the following C++ features in our code base:
- class
  - single inheritance
    - _reduce source code size_
  - copy / move constructors
    - _improve perf._
    - _much simplifies GC related logic_
  - virtual functions (very limited, advisable usage)
    - _reduce cross-module dependencies_
- constexpr
 - _improve perf, reduce code size_
- static_assert
 - _quality_
- static_cast (already used), reinterpret_cast
 - _quality_
- try/catch (only for ECMA exception handling, instead of completion_value)
 - _performance, simplifies code_
- enum classes (desirable: replace all enums with enum classes)
 - _quality (type safety)_
- assignment operator implementation (very limited usage)
  - may be, other operators implementation, for example, to replace ecma_number_add with simple '+') 
  - _simplifies code_
- explicitly instantiated templates (very limited, advisable usage, controlled with -fno-implicit-templates)
 - _reduce source code size _

What do you think about this?

zherczeg, 2015-07-27:
Ruben will decide this so I am reassigning this to him.

ruben-ayrapetyan, 2015-07-27:
@zherczeg, this list was prepared by @egavrin, @sand1k and me. We've shared it for review.

egavrin, 2015-07-27:
@zherczeg very strange behavior. We prepared this list to discuss with you.

akiss77, 2015-07-27:
Hi guys. I cannot wholeheartedly support C++ in a very constrained environment like what we are aiming to target with JerryScript. Although it is true that static_assert, casts, and enum classes can improve type safety, and several above-listed feaures reduce _source_ code size, but isn't _binary_ size (and runtime memory usage) what we would like to keep at minimum? E.g., I'm truly concerned about the EH frames that try/catch-es would introduce. Inheritance and templates also ring the alarm bell in the back of my mind. Is there a proof-of-concept work that could show the effect of the proposed changes to the measurable properties of the project (binary size, memory usage, performance)? I'd definitely like to see the results of some experiments that underpin the hypotheses above before we make a decision with such a future-defining effect.

zherczeg, 2015-07-28:
I agree with Akos. We should prove somehow that these features are useful from our perspective before using it. I think we should decide our target benchmark suites first, where we can measure progression. What do you think of SunSpider?

egavrin, 2015-07-28:
@zherczeg regarding benchmarks I see not mane options for now:
- ubench (fully supported)
- several SunSpider 1.0.2 benchmarks (several benchmarks are supported)
- possibly several Kraken ones

zherczeg, 2015-07-28:
These are ok for me. We should also decide the platform where we do measurements. I think the STM4 boards would be a good choice, since it has an ARM Thumb2 32-bit CPU, which is the most likely target for our engine.

Perhaps an automated measurement system would be the best option, where we can submit patches, and it measures these patches and compare them to trunk automatically. We should measure: runtime, memory consumption, and binary size (release with -Os optimization). The measurement system could upload the results to github automatically (part of the CI).

egavrin, 2015-07-28:
STM measurements may take much time. Additionally, these boards are not stable and sometimes require manual reconnect. I'd prefer ARM board with Linux on board, like RaspPi.

zherczeg, 2015-07-28:
Good point. RPi is also a good choice. Which revision? I am not sure rev1 supports thumb 2. Rev 2 has quad core ARMv7 CPU.

closed 2015-11-19 by egavrin