jerryscript-project/jerryscript #489: build error

simapple, 2015-07-29:
/usr/bin/cc cq librelease.jerry-libc.linux.lib.a  CMakeFiles/release.jerry-libc.linux.lib.dir/jerry-libc-fatals.c.o CMakeFiles/release.jerry-libc.linux.lib.dir/jerry-libc-pr   intf.c.o CMakeFiles/release.jerry-libc.linux.lib.dir/jerry-libc.c.o CMakeFiles/release.jerry-libc.linux.lib.dir/target/linux/jerry-libc-target.c.o CMakeFiles/release.jerry-l   ibc.linux.lib.dir/target/linux/jerry-asm.S.o
 gcc: error: cq: No such file or directory
gcc: error: librelease.jerry-libc.linux.lib.a: No such file or directory

akiss77, 2015-07-29:
Could you please specify: What's the compiler you are using? How did you invoke make? What platform are you building on/for?

simapple, 2015-07-29:
cmake 3.2.2
gcc 4.8.4
compile on gentoo  linux

akiss77, 2015-07-29:
My build log contains the following line:
```
/usr/bin/gcc-ar-4.8 cr librelease.jerry-libc.linux.lib.a  CMakeFiles/release.jerry-libc.linux.lib.dir/jerry-libc.c.o CMakeFiles/release.jerry-libc.linux.lib.dir/jerry-libc-printf.c.o CMakeFiles/release.jerry-libc.linux.lib.dir/jerry-libc-fatals.c.o CMakeFiles/release.jerry-libc.linux.lib.dir/target/linux/jerry-libc-target.c.o CMakeFiles/release.jerry-libc.linux.lib.dir/target/linux/jerry-asm.S.o
```

So, it seems that the `ar` tool is not determined correctly. The current cmake listfiles try to deduce its name/location from the location of `gcc` by appending `-ar`. E.g., on my Ubuntu installation, I have `gcc-4.8` and `gcc-ar-4.8` under `/usr/bin`. Is there a chance that you can set up some symlinks, until the issue persists?

LaszloLango, 2016-02-11:
@akiss77, may we close this issue?

akiss77, 2016-02-11:
@LaszloLango well, we still have the same gcc -> gcc-ar logic in the CMakeLists.txt. So, people may still run into this issue. However, I'm not sure when this logic will go away - if ever.

closed 2016-07-28 by zherczeg