jerryscript-project/jerryscript #544: JerryScript based on Espruino or Duktape?

niutech, 2015-08-06:
Why isn't JerryScript based on existing minimal open source ECMAScript engines: [Espruino](https://github.com/espruino/Espruino) or [Duktape](https://github.com/svaarala/duktape) (#494)? Is it a NIH syndrome which makes you reinvent the wheel? Or is JerryScript superior to them?

Thanks!

lemmaa, 2015-08-07:
@niutech  As far as I know, tiny-js is predecessor of Espruino, and it's not designed for full featured. So it's not under our consideration when we first started JerryScript project. Sure, we also realized soon that implementing full featured JavaScript engine with full set of built-in ECMA objects within 100KB is not easily possible, so we changed our first goal to subset version. Anyway, we got a great experience with that and now keep going to implement full featured one. As I mentioned in (https://github.com/Samsung/iotjs/issues/141), our purpose is not only developing another JavaScript engine, but also keep trying disruptive experiment. That's all. :)

seanshpark, 2015-12-11:
maybe close this issue?

egavrin, 2015-12-11:
I think issue is resolved. Closing.

closed 2015-12-11 by egavrin