jerryscript-project/jerryscript #546: Is there any plan to support the ESP8266?

usefulthink, 2015-08-07:
The [ESP8266 by Espressif-Systems][ESP8266] is currently the (by far) cheapest System-on-chip for IoT-purposes with around 40k of available RAM and usually >500kB of flash-memory. With [nodeMCU][] there already exists a firmware that sports a LUA-Interpreter so I would really love to see a JS-interpreter doing the same.

Is this possible at all, and is there any available Documentation on the steps needed to make jerryscript run on this Hardware?

[ESP8266]: http://espressif.com/en/products/esp8266/
[nodeMCU]: http://www.nodemcu.com/

lemmaa, 2015-08-07:
@usefulthink IMHO, JerryScript can running on ESP8266. But, first we have to check up the ESP8266 by ourselves.
Please reference the (https://github.com/Samsung/iotjs/issues/136) also.
Thanks!

JerrySievert, 2015-08-23:
i'm curious as well, as it seems is @maxogden - do you have a timeframe on when you'd be able to try it on an esp8266?

lemmaa, 2015-08-24:
@JerrySievert , We already ordered the ESP8266 but not delivered yet. So, it's hard to estimate the schedule now. I will keep update the progress here.

seanshpark, 2015-10-07:
To whom interested, I'm now working to make JerryScript run in ESP8266.
First thing is getting used to the build system.

usefulthink, 2015-10-07:
\o/ Thank you so much!

JerrySievert, 2015-10-07:
that's great news!

/cc @maxogden, @kenichi

seanshpark, 2015-10-22:
Hello, I've uploaded initial development branch [esp8266-dev](https://github.com/Samsung/jerryscript/commits/esp8266-dev) for ESP-01 board. There are some patches to the SDK that may be not correct. Please add comments. 
First commit just blinks LED from the JS. APIs not just for testing. 

I'll try to bind network API so that we can use it from the JS codes. 
But I am not sure where to start from, so anyone who knows please leave a comment.

seanshpark, 2015-10-22:
About the memory usage...
JerryHeap is set to 20KB and task stack size is 8K and free system heap memory is about 10K.

maxogden, 2015-10-22:
Awesomeness @seanshpark . I'm on Mac OS and will try and figure out how to get the ESP SDK working and report back if I succeed

seanshpark, 2015-10-23:
@maxogden , Nice !

nkolban, 2015-11-19:
Howdy ... I am a keen ESP8266 developer and have a free book on the subject available here ... http://neilkolban.com/tech/esp8266/

I would love to offer any free assistance I can on getting JerryScript running on the ESP8266 and ESP32.

I can be reached by email at kolban1@kolban.com

JerrySievert, 2015-11-19:
@nkolban do you have a source for esp32's?

seanshpark, 2015-11-20:
@nkolban , Wow, glad to hear from you. I've read the book and it helped a lot!. I've made a branch `embedding-dev`, previously `esp8266-dev` but now merged also with `mbed`, and JerryScript itself runs in ESP8266. I was working to enable network, in my personal repo, but have little time to do this while working with others.

nkolban, 2015-11-20:
@JerrySievert Earlier this year I worked hard and created a free book on the ESP8266.  I listed this in a previous post.  Each month I continued to update it.  The good folks at Espressif were pleased with the result and I have offered to update it and/or re-write it for the ESP32.  As such, they have invited me to their beta programme.  I have no idea when I'll receive an instance but when I do, I'll be using it for working on a PDF of notes for us all to build upon.   So ... to your question ... I have no source other than Espressif themselves and I don't expect to receive more than one and that would be for my study to allow me to write the PDF notes.  When it arrives, I plan to go dark on it for months and simply work on the write-up for release when it itself becomes generally available.

@seanshpark I'll be delighted to participate in any way I can.  I assisted with porting the Espruino open source JavaScript to ESP8266 and that is working well, am also tinkering with Smart.js JavaScript .... so I'm growing more and more knowledge on what it takes to get JavaScript going on these ESP8266s.  If you don't have time just now to work any further on the ESP8266 and there is no-one else contributing, then I'll step back until such time as work on it should restart.  I'm keen to collaborate ... but collaboration is the operative words here.  I'll be delighted to bring any and all ESP8266 skills and experience to the table, but I don't know anything yet about the internals of JerryScript ... so would be looking to be part of the community and not so much an active leader.

seanshpark, 2015-11-20:
@nkolban , first thing, thank you for writing down your kind thoughts. As english is not my first language writings may seem dry and somethings it's hard for to express my thoughts and emotions.
YES it would be a big help if you can guide me and others with JerryScript running on ESP8266 and may be yourself with the codes. Current status is that just running JerryScript in ESP8266 is somewhat done, and it would be an honor if you can look the codes [here](https://github.com/seanshpark/jerryscript/tree/embedding-dev/embedding/esp8266) and give comments.
Also wrote some summary I've done with in [here](https://github.com/seanshpark/help/wiki/ESP8266-build-environment-setting) and [here](https://github.com/seanshpark/help/wiki/ESP8266-JerryScript). It would be wonderful if you can give some notes about it.

But after that I had some problems with enabling network, but it was my AP that was working improperly and now I'm waiting for the new one. I've tried also at home and connection itself is ok, but not sure my test codes was proper or not. I've reference `nodemcu` codes. So if possible can you give some informations how to implement this? I'm thinking of exposing JS APIs so that users can control AP connection with JavaScript, or maybe is this a bad idea? Currently I cannot find my code. May be at home...

The next step I'm thinking is more difficult cause as memory is short I can't make IoT.js run so have to build-up more compact JS APis with TCP, UDP and HTTP or maybe CoAP...

seanshpark, 2016-01-14:
Hello, JerryScript on ESP8266 now in master branch, #733.

JerrySievert, 2016-01-14:
@seanshpark thank you!  that's fantastic news!

zherczeg, 2016-07-28:
A patch is recently landed wich adds the support.

closed 2016-07-28 by zherczeg