jerryscript-project/jerryscript #551: Assertion 'mem_pools == NULL' failed in mem_pools_finalize

renatahodovan, 2015-08-10:
###### Jerry version:
    Checked revision: 58851bc
    Build: debug.linux
###### OS: 
    Ubuntu 15.04, x86_64
###### Test case:
```javascript
JSON.stringify([], [ 'foo', 'foo' ]);
```
###### Note:
Although the assertion and the backtrace look similar to the issue will be fixed by #542 but this problem seems unrelated to that one.

###### Backtrace:
```javascript
ICE: Assertion 'mem_pools == NULL' failed at jerryscript/jerry-core/mem/mem-poolman.cpp(mem_pools_finalize):89.
Error: ERR_FAILED_INTERNAL_ASSERTION

Program received signal SIGABRT, Aborted.
0x00000000004a8443 in syscall_2_asm () at jerryscript/jerry-libc/target/linux/jerry-asm.S:32
32    SYSCALL_2
(gdb) bt
# 0  0x00000000004a8443 in syscall_2_asm () at jerryscript/jerry-libc/target/linux/jerry-asm.S:32
# 1  0x0000000000400847 in syscall_2 (syscall_no=62, arg1=8145, arg2=6) at jerryscript/jerry-libc/target/linux/jerry-libc-target.c:96
# 2  0x00000000004009e8 in abort () at jerryscript/jerry-libc/target/linux/jerry-libc-target.c:175
# 3  0x0000000000483ac5 in jerry_fatal (code=ERR_FAILED_INTERNAL_ASSERTION) at jerryscript/jerry-core/jrt/jrt-fatals.cpp:65
# 4  0x0000000000483b14 in jerry_assert_fail (assertion=0x4cb3f3 "mem_pools == NULL", file=0x4cb3b0 "jerryscript/jerry-core/mem/mem-poolman.cpp", 
    function=0x4a9450 <_ZZ18mem_pools_finalizevE8__func__.lto_priv.1071> "mem_pools_finalize", line=89) at jerryscript/jerry-core/jrt/jrt-fatals.cpp:92
# 5  0x00000000004a5827 in mem_pools_finalize () at jerryscript/jerry-core/mem/mem-poolman.cpp:89
# 6  0x00000000004a4265 in mem_finalize (is_show_mem_stats=false) at jerryscript/jerry-core/mem/mem-allocator.cpp:58
# 7  0x00000000004a2661 in jerry_cleanup () at jerryscript/jerry-core/jerry.cpp:1269
# 8  0x00000000004a137b in main (argc=3, argv=0x7fffffffdd08) at jerryscript/main-linux.cpp:299
```

rtakacs, 2015-08-17:
PR: #575

kkristof, 2015-08-19:
This issue is solved by #575.

closed 2015-08-19 by kkristof