jerryscript-project/jerryscript #572: error: control reaches end of non-void function

seanshpark, 2015-08-13:
Some functions like `lit_get_magic_string_utf8()` ends with 
```
  JERRY_UNREACHABLE ();
}
```
and does not return value. 
In this case without `-Wno-return-type` mips compiler fails with an `error: control reaches end of non-void function`. Should I use this option `-Wno-return-type` or should I add something like this?
```
  JERRY_UNREACHABLE ();
  return NULL; /* make compiler happy */
}
```

egavrin, 2015-08-13:
This flag is useful, so I'd prefer to add `return NULL;`.

ruben-ayrapetyan, 2015-08-14:
`JERRY_UNREACHABLE` macro calls `jerry_unreachable` function that is marked with `noreturn` attribute (`jerry_unreachable` leads to either `exit` or `abort`).
So, seems that the warning is false-positive.
To fix this, maybe, `while (true) {}` loop could be added after call to `jerry_unreachable` in the macro.

seanshpark, 2015-08-17:
adding `while (true) {}` to end of `jerry_fatal()` works. 
there are `jerry_assert_fail()`, `jerry_unreachable()` and `jerry_unimplemented()` that calls `jerry_fatal()` and all of them has `noreturn` attribute.
@ruben-ayrapetyan , if you confirm I'll make a PR.

egavrin, 2015-08-17:
@seanshpark @ruben-ayrapetyan 
```cpp
#define JERRY_UNREACHABLE() \
  do \
  { \
    jerry_unreachable (NULL, __FILE__, __func__, __LINE__); \
    __builtin_unreachable(); \
  } while (0)
```

seanshpark, 2015-08-17:
@egavrin , that will change 6 lines in jrt.h, is it ok? :)

egavrin, 2015-08-17:
@seanshpark :+1:

seanshpark, 2015-08-28:
Close as #586 landed

closed 2015-08-28 by seanshpark