jerryscript-project/jerryscript #577: Cross build is broken after #574

egavrin, 2015-08-17:
Seams that issue in this line:
```cpp
JERRY_ASSERT (dest_p - character_buffer == result_string_length);
```

zherczeg, 2015-08-17:
What is the compiler error? Neither make precommit nor make push do not have any problem with it.

zherczeg, 2015-08-17:
Is it a type conversion error?

egavrin, 2015-08-17:
Yes, type conversion.

zherczeg, 2015-08-17:
I don't see this error so I need your help. Please copy the compiler error here, and I will give you a speculative fix. Please try whether it works and then I can create a patch.

egavrin, 2015-08-17:
Do you have cross toolchain? You can reproduce results with the following command:
`make release.linux   -j4 TOOLCHAIN="./build/configs/toolchain_linux_armv7l-el.cmake"`

I'll rebuilt and share output to you.

zherczeg, 2015-08-17:
I have what is downloaded by precommit/push. It has cross compiler dependencies, but they work

egavrin, 2015-08-17:
Please check
```
/home/user/jerry-gh/jerry-core/ecma/builtin-objects/ecma-builtin-helpers-date.cpp: 
In function ‘ecma_completion_value_t
ecma_date_value_to_string(ecma_number_t)’:
/home/user/jerry-gh/jerry-core/ecma/builtin-objects/ecma-builtin-helpers-date.cpp:1090:3: 
error: comparison between signed and unsigned integer expressions [-Werror=sign-compare]
```

zherczeg, 2015-08-17:
ASSERT ((uint32_t) (dest_p - character_buffer) == result_string_length);

Does this fix it?

sand1k, 2015-08-17:
@zherczeg, yes, it should fix the issue.
Please, check by running:
```
make release.linux -j4 TOOLCHAIN=./build/configs/toolchain_linux_armv7l-el.cmake
```

zherczeg, 2015-08-17:
PR #582

closed 2015-08-20 by egavrin