jerryscript-project/jerryscript #590: Assertion 'mem_free_chunks_number == 0' failed in mem_pools_finalize

renatahodovan, 2015-08-18:
###### Jerry version:
    Checked revision: ab47365
    Build: debug.linux
###### OS: 
    Ubuntu 15.04, x86_64
###### Test case:
```javascript
JSON.stringify([], [1, 1]);
```
###### Backtrace:
```javascript
ICE: Assertion 'mem_free_chunks_number == 0' failed at jerryscript/jerry-core/mem/mem-poolman.cpp(mem_pools_finalize):171.
Error: ERR_FAILED_INTERNAL_ASSERTION

Program received signal SIGABRT, Aborted.
0x00000000004a9023 in syscall_2_asm () at jerryscript/jerry-libc/target/linux/jerry-asm.S:32
32	  SYSCALL_2
# 0  0x00000000004a9023 in syscall_2_asm () at jerryscript/jerry-libc/target/linux/jerry-asm.S:32
# 1  0x0000000000400847 in syscall_2 (syscall_no=62, arg1=16519, arg2=6) at jerryscript/jerry-libc/target/linux/jerry-libc-target.c:96
# 2  0x00000000004009e8 in abort () at jerryscript/jerry-libc/target/linux/jerry-libc-target.c:175
# 3  0x000000000046f1f5 in jerry_fatal(jerry_fatal_code_t) (code=ERR_FAILED_INTERNAL_ASSERTION) at jerryscript/jerry-core/jrt/jrt-fatals.cpp:65
# 4  0x000000000046f244 in jerry_assert_fail(char const*, char const*, char const*, unsigned int) (assertion=0x4cbd1b "mem_free_chunks_number == 0", 
    file=0x4cbcd8 "jerryscript/jerry-core/mem/mem-poolman.cpp", function=0x4aa0b0 <_ZZ18mem_pools_finalizevE8__func__.lto_priv.1054> "mem_pools_finalize", line=171)
    at jerryscript/jerry-core/jrt/jrt-fatals.cpp:92
# 5  0x00000000004a5f07 in mem_pools_finalize () at jerryscript/jerry-core/mem/mem-poolman.cpp:171
# 6  0x00000000004a48f4 in mem_finalize (is_show_mem_stats=false) at jerryscript/jerry-core/mem/mem-allocator.cpp:58
# 7  0x00000000004a2cf0 in jerry_cleanup () at jerryscript/jerry-core/jerry.cpp:1351
# 8  0x00000000004a17e3 in main (argc=3, argv=0x7fffffffdd88) at jerryscript/main-linux.cpp:299
```

kkristof, 2015-08-19:
It is very similar to #551 so I think the PR #575 also solve this. I'm going to check this.

kkristof, 2015-08-19:
This issue is solved by #575.

closed 2015-08-19 by kkristof