jerryscript-project/jerryscript #598: Could not build on OSX

mdarse, 2015-08-26:
I can't build JerryScript on Mac OS X Mavericks (10.9.5), I get the following error :
```
❯ make
Setting up prerequisites... (log file: ./build/prerequisites/prerequisites.log)
Prerequisites setup failed. See ./build/prerequisites/prerequisites.log for details.
make: *** [build/prerequisites/.prerequisites] Error 1
```

```
❯ cat build/prerequisites/prerequisites.log 
mktemp: illegal option -- -
usage: mktemp [-d] [-q] [-t prefix] [-u] template ...
       mktemp [-d] [-q] [-u] -t prefix 
/stm32f3.zip: Permission denied
Failed to setup 'stm32f3' prerequisite. Cannot download 'http://www.st.com/st-web-ui/static/active/en/st_prod_software_internet/resource/technical/software/firmware/stm32f3discovery_fw.zip' zip archive.
```

egavrin, 2015-08-26:
By default `make` builds all available targets.
Please, try `make release.linux -j4` which should build the engine: `./build/bin/release.linux/jerry`

poussa, 2015-08-26:
Is building on OS X really supported ? 

On OS X I get the error (since I have clang, not native g++):

```CMake Error at CMakeLists.txt:20 (message):
  g++ compiler is required```

```spoussa@jerryscript: g++ --version
Configured with: --prefix=/Applications/Xcode.app/Contents/Developer/usr --with-gxx-include-dir=/usr/include/c++/4.2.1
Apple LLVM version 6.1.0 (clang-602.0.53) (based on LLVM 3.6.0svn)
Target: x86_64-apple-darwin14.5.0
Thread model: posix```

egavrin, 2015-08-26:
@poussa we're using Ubuntu 14.04+ for development and build system only guaranteed to work on this system: please check http://samsung.github.io/jerryscript/how-to/
Build on OS X is possible, but it may be tricky - let me check.

lemmaa, 2015-08-27:
@mdarse There are some problems during prerequisites set up. It's down to command mismatch between Linux and OS X, such as `mktemp`, `sha256sum` etc. I make a  PR https://github.com/Samsung/jerryscript/pull/600 for it.

lemmaa, 2015-08-30:
@mdarse @poussa I just make a new PR https://github.com/Samsung/jerryscript/pull/607 for it. After merging, you can build JerryScript on Mac OS X using apple C/C++ compiler.
Please, try `make release.darwin -j4`, and check `./build/bin/release.darwin/jerry`.

poussa, 2015-12-11:
@lemmaa thanks for the fix. It now builds for me (on Mac). You can close the issue.

closed 2015-12-12 by egavrin