jerryscript-project/jerryscript #614: Assertion '!ecma_is_value_empty (reg_value)' failed in get_variable_value

renatahodovan, 2015-09-03:
###### Jerry version:
    Checked revision: a26c454
    Build: debug.linux
###### OS: 
    Ubuntu 15.04, x86_64
###### Test case:
```javascript
JSON.stringify & (Date = 1)
```

###### Backtrace:
```javascript

ICE: Assertion '!ecma_is_value_empty (reg_value)' failed at jerryscript/jerry-core/vm/opcodes-helpers-variables.cpp(get_variable_value):87.
Error: ERR_FAILED_INTERNAL_ASSERTION

Program received signal SIGABRT, Aborted.
0x00000000004b445f in syscall_2_asm () at jerryscript/jerry-libc/target/linux/jerry-asm.S:32
32    SYSCALL_2
(gdb) bt
# 0  0x00000000004b445f in syscall_2_asm () at jerryscript/jerry-libc/target/linux/jerry-asm.S:32
# 1  0x00000000004003c2 in syscall_2 (syscall_no=62, arg1=4162, arg2=6) at jerryscript/jerry-libc/target/linux/jerry-libc-target.c:96
# 2  0x0000000000400563 in abort () at jerryscript/jerry-libc/target/linux/jerry-libc-target.c:175
# 3  0x000000000043de52 in jerry_fatal(jerry_fatal_code_t) (code=ERR_FAILED_INTERNAL_ASSERTION)
    at jerryscript/jerry-core/jrt/jrt-fatals.cpp:65
# 4  0x000000000043dea1 in jerry_assert_fail(char const*, char const*, char const*, unsigned int) (assertion=0x4d99b0 "!ecma_is_value_empty (reg_value)", 
    file=0x4d9918 "jerryscript/jerry-core/vm/opcodes-helpers-variables.cpp", 
    function=0x4b5960 <_ZZ18get_variable_valueP14vm_frame_ctx_thbE8__func__.lto_priv.1158> "get_variable_value", line=87)
    at jerryscript/jerry-core/jrt/jrt-fatals.cpp:97
# 5  0x00000000004aa574 in get_variable_value (frame_ctx_p=0x7fffffffd550, var_idx=131 '\203', do_eval_or_arguments_check=false)
    at jerryscript/jerry-core/vm/opcodes-helpers-variables.cpp:87
# 6  0x00000000004b15d5 in opfunc_b_and (instr=..., frame_ctx_p=0x7fffffffd550) at jerryscript/jerry-core/vm/opcodes-ecma-bitwise.cpp:131
# 7  0x00000000004b0b93 in vm_loop (frame_ctx_p=0x7fffffffd550, run_scope_p=0x0) at jerryscript/jerry-core/vm/vm.cpp:484
# 8  0x00000000004b0e2c in vm_run_from_pos (instrs_p=0x7ef508 <mem_heap+12808>, start_pos=1, this_binding_value=1275, lex_env_p=0x7ef448 <mem_heap+12616>, 
    is_strict=false, is_eval_code=false) at jerryscript/jerry-core/vm/vm.cpp:575
# 9  0x00000000004b09fd in vm_run_global () at jerryscript/jerry-core/vm/vm.cpp:413
# 10 0x00000000004b3b9d in jerry_run () at jerryscript/jerry-core/jerry.cpp:1445
# 11 0x00000000004b2556 in main (argc=3, argv=0x7fffffffd918) at jerryscript/main-linux.cpp:295
```

galpeter, 2015-09-16:
I've created a different testcase:
```js
this.a & (b = 1)
```
The assert is the same. I've also dumped the opcodes:
```
// this.a & (b = 1)
  0:                 meta   12  253  255     // [to be rewritten]
  1:         reg_var_decl  128  253  253     // var tmp128 .. tmp253;
  2:         this_binding  130               // tmp130 = this;
  3:           assignment  131    5  254     // tmp131 = 'a': STRING;
  4:          prop_getter  132  130  131     // tmp132 = tmp130[tmp131];
  4:           assignment  133    1    1     // tmp133 = 1: SMALLINT;
  5:          prop_setter  130  131  133     // tmp130[tmp131] = tmp133;
// 
  6:                b_and  134  132  133     // tmp134 = tmp132 & tmp133;
  0:                 meta   12    6  255     // [no 'arguments'] [no 'eval']  // REWRITE
  1:         reg_var_decl  128  134    0     // var tmp128 .. tmp134; // REWRITE
  7:                  ret                    // ret;
LITERALS:
0x8e18c4 [  8] a : STRING
0x8e18cc [  8] b : STRING
0x8e18d4 [ 44]  : EMPTY RECORD
  0:                 meta   12    6  255     // [no 'arguments'] [no 'eval'] 
  1:         reg_var_decl  128  134    0     // var tmp128 .. tmp134;
  2:         this_binding  130               // tmp130 = this;
  3:           assignment  131    5    0     // tmp131 = 'a': STRING;
  4:           assignment  133    1    1     // tmp133 = 1: SMALLINT;
  5:          prop_setter  130  131  133     // tmp130[tmp131] = tmp133;
  6:                b_and  134  132  133     // tmp134 = tmp132 & tmp133;
  7:                  ret                    // ret;
```

It seems for me that there is an incorrect `prop_setter` bytecode generated.

egavrin, 2015-11-19:
Works on master (f17f785b6ec0c71c2ef37e4e80a98aff1eff0963)

closed 2015-11-19 by egavrin