jerryscript-project/jerryscript #615: Partial jerry_api_eval()

seanshpark, 2015-09-03:
@ruben-ayrapetyan , regarding issue comment https://github.com/Samsung/iotjs/issues/212#issuecomment-137080946
can it be possible to extend or add a new function besides jerry_api_eval() so that
```
jerry_api_eval(wrapper[0], ...);
jerry_api_eval(source, ...);
jerry_api_eval(wrapper[1], ...);
```
may work ?

This may be one way not to make additional string buffer for concatenation. Simple or better ways are welcome.

ruben-ayrapetyan, 2015-09-03:
@seanshpark, the API could be added. In the case, jerryscript would perform the concatenation inside the engine. The issue could also be solved in the following way:
 - for external modules, compiled through "Compile" interface, the wrapper strings could be added by JS code in `Module.prototype.compile` function, so using jerryscript's built-in concatenation mechanism; in the case, jerryscript would return buffer size in `GetString` and reallocation would not be necessary;
 - for native modules, the wrapper strings could be added in `js2c.py` script.

seanshpark, 2015-09-03:
@ruben-ayrapetyan , thank you! as an experiment I can try with the second one. 
For the first method, can it be possible without source concatenation? I mean progressive parsing and when the last one is added it can tell Jerry with  some flags so that it can go one with execution.

egavrin, 2015-09-04:
This is possible with the only limitation: passed string should be correct JavaScript expression without syntax errors and this can be `relatively easy` handled:

```cpp
jerry_api_add_eval_line ("function a () {}", ...);
jerry_api_add_eval_line ("function b () {}", ...);
jerry_api_add_eval_line ("function c () {}", ...);
```

But this'll require to handle cases like this without concatenation, which may not be efficient in terms of memory or performance:
```cpp
jerry_api_partial_eval ("function a () {", ...); // save parsing context
jerry_api_partial_eval ("b++", ...); // save parsing context
jerry_api_partial_eval ("}", ...); // save parsing context
```

We can handle this by adding `jerry_api_partial_eval` and `jerry_api_partial_eval_complete`, but this will not reduce memory consumption:
```cpp
jerry_api_partial_eval ("function a () {", ...);
jerry_api_partial_eval ("b++", ...);
jerry_api_partial_eval ("}", ...);
jerry_api_partial_eval_complete (); // concat and parse
```

seanshpark, 2015-09-05:
@egavrin , if the last one performs "concat and parse" I also think it'll be the same. I was thinking of for each

zherczeg, 2016-07-21:
Does this function still needed?

seanshpark, 2016-07-27:
Long time ago hope that got lost :)
Close this issue for now and maybe when required, create a new one.

closed 2016-07-27 by seanshpark