jerryscript-project/jerryscript #636: Support NuttX platform as an interpreter

seanshpark, 2015-09-21:
Adding this issue as a task to support for NuttX.

egavrin, 2015-09-24:
Do we really need to have two separate processes for IoTjs and Jerry?

seanshpark, 2015-09-24:
This is what I'm thinking of. JerryScript and IoT.js are tightly related but it's something different. Providing as many platforms as we can, can help others using JerryScript as an engine for ones purpose. So I want to provide for the ones I can. By doing so we can extend the make related files and codes  and may ease others porting to their platforms. And providing auto-testing can be a huge task and I hope there will be contributors out there.

So the output result, currently for nuttx are two things. One is JerryScript as interpreter. It executes `.js` files. Providing system(file, I/O, ...) may be another thing. There may be no relation with IoT.js. Its program name maybe `jerryscript`. Entry code for this, as I am writing, is [ jerryscript.cxx](https://bitbucket.org/seanshpark/nuttx_apps/src/026d23743263b700d4f9297eb054532fe2c7611e/interpreters/jerryscript/jerryscript.cxx?at=iotjs)

I had some problem with current code so added a simple patch at my fork, https://github.com/seanshpark/jerryscript/commit/18253bcbe0c364504556f7ba8e3406f27e91541b. The main thing is to provide code for running the engine, [main-nuttx.cpp](https://github.com/seanshpark/jerryscript/blob/18253bcbe0c364504556f7ba8e3406f27e91541b/main-nuttx.cpp) and some build related files. I think you can help to merge the main branch cause I'm not sure of current separate [Makefile.nuttx](https://github.com/seanshpark/jerryscript/blob/18253bcbe0c364504556f7ba8e3406f27e91541b/Makefile.nuttx) is a good thing. 

Another is IoT.js, as you know, providing APIs described in IoT.js wiki pages. Program name for this is `iotjs`. Entry code for this is https://bitbucket.org/seanshpark/nuttx_apps/src/026d23743263b700d4f9297eb054532fe2c7611e/system/iotjs/?at=iotjs

closed 2016-08-16 by seanshpark