jerryscript-project/jerryscript #639: Assertion 'lit_literal_exists (lit)' failed in lit_get_literal_by_cp

renatahodovan, 2015-09-21:
###### Jerry version:
    Checked revision: 6697523
    Build: debug.linux
###### OS: 
    Ubuntu 15.04, x86_64
###### Test case:
```javascript
Math.min(10); 
function f_0() {};
function f_1() {
  var f_1 = Date.prototype.getMilliseconds(Date.prototype.setUTCMonth);
}
```

###### Backtrace:
```
ICE: Assertion 'lit_literal_exists (lit)' failed at jerryscript/jerry-core/lit/lit-literal.cpp(lit_get_literal_by_cp):554.
Error: ERR_FAILED_INTERNAL_ASSERTION

Program received signal SIGABRT, Aborted.
0x00000000004acb25 in syscall_2_asm () at jerryscript/jerry-libc/target/linux/jerry-asm.S:32
32    SYSCALL_2
(gdb) bt
# 0  0x00000000004acb25 in syscall_2_asm () at jerryscript/jerry-libc/target/linux/jerry-asm.S:32
# 1  0x00000000004003c2 in syscall_2 (syscall_no=62, arg1=8646, arg2=6) at jerryscript/jerry-libc/target/linux/jerry-libc-target.c:96
# 2  0x0000000000400563 in abort () at jerryscript/jerry-libc/target/linux/jerry-libc-target.c:175
# 3  0x0000000000436073 in jerry_fatal(jerry_fatal_code_t) (code=ERR_FAILED_INTERNAL_ASSERTION)
    at jerryscript/jerry-core/jrt/jrt-fatals.cpp:65
# 4  0x00000000004360c2 in jerry_assert_fail(char const*, char const*, char const*, unsigned int) (assertion=0x4d0253 "lit_literal_exists (lit)", 
    file=0x4d01d8 "jerryscript/jerry-core/lit/lit-literal.cpp", 
    function=0x4ad3e0 <_ZZ21lit_get_literal_by_cpN15rcs_recordset_t8record_t10cpointer_tEE8__func__.lto_priv.1149> "lit_get_literal_by_cp", line=554)
    at jerryscript/jerry-core/jrt/jrt-fatals.cpp:97
# 5  0x00000000004a1402 in lit_get_literal_by_cp (lit_cp=...) at jerryscript/jerry-core/lit/lit-literal.cpp:554
# 6  0x000000000045ba78 in ecma_init_ecma_string_from_lit_cp (string_p=0x8e2990 <mem_heap+1552>, lit_cp=..., is_stack_var=false)
    at jerryscript/jerry-core/ecma/base/ecma-helpers-string.cpp:331
# 7  0x000000000045c187 in ecma_new_ecma_string_from_lit_cp (lit_cp=...) at jerryscript/jerry-core/ecma/base/ecma-helpers-string.cpp:551
# 8  0x0000000000433382 in function_declaration (frame_ctx_p=0x7fffffffcf50, function_name_lit_cp=..., formal_params_collection_p=0x8e2988 <mem_heap+1544>)
    at jerryscript/jerry-core/vm/opcodes.cpp:514
# 9  0x000000000043347d in opfunc_func_decl_n (instr=..., frame_ctx_p=0x7fffffffcf50) at jerryscript/jerry-core/vm/opcodes.cpp:558
# 10 0x00000000004a5ae5 in vm_loop (frame_ctx_p=0x7fffffffcf50, run_scope_p=0x0) at jerryscript/jerry-core/vm/vm.cpp:545
# 11 0x00000000004a5d89 in vm_run_from_pos (header_p=0x8e2840 <mem_heap+1216>, start_pos=1, this_binding_value=523, lex_env_p=0x8e27a0 <mem_heap+1056>, 
    is_strict=false, is_eval_code=false) at jerryscript/jerry-core/vm/vm.cpp:637
# 12 0x00000000004a5805 in vm_run_global () at jerryscript/jerry-core/vm/vm.cpp:415
# 13 0x00000000004abebe in jerry_run () at jerryscript/jerry-core/jerry.cpp:1464
# 14 0x00000000004aa878 in main (argc=3, argv=0x7fffffffd558) at jerryscript/main-linux.cpp:458
```

renatahodovan, 2015-09-21:
Removing a few more lines the failure happens two lines earlier:

```javascript
function f_0() {
  var f_0 = print;
}
```

Note: Any function reference can be on the right-hand side of the assignment.

###### Backtrace:
```
ICE: Assertion 'lit_cp.packed_value != MEM_CP_NULL' failed at jerryscript/jerry-core/lit/lit-literal.cpp(lit_get_literal_by_cp):552.
Error: ERR_FAILED_INTERNAL_ASSERTION

Program received signal SIGABRT, Aborted.
0x00000000004acb25 in syscall_2_asm () at jerryscript/jerry-libc/target/linux/jerry-asm.S:32
32    SYSCALL_2
(gdb) bt
# 0  0x00000000004acb25 in syscall_2_asm () at jerryscript/jerry-libc/target/linux/jerry-asm.S:32
# 1  0x00000000004003c2 in syscall_2 (syscall_no=62, arg1=9260, arg2=6) at jerryscript/jerry-libc/target/linux/jerry-libc-target.c:96
# 2  0x0000000000400563 in abort () at jerryscript/jerry-libc/target/linux/jerry-libc-target.c:175
# 3  0x0000000000436073 in jerry_fatal(jerry_fatal_code_t) (code=ERR_FAILED_INTERNAL_ASSERTION)
    at jerryscript/jerry-core/jrt/jrt-fatals.cpp:65
# 4  0x00000000004360c2 in jerry_assert_fail(char const*, char const*, char const*, unsigned int) (assertion=0x4d0230 "lit_cp.packed_value != MEM_CP_NULL", 
    file=0x4d01d8 "jerryscript/jerry-core/lit/lit-literal.cpp", 
    function=0x4ad3e0 <_ZZ21lit_get_literal_by_cpN15rcs_recordset_t8record_t10cpointer_tEE8__func__.lto_priv.1149> "lit_get_literal_by_cp", line=552)
    at jerryscript/jerry-core/jrt/jrt-fatals.cpp:97
# 5  0x00000000004a13be in lit_get_literal_by_cp (lit_cp=...) at jerryscript/jerry-core/lit/lit-literal.cpp:552
# 6  0x000000000045ba78 in ecma_init_ecma_string_from_lit_cp (string_p=0x8e28d0 <mem_heap+1360>, lit_cp=..., is_stack_var=false)
    at jerryscript/jerry-core/ecma/base/ecma-helpers-string.cpp:331
# 7  0x000000000045c187 in ecma_new_ecma_string_from_lit_cp (lit_cp=...) at jerryscript/jerry-core/ecma/base/ecma-helpers-string.cpp:551
# 8  0x0000000000433382 in function_declaration (frame_ctx_p=0x7fffffffcf50, function_name_lit_cp=..., formal_params_collection_p=0x8e28c8 <mem_heap+1352>)
    at jerryscript/jerry-core/vm/opcodes.cpp:514
# 9  0x000000000043347d in opfunc_func_decl_n (instr=..., frame_ctx_p=0x7fffffffcf50) at jerryscript/jerry-core/vm/opcodes.cpp:558
# 10 0x00000000004a5ae5 in vm_loop (frame_ctx_p=0x7fffffffcf50, run_scope_p=0x0) at jerryscript/jerry-core/vm/vm.cpp:545
# 11 0x00000000004a5d89 in vm_run_from_pos (header_p=0x8e2840 <mem_heap+1216>, start_pos=1, this_binding_value=523, lex_env_p=0x8e27a0 <mem_heap+1056>, 
    is_strict=false, is_eval_code=false) at jerryscript/jerry-core/vm/vm.cpp:637
# 12 0x00000000004a5805 in vm_run_global () at jerryscript/jerry-core/vm/vm.cpp:415
# 13 0x00000000004abebe in jerry_run () at jerryscript/jerry-core/jerry.cpp:1464
# 14 0x00000000004aa878 in main (argc=3, argv=0x7fffffffd558) at jerryscript/main-linux.cpp:458
```

egavrin, 2015-11-19:
Works on master (f17f785b6ec0c71c2ef37e4e80a98aff1eff0963)

closed 2015-11-19 by egavrin